espressif/esp-idf #9987: Change in behaviour between IDFv4.3-dev (and earlier) and IDFv4.4.2 re partition table MD5 is bricking devices during OTA

william-ferguson-au, 2022-10-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-312-g08fa67fe92

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM32

### Power Supply used.

USB

### What is the expected behavior?

I expect that after an OTA update that the app should be able to load an existing partition that it was able to load before the OTA update.

### What is the actual behavior?

For an OTA update of an app built using IDF v4.4.2 if CONFIG_PARTITION_TABLE_MD5=y then the app cannot load any existing partitions if the partition table was originally created without an MD5 checksum.

If the app was built with CONFIG_PARTITION_TABLE_MD5 unset then partition tables can be loaded.

In IDF version 4.3 and earlier, partitions could still be loaded regardless of the value of CONFIG_PARTITION_TABLE_MD5.

### Steps to reproduce.

1. Build and flash an app with CONFIG_PARTITION_TABLE_MD5=y which will generate a partition table with an MD5 checksum.
2. In the app, initialise the flash partition and attempt to open it. Note that that works.
3. Build the same app with CONFIG_PARTITION_TABLE_MD5 unset (do not flash it)
4. Update the app via OTA.
2. In the app, attempt to initialise and open the flash partition and attempt to open it. 
3. Note that this fails - which for many/most app is not recoverable.

### Debug Logs.

```plain
I (29) boot: ESP-IDF v3.3-51-g5e321badd 2nd stage bootloader
I (29) boot: compile time 12:59:52
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (73) boot:  2 app0             OTA app          00 10 00010000 001d0000
I (80) boot:  3 app1             OTA app          00 11 001e0000 001d0000
I (88) boot:  4 hardware         WiFi data        01 02 003b0000 00003000
I (95) boot:  5 spiffs           Unknown data     01 82 003b3000 0004d000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x34bb8 (215992) map
I (192) esp_image: segment 1: paddr=0x00044be0 vaddr=0x3ff80063 size=0x00008 (     8) load
I (192) esp_image: segment 2: paddr=0x00044bf0 vaddr=0x3ffb0000 size=0x03b34 ( 15156) load
I (204) esp_image: segment 3: paddr=0x0004872c vaddr=0x40080000 size=0x078ec ( 30956) load
0x40080000: _WindowOverflow4 at C:/Dev/espressif/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1736

I (220) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0xc945c (824412) map
0x400d0020: _stext at ??:?

I (510) esp_image: segment 5: paddr=0x00119484 vaddr=0x400878ec size=0x0fc98 ( 64664) load
0x400878ec: get_rate_fcc_index at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:1762

I (537) esp_image: segment 6: paddr=0x00129124 vaddr=0x400c0000 size=0x00064 (   100) load
I (537) esp_image: segment 7: paddr=0x00129190 vaddr=0x50000000 size=0x00010 (    16) load
I (557) boot: Loaded app from partition at offset 0x10000
I (557) boot: Disabling RNG early entropy source...
I (558) cpu_start: Pro cpu up.
I (562) cpu_start: Starting app cpu, entry point is 0x40081584
0x40081584: call_start_cpu1 at C:/Dev/espressif/esp-idf/components/esp_system/port/cpu_start.c:148

I (554) cpu_start: App cpu up.
I (583) cpu_start: Pro cpu start user code
I (583) cpu_start: cpu freq: 160000000
I (584) cpu_start: Application information:
I (588) cpu_start: Project name:     my-app
I (594) cpu_start: App version:      23
I (598) cpu_start: Compile time:     Oct 15 2022 20:35:02
I (604) cpu_start: ELF file SHA256:  8845c0977d48c026...
I (610) cpu_start: ESP-IDF:          v4.4.2-312-g08fa67fe92
I (617) heap_init: Initializing. RAM available for dynamic allocation:
I (624) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (630) heap_init: At 3FFB9200 len 00026E00 (155 KiB): DRAM
I (636) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (642) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (649) heap_init: At 40097584 len 00008A7C (34 KiB): IRAM
I (657) spi_flash: detected chip: generic
I (660) spi_flash: flash io: dio
I (666) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (680) trinity_main: Logger initialised
E (685) partition: No MD5 found in partition table
E (690) partition: load_partitions returned 0x105
I (695) wt_hardware: No 'hardware' partition found. Assuming version 1 hardware.
I (703) trinity_main: hardware_version=1
E (708) partition: No MD5 found in partition table
E (713) partition: load_partitions returned 0x105
I (718) wt_flash: Initialised flash errorCode=261
E (724) partition: No MD5 found in partition table
E (729) partition: load_partitions returned 0x105
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4008a854
0x4008a854: _esp_error_check_failed at C:/Dev/espressif/esp-idf/components/esp_system/esp_err.c:42

file: "./components/wylas-libs/wylas-common/src/wt_flash.cpp" line 34
func: void init_flash()
expression: nvs_flash_erase()

abort() was called at PC 0x4008a857 on core 0
0x4008a857: _esp_error_check_failed at C:/Dev/espressif/esp-idf/components/esp_system/esp_err.c:43

Backtrace: 0x40081e5a:0x3ffbbd30 0x4008a861:0x3ffbbd50 0x40091efa:0x3ffbbd70 0x4008a857:0x3ffbbde0 0x400e2431:0x3ffbbe00 0x400d8343:0x3ffbbe20 0x40197c4c:0x3ffbbeb0 0x4008e23d:0x3ffbbed0
0x40081e5a: panic_abort at C:/Dev/espressif/esp-idf/components/esp_system/panic.c:402

0x4008a861: esp_system_abort at C:/Dev/espressif/esp-idf/components/esp_system/esp_system.c:128

0x40091efa: abort at C:/Dev/espressif/esp-idf/components/newlib/abort.c:46

0x4008a857: _esp_error_check_failed at C:/Dev/espressif/esp-idf/components/esp_system/esp_err.c:43

0x400e2431: init_flash() at C:/Source/wylas/trinity/components/wylas-libs/wylas-common/src/wt_flash.cpp:34 (discriminator 1)

0x400d8343: app_main at C:/Source/wylas/trinity/main/src/main.cpp:63

0x40197c4c: main_task at C:/Dev/espressif/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x4008e23d: vPortTaskWrapper at C:/Dev/espressif/esp-idf/components/freertos/port/xtensa/port.c:131
```

### More Information.

Note that this wouldn't potentially be so bad except that the behaviour of the build system changed at some point. 
Originally CONFIG_PARTITION_TABLE_MD5=y meant that the partition did **NOT** get an MD5 checksum.

This means that there are builds out there where the owner thinks they have always been generating an MD5 checksum, but at some point in the past they weren't. But with IDF v4.4.2 CONFIG_PARTITION_TABLE_MD5=y seems to mean 2 things:

1. Generate an MD5 checksum on the partition table
2. **REQUIRE** an MD5 checksum on the partition table.

This means many updates to IDF v4.4.2 will effectively brick devices.

See 

- https://www.esp32.com/viewtopic.php?t=12794
- https://www.esp32.com/viewtopic.php?f=13&t=30094

kriegste, 2022-10-15:
That means, the check is done by the app and not the bootloader?

william-ferguson-au, 2022-10-16:
Correct

Alvin1Zhang, 2022-10-20:
Thanks for reporting, seems the issue has been tracked in https://www.esp32.com/viewtopic.php?f=13&t=30094, will close the issue. Feel free to reopen.

closed 2022-10-20 by Alvin1Zhang

william-ferguson-au, 2022-10-23:
I don't think this has been dealt with in a useful fashion.

Just like every other project, we will be be building the application to be sent OTA as well as flashed onto new devices.

We could build multiple configurations of the same app, one for OTA, one for new devices, but what would be the point?
The new devices would eventually received the OTAs, so they gain no benefit.

Surely the sensible approach is for  
[list=]
[* the build to always generate an MD5 in the partition table if CONFIG_PARTITION_TABLE_MD5=y]
[* If CONFIG_ESP32_COMPATIBLE_PRE_V3_1_BOOTLOADERS=y then to log at warn if the MD5 is missing but allow access]
[/list]

william-ferguson-au, 2022-10-23:
@Alvin1Zhang how do I reopen the issue?