espressif/esp-idf #9989: Menuconfig unexpected behaviour using VS Code with ESP32S3 target

idea--list, 2022-10-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using VS Code (1.72.2) and Espressif IDF plugin (1.5.0) and ESP-IDF (4.4.2) on my Win10x64 machine to write code.
On my ESP32S3 based board i was often facing unexpected behaviour when trying to use light sleep or auto light sleep.
Today i noticed that if i click the cogwheel icon to open menuconfig inside VS Code, then the `Disable all GPIO when chip at sleep` option gets always automatically enabled in case `light sleep GPIO reset workaround` option is also enabled.

This is not what i was expecting. If i manually uncheck `Disable all GPIO when chip at sleep` option and compile the code it will work fine. But as soon as i reopen menuconfig to change any values, then the option is check automatically again and the recompiled code just breaks again. Was not aware of this until now and it is easy to miss as there are so many options. Will have to check if this might the reason behind my previous issue reports with light sleep on the ESP32S3.

I also noticed that the above described behaviour does not occur if `light sleep GPIO reset workaround` is disabled.
Not sure if this only occurs using VS Code and the plugin or it may be even a workaround for some other bug.
Can someone please confirm if it occurs even using ESP-IDF natively or is it VS Code specific?

ESP-Marius, 2023-04-27:
@idea--list Hey, sorry for the very delayed response. Seems like the issue got lost in our backlog.

I think what you are seeing here is as expected. `light sleep GPIO reset workaround` `select `PM_SLP_DISABLE_GPIO`, so if you activate that one it will automatically also enable `Disable all GPIO when chip at sleep`

closed 2023-04-27 by ESP-Marius

idea--list, 2023-04-28:
@ESP-Marius 

Thanks for the feedback. Since that i am configuring all the GPIOs using `gpio_set_direction` `gpio_sleep_set_direction` `gpio_sleep_sel_dis` , etc APIs and those seem to solve most issues i faced before with automatic/default settings. Also power consumption seems to be somewhat lower this way, which is always great.