espressif/esp-idf #9991: IDF 5.0 RMT peripheral features

mitchellcairns, 2022-10-16:
### Is your feature request related to a problem?

With the current RMT peripheral, I can pre-load data into the RMT memory with direct access and control exactly when the transaction is sent. With the new 5.0 driver, direct memory access is depreciated, and there is no means to pre-load transaction data before we initiate the transaction.

### Describe the solution you'd like.

We need some functional equivalent to pre-load the RMT memory before starting the transaction. 

### Describe alternatives you've considered.

There is still the legacy driver; this is obviously not a great long-term solution. We should be able to use the new encoder and driver types to preload the RMT channel memory before sending the transaction. This is huge for time-critical implementations where we need to start a transaction at an exact moment.

suda-morris, 2022-10-17:
Hi @mitchellcairns Direct access to RMT hardware memory is not taken into consideration in the new driver, this is mainly because, on ESP32S3, the RMT also has the DMA access ability. We introduced the **encoder** interface to extract the difference between various kinds of memory access modes. Can your requirement be addressed by customizing another **encoder**? Or do you think we should allow user to pass a `NULL` encoder handle, and in that case, we don't do encoding for the user, and just start the transmission? (this way, user has to manipulate the RMT memory in the application layer, but not easy to implement a ping-pong buffer)

mitchellcairns, 2022-10-17:
I think as an example, the i2c peripheral handles the workflow fairly well. Being able to set up the information that needs to be sent out ahead of time, then only needing to call one function to start the transmission is great. 

Something that functions as below
rmt_tx_encode(rmt_channel_handle_t tx_channel, rmt_encoder_handle_t encoder, const void *payload, size_t payload_bytes)

Ideally, there should be a NULL encoder which allows us to form our own rmt_symbol_word_t buffers without the need to create an encoder. There are a lot of situations where having to set up an encoder only makes the codebase more obtuse when we just need to get some pre-constructed data out. 

I think keeping rmt_transmit with the current function is fine, but being able to split up the acts of loading the RMT channel memory and starting the transmission is vital.

Then just a simple
rmt_tx_start(rmt_channel_handle_t tx_channel, bool continuous_transmission)

There should also be
rmt_tx_stop(rmt_channel_handle_t tx_channel) to stop continuous transmission events. There are instances where we'd want to stop transmission but not stop the whole RMT channel.