espressif/esp-idf #9992: Consequtive Send USB HID Message Cause EP Unusable

MarsTechHAN, 2022-10-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1264-g1c84cfde14

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3 | Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I am trying to process the data from *tud_hid_set_report_cb* and send the reply. The reply may contain multiple frames (the total length may be much greater than 64bytes).
 
I tried to both directly send data in the *tud_hid_set_report_cb*or let *tud_hid_set_report_cb* start a new task to process. Either method would be stuck at tud_hid_n_ready after the first packet is sent.

I am not sure if this is a usage problem or a bug, but I expected the device should be able to send multiple HID reports consecutively.

### What is the actual behavior?

After the first report is sent using *tud_hid_n_report*, the *tud_hid_n_ready* never becomes True.

### Steps to reproduce.

I can just plug the following code into the idf's USB demo.
> esp-idf/examples/peripherals/usb/device/tusb_hid/main/tusb_hid_example_main.c.

```C
void tud_hid_set_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize)
{
    uint8_t frame[64] = {0};
    while (1) {
        ESP_LOGI(TAG, "Seq: %u", frame[0]++);
            while(!tud_hid_n_ready(0));
            tud_hid_n_report(0, 0, frame, 64);
    }
}
```

and the log from the above code:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x16b4
load:0x403c9700,len:0xbd0
load:0x403cc700,len:0x2c64
entry 0x403c9900
[... 500 lines cut ...]
I (14933) FIDO2-IF: New Channel Allocated, CID: 0x6E276269
I (14943) FIDO2-HID: Sending frame, seq: 0
E (19863) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (19863) task_wdt:  - IDLE (CPU 0)
E (19863) task_wdt: Tasks currently running:
E (19863) task_wdt: CPU 0: TinyUSB
E (19863) task_wdt: CPU 1: IDLE
E (19863) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4200CD12:0x3FC92220 0x4200CEBE:0x3FC92240 0x40376DE5:0x3FC92260 0x42008E35:0x3FCEAC50 0x42007EDC:0x3FCEAC70 0x42008171:0x3FCEACD0 0x42007DCD:0x3FCEAD10 0x42007E86:0x3FCEAD30 0x42008656:0x3FCEAD60 0x4200897E:0x3FCEAD80 0x42009A85:0x3FCEADA0 0x4200A01F:0x3FCEADD0 0x4037C8F1:0x3FCEADF0
0x4200cd12: task_wdt_timeout_handling at *****/esp-idf/components/esp_system/task_wdt/task_wdt.c:461 (discriminator 3)

0x4200cebe: task_wdt_isr at *****/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40376de5: _xt_lowint1 at *****/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1117

0x42008e35: tud_mounted at *****/esp-idf/components/tinyusb/tinyusb/src/device/usbd.c:367

0x42007edc: u2f_hid_if_send at *****/fido2_hid.c:126 (discriminator 1)
```
```

### More Information.

I can keep sending the hid report in a thread started by *app_main*, but not the thread started by callbacks.

mitchellcairns, 2022-11-06:
I believe I may be experiencing a related issue to this. When I receive more than one OUT report from the host device, only the first OUT is received. The callback does not seem to fire more than once for whatever reason.

mitchellcairns, 2022-11-07:
I think I found out a solution and this may point to a permanent fix in code to avoid future confusion? It seems that I needed to increase the max buffer size in my Endpoint descriptor. The max packet size was only 5 bytes, but this needed to be increased to 6 to get the actual 5 bytes coming in. Hope this can help someone else. Maybe there's something that's not starting at the right index or is incrementing one. Regardless, I'd consider this a bug on the TinyUSB stack.

chegewara, 2022-11-07:
> I think I found out a solution and this may point to a permanent fix in code to avoid future confusion? It seems that I needed to increase the max buffer size in my Endpoint descriptor. The max [...]

In USB hid packet first byte is always reportID, thats why you need to increase buffer size.

mitchellcairns, 2022-11-07:
> In USB hid packet first byte is always reportID, thats why you need to increase buffer size.

In my program I'm including the report byte in the buffer size. 

Byte 0 - report
Byte 1-4 - data

I do think this is a bug. I have to set it as 6 bytes to get that amount of data through.