espressif/esp-idf #9994: idf v4.3.2运行main_task到vTaskDelete(NULL);后没有释放main_task的heap资源

huangshenghui, 2022-10-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

custom 

### Power Supply used.

External 5V

### What is the expected behavior?

idf v4.3.2运行main_task到vTaskDelete(NULL);后没有释放main_task的heap资源

### What is the actual behavior?

idf v4.3.2运行main_task到vTaskDelete(NULL);后没有释放main_task的heap资源

### Steps to reproduce.

1. Step
2. Step
3. Step
...
idf v4.3.2运行main_task到vTaskDelete(NULL);后没有释放main_task的heap资源

huangshenghui, 2022-10-17:
static void cli_diag_mem_task(void)
{
#ifdef CONFIG_HEAP_TASK_TRACKING
	static size_t count;
	static heap_task_totals_t totals[MAX_TASK_NUM];
	static heap_task_block_t blocks[MAX_BLOCK_NUM];
	heap_task_info_params_t heap_info;
	size_t total_size = 0;
	size_t total_count = 0;
	char *task_name;
	char *p;
	char c;
	int i;
	int j;

	count = 0;
	memset(&heap_info, 0, sizeof(heap_info));

	heap_info.totals = totals;
	heap_info.num_totals = &count;
	heap_info.max_totals = MAX_TASK_NUM;
	heap_info.blocks = blocks;
	heap_info.max_blocks = MAX_BLOCK_NUM;

	heap_caps_get_per_task_info(&heap_info);

	printcli("\nheap usage by task:");
	printcli("   %20s %6s %6s %12s", "task", "size", "count", "handle");

	for (i = 0 ; i < count; i++) {
		task_name = "_none_";
		p = NULL;
		if (heap_info.totals[i].task) {
			task_name = "_unknown_";
			p = pcTaskGetTaskName(heap_info.totals[i].task);
			/*
			 * Only use returned name if it is valid ASCII.
			 * FreeRTOS assumes the task handle is good when it
			 * may not be.
			 */
			for (j = 0; j < 20; ++j) {
				c = p[j];
				if (c < ' ' || c > '~')  {
					break;
				}
			}
			if (j < 20 && c == '\0') {
				task_name = p;
			}
		}
		printcli("   %20s %6u %6u %12p",
		    task_name,
		    heap_info.totals[i].size[0],
		    heap_info.totals[i].count[0],
		    heap_info.totals[i].task);
		total_size += heap_info.totals[i].size[0];
		total_count += heap_info.totals[i].count[0];
	}
	printcli("   %20s %6u %6u", "TOTAL", total_size, total_count);
#endif
}

[image]

ESP-Marius, 2022-10-17:
I think this is as expected. `heap_caps_get_per_task_info` will show which task allocated memory, but this memory is not supposed to be cleaned-up when the task is deleted. 

FreeRTOS will only free memory that is directly allocated for the purpose of task creation, e.g. a task's stack, task control block++.

Any memory allocated in the task function itself (i.e. in `main_task` in this case) must be cleaned up by the user by calling the appropriate clean up functions (e.g. `free` if you allocated the memory yourself with `malloc`)

huangshenghui, 2022-10-17:
static void main_task(void* args)
{
#if !CONFIG_FREERTOS_UNICORE
	// Wait for FreeRTOS initialization to finish on APP CPU, before replacing its startup stack
	while (port_xSchedulerRunning[1] == 0) {
		;
	}
#endif

	// [refactor-todo] check if there is a way to move the following block to esp_system startup
	heap_caps_enable_nonos_stack_heaps();

	// Now we have startup stack RAM available for heap, enable any DMA pool memory
#if CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL
	if (g_spiram_ok) {
		esp_err_t r = esp_spiram_reserve_dma_pool(CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL);
		if (r != ESP_OK) {
			ESP_EARLY_LOGE(TAG, "Could not reserve internal/DMA pool (error 0x%x)", r);
			abort();
		}
	}
#endif

	//Initialize task wdt if configured to do so
#ifdef CONFIG_ESP_TASK_WDT_PANIC
	ESP_ERROR_CHECK(esp_task_wdt_init(CONFIG_ESP_TASK_WDT_TIMEOUT_S, true));
#elif CONFIG_ESP_TASK_WDT
	ESP_ERROR_CHECK(esp_task_wdt_init(CONFIG_ESP_TASK_WDT_TIMEOUT_S, false));
#endif

	//Add IDLE 0 to task wdt
#ifdef CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
	TaskHandle_t idle_0 = xTaskGetIdleTaskHandleForCPU(0);
	if(idle_0 != NULL){
		ESP_ERROR_CHECK(esp_task_wdt_add(idle_0));
	}
#endif
	//Add IDLE 1 to task wdt
#ifdef CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1
	TaskHandle_t idle_1 = xTaskGetIdleTaskHandleForCPU(1);
	if(idle_1 != NULL){
		ESP_ERROR_CHECK(esp_task_wdt_add(idle_1));
	}
#endif

	app_main();
	vTaskDelete(NULL);
}
有没有示例去释放上面的heap_caps_enable_nonos_stack_heaps();所分配的heap

ESP-Marius, 2022-10-18:
`heap_caps_enable_nonos_stack_heaps` release stack/memory used during the bootup process and adds it to the heap pool, but this is done automatically for you before `app_main` runs. 

Not sure what exactly you are asking for here. Please try to describe in details what your issue is and only include code where it is actually relevant for the question.

ESP-Marius, 2022-11-16:
Marking this as answered, feel free to reopen if there is anything left unresolved.

closed 2022-11-16 by ESP-Marius