espressif/esp-idf #10000: MQTT Broker disconnect causes tcp_read error spam

allard-potma, 2022-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1-dev-962-g787c69a2cc

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

No spam.

### What is the actual behavior?

When the brokers shuts down, the mqtt client tcp connection will start to spam 'transport_base: tcp_read error, errno=Socket is not connected' messages. The frequency of these messages causes starvation and triggers the task watchdog. 

### Steps to reproduce.

1. Connect to broker (mosquitto)
2. Exit the broker
3. Spam begins

### Debug Logs.

```plain
<CUT, many more of the same message here>
E (78504) transport_base: tcp_read error, errno=Socket is not connected
E (78514) transport_base: tcp_read error, errno=Socket is not connected
E (78514) transport_base: tcp_read error, errno=Socket is not connected
E (78524) transport_base: tcp_read error, errno=Socket is not connected
E (78534) transport_base: tcp_read error, errno=Socket is not connected
E (78534) transport_base: tcp_read error, errno=Socket is not connected
E (78544) transport_base: tcp_read error, errno=Socket is not connected
E (78554) transport_base: tcp_read error, errno=Socket is not connected
E (78564) transport_base: tcp_read error, errno=Socket is not connected
E (78564) transport_base: tcp_read error, errno=Socket is not connected
E (78574) transport_base: tcp_read error, errno=Socket is not connected
E (78584) transport_base: tcp_read error, errno=Socket is not connected
E (78594) transport_base: tcp_read error, errno=Socket is not connected
E (78594) transport_base: tcp_read error, errno=Socket is not connected
E (78634) transport_base: tcp_read error, errno=Socket is not connected
E (78634) transport_base: tcp_read error, errno=Socket is not connected
E (78634) transport_base: tcp_read error, errno=Socket is not connected
E (78644) transport_base: tcp_read error, errno=Socket is not connected
E (78654) transport_base: tcp_read error, errno=Socket is not connected
E (78664) transport_base: tcp_read error,E (78664) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (78664) task_wdt:  - IDLE (CPU 0)
E (78664) task_wdt: Tasks currently running:
E (78664) task_wdt: CPU 0: mqtt_task
E (78664) task_wdt: CPU 1: IDLE
E (78664) task_wdt: Print CPU 0 (current core)

<CUT backtrace shows it was stuck in uart_write/log_write>

<CUT many more tcp_read error here>

E (82624) transport_base: tcp_read error, errno=Socket is not connected
E (82634) transport_base: tcp_read error, errno=Socket is not connected
E (82634) transport_base: tcp_read error, errno=Socket is not connected
E (82644) transport_base: tcp_read error, errno=Socket is not connected
E (82654) transport_base: tcp_read error, errno=Socket is not connected
E (82654) transport_base: poll_read select error 104, errno = Connection reset by peer, fd = 54
E (82664) mqtt_client: Poll read error: 128, aborting connection
```

david-cermak, 2022-11-02:
Thanks for reporting this bug! We missed one error translation in tcp-transport component, this patch should fix the issue:
[tcp_transport-Fix-error-propagation-on-tcp_read.patc.txt](https://github.com/espressif/esp-idf/files/9922975/tcp_transport-Fix-error-propagation-on-tcp_read.patc.txt)

AxelLin, 2022-12-16:
v5.1-dev: dc1f6b3aefe26541e884e4c59b81287b3e643f0a
v5.0: 3caeaece6b62b29b6d8b8cde312c10869aa6dd25

Alvin1Zhang, 2022-12-16:
Thanks for sharing the fix commits, feel free to reopen.

closed 2022-12-16 by Alvin1Zhang