espressif/esp-idf #10002: Crash when printing after light sleep

EdMlt, 2022-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, i have an esp32 running multiple tasks, I get this issue when I try to printf after light sleep:
[image]

Any idea if it's an idf issue or else? Ask if you need more information?

ESP-Marius, 2022-10-19:
Hi @EdMlt, thanks for reporting this!

Would you be able to describe a bit more in details about the setup when the crash is happening? E.g. how are you entering light sleep, are you using `FREERTOS_USE_TICKLESS_IDLE` or calling `esp_light_sleep_start` manually?

The best would be if you are able to reproduce a minimal example that is able to produce a crash like this.

EdMlt, 2022-10-19:
So I use this function: 
`uint8_t low_power_light_sleep(uint64_t timetosleep_ms){
    esp_sleep_wakeup_cause_t dummy;
    xSemaphoreTake(power_mutex, portMAX_DELAY);
    low_i2c_mutex_take();
    low_nvs_mutex_take();
    
    ESP_LOGI(TAG,"Light sleep mode activated.");

    int64_t start = esp_timer_get_time()/1000;

    fflush(stdout);
    uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
    
    timetosleep_ms = timetosleep_ms *1000;
    esp_sleep_enable_timer_wakeup( timetosleep_ms);
    esp_light_sleep_start();

    low_nvs_mutex_give();
    low_i2c_mutex_give();

    vTaskDelay(150/ portTICK_PERIOD_MS);
    printf("\n\n\r");
}`

It makes the crash at startup when I want to print something, really don't have a clue of what is happening, it do it continuesly. If I reflash the code the error disappear.

EdMlt, 2022-10-19:
I use esp-idf v4.4.2

ESP-Marius, 2022-10-20:
Hmm, when just testing your function (after removing the `low_X` calls) it seems to work fine for me. 

Could you also share:

- Your sdkconfig file
- The content`low_power_wakeup_cause` function where the crash happens

EdMlt, 2022-10-21:
My sdkconfig file:
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

[... 1430 lines cut ...]
uint8_t low_power_wakeup_cause(esp_sleep_wakeup_cause_t *cause){
    *cause = esp_sleep_get_wakeup_cause();
    switch (*cause ) {
        case ESP_SLEEP_WAKEUP_EXT0:
            ESP_LOGI(TAG,"Wake up from EXT0.");
            break;
        case ESP_SLEEP_WAKEUP_EXT1:
            ESP_LOGI(TAG,"Wake up from EXT1.");
            break;
        case ESP_SLEEP_WAKEUP_UART:
            ESP_LOGI(TAG,"Wake up from UART.");
            break;
        case ESP_SLEEP_WAKEUP_TIMER:
            ESP_LOGI(TAG,"Wake up from TIMER.");
            break;
        case ESP_SLEEP_WAKEUP_UNDEFINED:
        default:
            ESP_LOGW(TAG,"Not a deep sleep reset.");
            break;
    }
    return 0;
}
```

The crash might happen even if I printf something just after the light sleep

EdMlt, 2022-10-24:
1. Sorry but I can't reproduce it in a minimal example, it's a small part in a complexe project...
2. I can try but the error happen once in 10 flashing, it's hard to reproduce

EdMlt, 2022-11-06:
It seems happening when I flash the binary with esp-tool while the esp32 is coming in/out light sleep. I can't verify as the monitoring is closes while I begin to flash

ESP-Marius, 2022-11-16:
> It seems happening when I flash the binary with esp-tool while the esp32 is coming in/out light sleep. I can't verify as the monitoring is closes while I begin to flash

Do you mean that it crashes when you start the flashing process? Or does the crash occur after this flashing process at first boot up?

EdMlt, 2022-11-16:
No it crash after the flashing process at first boot up:
When I go to light sleep, it crashes at first print after light sleep and do it everytime until I reflash.
I wonder if it's not linked with the uart and this lines of code:
```
    fflush(stdout);
    uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
```

ESP-Marius, 2022-11-17:
Hmm, this seems very strange to me, dont really have any ideas.

Don't see how re-flashing should fix it if it was a SW error. 

Just to gather some more information that could potentially help give an idea:
* What kind of HW setup is? Dev-kit? Custom PCB? What is connected to the chip?
* How are you flashing?

ESP-Marius, 2023-01-29:
@EdMlt Are you still experiencing this issue and want to continue investigating it?

If not i'll close the issue for now as we cannot replicate it.

closed 2023-03-01 by ESP-Marius