espressif/esp-idf #10003: tinyusb_cdcacm_write_flush timed out while the host receives the message

wuyuanyi135, 2022-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I have a question regarding the tinyUSB CDC device with the `tinyusb_cdcacm_write_flush` function.

I wrote some data to the buffer with `tinyusb_cdcacm_write_queue` then `tinyusb_cdcacm_write_flush` (with zero time out) and it failed with `ESP_ERR_TIMEOUT`. **However, the receiver still receive the message**. I assume this timeout error indicates the flush was not finished in time.

If I allow 10ms timeout it will suppress the error, but as the [documentation ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_device.html#_CPPv426tinyusb_cdcacm_write_flush20tinyusb_cdcacm_itf_t8uint32_t)states:

>  WARNING! TinyUSB can block output Endpoint for several RX callbacks, after will do additional flush
   after the each trasfer. That can leads to the situation when you requested a flush, but it will fail until
   ont of the next callbacks ends.
   SO USING OF THE FLUSH WITH TIMEOUTS IN CALLBACKS IS NOT RECOMENDED - YOU CAN GET A LOCK FOR THE TIMEOUT

So, if the timeout is not recommended in callback. Is there an option to request flush asynchronously? Otherwise, should I simply not to check the ESP_ERR_TIMEOUT error returned by `tinyusb_cdcacm_write_flush`?

Thanks

roma-jam, 2022-11-15:
@wuyuanyi135 Hi! Thanks for the pointing out the problem. I'll try to do as much as can to help.
Can you please clarify a little bit more details about what you are doing to get this ESP_ERR_TIMEOUT? It would helps a lot if I had known the following:
1. What is length of the data have been written? 
2. How often have you write the data to the buffer? (how much s/ms or us between the data writings)

Thanks.

wuyuanyi135, 2022-11-15:
Hi, @roma-jam 
Thanks for taking care of this problem. However, my code for reproducing this code cannot be found. I will try to reproduce it later.

To answer your question:
1. The length is shorter than the tx buffer length so that without flush it won't be sent to the host. If my memory serves, I was writing "Hello world\n" to the host.

2. I remember it was writing "Hello world\n", then flush, then `vTaskDelay` for 1 second. 

Sorry for the fragmented information. I will try to provide a reproducible code later.

roma-jam, 2022-11-15:
@wuyuanyi135 No worries, thanks for the information that has been provided. I think there is nothing to provide more, it is enough. 
Thanks.

wuyuanyi135, 2022-11-15:
Thank you @roma-jam. Let me know if it failed to reproduce the error.

roma-jam, 2022-11-16:
Hi @wuyuanyi135, hope you have a great day. 

Thanks, I have reproduced the error and it seems that I can provide the following information for now. 

1. _Is there an option to request flush asynchronously?_
There are only two options to flush for now: due to the overflow (when the amount of stored data  is more than 64 bytes in fifo) and manual (with the exactly the tinyusb_cdcacm_write_flush function). But there are some extra information which needs to be considered, please read the additional information below. 

2. _Otherwise, should I simply not to check the ESP_ERR_TIMEOUT error returned by tinyusb_cdcacm_write_flush?_
For now, unfortunately yes. When the tinyusb_cdcacm_write_flush returned ESP_ERR_TIMEOUT (when it was called with timeout=0 argument) that means that the data fifo is already flushed and do not contain any data to send. Otherwise, function returns ESP_FAIL. 

**Additional information**
Should be considered that to provide correct data sending the actual sending should be done in the task which is not the same with rx_callback (based on the tusb_serial_device example, and it can be verified with xTaskGetCurrentTaskHandle() macro). 
That is necessary to handle tx completion right. 

I mean, If you would like to try send some data within tinyusb_cdc_rx_callback (e.g to modify existing example), thus during tinyusb_cdcacm_write_flush the actual flush won't be done. This happens because rx callback is run within the same context as tx completion processing and the tx processing is blocked completion event (which is the WARNING is exactly about). 

To determine tx completeion there is a special callback which is called tud_cdc_tx_complete_cb (https://github.com/espressif/arduino-esp32/blob/8ee5f0a11e5423a018e0f89146e05074466274db/tools/sdk/esp32s3/include/arduino_tinyusb/tinyusb/src/class/cdc/cdc_device.h#L139)

And the example of the callback usage is next.
1. Create a separate task to send data after USB initialization is completed
`    
    BaseType_t task_created = xTaskCreate(cdc_xfer_demo_task, "cdc_xfer_demo_task", 2048, (void *) &acm_cfg, 2, NULL);
    assert(task_created); 
`

2. Write task body with sending data (for example, send string "HelloWorld\r" every 500 ms)
`
static void cdc_xfer_demo_task(void *args)
{
	tinyusb_config_cdcacm_t *acm_cfg = (tinyusb_config_cdcacm_t *)args;
	tinyusb_cdcacm_itf_t itf = acm_cfg->cdc_port;

	uint8_t tmp_buf[32] = { 'H', 'e', 'l', 'l', 'o', 'W', 'o', 'r', 'l', 'd', '\r' };

	while (1) {
		vTaskDelay(pdMS_TO_TICKS(500));

		tinyusb_cdcacm_write_queue(itf, tmp_buf, 11);
		tinyusb_cdcacm_write_flush(itf, 0);
	}

	vTaskDelete(NULL);
}
`

3. Declare the callback function:
`
void tud_cdc_tx_complete_cb(uint8_t itf)
{
	ESP_LOGI(TAG, "tx complete cb");
}
`

At this case, the tud_cdc_tx_complete_cb will be triggered every time, when the tx fifo cleared and ready to the next portion of a data. 

Hope, that can help. If there will be more question, please let me know. 

Lots of best wishes,
RJ

Alvin1Zhang, 2023-01-12:
Close with https://github.com/espressif/idf-extra-components/pull/115#pullrequestreview-1222747039. Feel free to reopen.

closed 2023-01-12 by Alvin1Zhang