espressif/esp-idf #10005: esp_hid_host fails to pair with BT Classic 2.0 trackpad: Authentication fail reason 5

boernsen-development, 2022-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1264-g1c84cfde14

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32 Dev Kit C V4

### Power Supply used.

USB

### What is the expected behavior?

I expected the esp_hid_host demo to successfully pair with my BT Classic 2.0 trackpad or print an informative error or reference to further documentation for help to debug this.

### What is the actual behavior?

Instead it prints:

> E (9083) BT_HIDH: OPEN ERROR: ERR
> E (9083) ESP_HIDH_DEMO:  OPEN failed!
> W (9233) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x5
> E (9243) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5

### Steps to reproduce.

1. Check out 1c84cfde14
2. Build, flash and monitor esp_hid_host example
3. Put the trackpad into pairing mode (I used an Apple Magic Trackpad 1. gen)
4. Reset the ESP32
5. ESP32 scan seems to find the trackpad and it tries to connect
6. **BT_HIDH: OPEN ERROR: ERR**

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7032
load:0x40078000,len:15376
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080648
I (27) boot: ESP-IDF v5.1-dev-1264-g1c84cfde14-dirty 2nd stage bootloader
I (27) boot: compile time Oct 18 2022 21:04:54
I (30) boot: chip revision: V300
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19050h (102480) map
I (150) esp_image: segment 1: paddr=00029078 vaddr=3ffbdb60 size=04930h ( 18736) load
I (157) esp_image: segment 2: paddr=0002d9b0 vaddr=40080000 size=02668h (  9832) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6a114h (434452) map
I (321) esp_image: segment 4: paddr=0009a13c vaddr=40082668 size=14fa4h ( 85924) load
I (357) esp_image: segment 5: paddr=000af0e8 vaddr=50000000 size=00010h (    16) load
I (369) boot: Loaded app from partition at offset 0x10000
I (369) boot: Disabling RNG early entropy source...
I (381) cpu_start: Pro cpu up.
I (381) cpu_start: Starting app cpu, entry point is 0x400812d8
0x400812d8: call_start_cpu1 at /home/boernsen/esp-idf/master/esp-idf/components/esp_system/port/cpu_start.c:146

I (0) cpu_start: App cpu up.
I (397) cpu_start: Pro cpu start user code
I (397) cpu_start: cpu freq: 160000000 Hz
I (397) cpu_start: Application information:
I (402) cpu_start: Project name:     esp_hid_host_master_20221014155
I (409) cpu_start: App version:      01de37b
I (414) cpu_start: Compile time:     Oct 18 2022 21:06:39
I (420) cpu_start: ELF file SHA256:  a6d5810410595401...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 77d302ee3824615048f4b1ee6271079fbde723d30cec105d15b0b7ced1f6cfe0
I (426) cpu_start: ESP-IDF:          v5.1-dev-1264-g1c84cfde14-dirty
I (433) heap_init: Initializing. RAM available for dynamic allocation:
I (440) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (446) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (452) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (458) heap_init: At 3FFC6680 len 00019980 (102 KiB): DRAM
I (465) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (471) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (477) heap_init: At 4009760C len 000089F4 (34 KiB): IRAM
I (485) spi_flash: detected chip: generic
I (488) spi_flash: flash io: dio
W (492) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (507) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (553) ESP_HIDH_DEMO: setting hid gap, mode:2
I (553) BTDM_INIT: BT controller compile version [8020d24]
I (553) system_api: Base MAC address is not set
I (553) system_api: read default base MAC address from EFUSE
I (563) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1233) ESP_HIDH_DEMO: EVENT: 5
I (1233) ESP_HIDH_DEMO: SCAN...
BT : aa:bb:cc:dd:ee:ff, COD: major: PERIPHERAL, minor: 37, service: 0x001, RSSI: -43
BT : aa:bb:cc:dd:ee:ff, COD: major: PERIPHERAL, minor: 37, service: 0x001, RSSI: -42
BT : aa:bb:cc:dd:ee:ff, BDNAME: Trackpad
I (5403) ESP_HIDH_DEMO: SCAN: 1 results
  BT : aa:bb:cc:dd:ee:ff, RSSI: -42, USAGE: GENERIC, COD: PERIPHERAL[MOUSE+TABLET] srv 0x001, UUID16: 0x0000, NAME: Trackpad 
W (8413) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (8573) BT_SDP: process_service_attr_rsp

W (8883) BT_SDP: process_service_attr_rsp

E (9083) BT_HIDH: OPEN ERROR: ERR
E (9083) ESP_HIDH_DEMO:  OPEN failed!
W (9233) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x5
E (9243) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5
```

### More Information.

Possibly related to #8027.

boernsen-development, 2022-10-21:
The workaround from #8027 (i.e. using ESP-IDF v4.3.1) did not work (see [comment](https://github.com/espressif/esp-idf/issues/8027#issuecomment-1287210321)), so this might be a different issue here.

boernsen-development, 2022-10-21:
Could it have something to do with the pairing passkey/PIN of the trackpad? It has a default configured passkey 0000 and I can't change it. When running the demo I am never asked about a passkey, nor did I configure anything related.

boernsen-development, 2022-10-22:
Test with an Apple Magic Mouse (**no passkey!**): same result

> I (13353) ESP_HIDH_DEMO: SCAN: 1 results
>   BT : aa:bb:cc:dd:ee:ff, RSSI: -51, USAGE: GENERIC, COD: PERIPHERAL[MOUSE] srv 0x001, UUID16: 0x0000, NAME: Apple Magic Mouse 
> W (17793) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
> W (17923) BT_SDP: process_service_attr_rsp
> 
> W (18183) BT_SDP: process_service_attr_rsp
> 
> E (18363) BT_HIDH: OPEN ERROR: ERR
> E (18363) ESP_HIDH_DEMO:  OPEN failed!
> W (18503) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x5
> E (18513) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: aa:bb:cc:dd:ee:ff
> E (18523) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5

So far I could only test Apple products as I don't have any other devices available right now. It would be interesting to see whether the same happens with a BT mouse/keyboard of a different brand, just to rule out brand specific reasons. The pairing passkey apparently doesn't change anything.

BetterJincheng, 2022-10-24:
@boernsen-development Please try with 
```
static void bt_gap_event_handler(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    ESP_LOGW(TAG, "==> BT GAP EVENT %s", bt_gap_evt_str(event));

    switch (event) {
    case ESP_BT_GAP_DISC_STATE_CHANGED_EVT: {
        ESP_LOGV(TAG, "BT GAP DISC_STATE %s", (param->disc_st_chg.state == ESP_BT_GAP_DISCOVERY_STARTED) ? "START" : "STOP");
        if (param->disc_st_chg.state == ESP_BT_GAP_DISCOVERY_STOPPED) {
            SEND_BT_CB();
        }
        break;
    }
    case ESP_BT_GAP_DISC_RES_EVT: {
        handle_bt_device_result(&param->disc_res);
        break;
    }
    case ESP_BT_GAP_KEY_NOTIF_EVT:
        ESP_LOGI(TAG, "BT GAP KEY_NOTIF passkey:%d", param->key_notif.passkey);
        break;
    case ESP_BT_GAP_MODE_CHG_EVT:
        ESP_LOGI(TAG, "BT GAP MODE_CHG_EVT mode:%d", param->mode_chg.mode);
        break;
    case ESP_BT_GAP_CFM_REQ_EVT:
        ESP_LOGI(BT_AV_TAG, "ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: %d", param->cfm_req.num_val);
        esp_bt_gap_ssp_confirm_reply(param->cfm_req.bda, true);
        break;
    default:
        ESP_LOGV(TAG, "BT GAP EVENT %s", bt_gap_evt_str(event));
        break;
    }
}
```
Please attach the logs with modified `bt_gap_event_handler()` in `examples/bluetooth/esp_hid_host/main/esp_hid_gap.c`.

boernsen-development, 2022-10-26:
Thank you for your response!

I tried that and got this build error with the code above:
> ../main/esp_hid_gap.c:411:18: error: 'BT_AV_TAG' undeclared (first use in this function)

So I replaced _BT_AV_TAG_ by _TAG_.
Hope this was correct?

This is the error log I get:
> E (12391) BT_HIDH: OPEN ERROR: ERR
> E (12391) ESP_HIDH_DEMO:  OPEN failed!
> W (12541) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x5
> E (12541) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: aa:bb:cc:dd:ee:ff
> E (12551) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5
> W (12551) ESP_HID_GAP: ==> BT GAP EVENT AUTH_CMPL

Does this help? Please let me know if you need more.

BetterJincheng, 2022-10-27:
Please attach the whole log.
Thank you!

BetterJincheng, 2022-10-27:
@boernsen-development  I guess I have located the problem. Please test again with:

```
static void bt_gap_event_handler(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    switch (event) {
    case ESP_BT_GAP_DISC_STATE_CHANGED_EVT: {
        ESP_LOGV(TAG, "BT GAP DISC_STATE %s", (param->disc_st_chg.state == ESP_BT_GAP_DISCOVERY_STARTED) ? "START" : "STOP");
        if (param->disc_st_chg.state == ESP_BT_GAP_DISCOVERY_STOPPED) {
            SEND_BT_CB();
        }
        break;
    }
    case ESP_BT_GAP_DISC_RES_EVT: {
        handle_bt_device_result(&param->disc_res);
        break;
    }
    case ESP_BT_GAP_KEY_NOTIF_EVT:
        ESP_LOGI(TAG, "BT GAP KEY_NOTIF passkey:%d", param->key_notif.passkey);
        break;
    case ESP_BT_GAP_MODE_CHG_EVT:
        ESP_LOGI(TAG, "BT GAP MODE_CHG_EVT mode:%d", param->mode_chg.mode);
        break;
    case ESP_BT_GAP_PIN_REQ_EVT:
        ESP_LOGI(TAG, "BT GAP MODE_CHG_EVT mode:%d", param->mode_chg.mode);
        if (param->pin_req.min_16_digit) {
            ESP_LOGI(TAG, "Input pin code: 0000 0000 0000 0000");
            esp_bt_pin_code_t pin_code = {0};
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 16, pin_code);
        } else {
            ESP_LOGI(TAG, "Input pin code: 1234 or 0000");
            esp_bt_pin_code_t pin_code;
            /* this type */
            pin_code[0] = '1';
            pin_code[1] = '2';
            pin_code[2] = '3';
            pin_code[3] = '4';
            // /* or, this type */
            // pin_code[0] = '0';
            // pin_code[1] = '0';
            // pin_code[2] = '0';
            // pin_code[3] = '0';
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 4, pin_code);
        }
        break;
    default:
        ESP_LOGV(TAG, "BT GAP EVENT %s", bt_gap_evt_str(event));
        break;
    }
}
```
and
```
static esp_err_t init_bt_gap(void)
{
    esp_err_t ret;
    // esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    // esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    // esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
    /*
     * Set default parameters for Legacy Pairing
     * Use fixed pin code
     */
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
    esp_bt_pin_code_t pin_code;
    esp_bt_gap_set_pin(pin_type, 0, pin_code);

    if ((ret = esp_bt_gap_register_callback(bt_gap_event_handler)) != ESP_OK) {
        ESP_LOGE(TAG, "esp_bt_gap_register_callback failed: %d", ret);
        return ret;
    }

    // Allow BT devices to connect back to us
    if ((ret = esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE)) != ESP_OK) {
        ESP_LOGE(TAG, "esp_bt_gap_set_scan_mode failed: %d", ret);
        return ret;
    }
    return ret;
}
```

Wait for your reply.
Thanks!

BetterJincheng, 2022-10-27:
@boernsen-development As your device is BT Classic 2.0 which does not support Secure Simple Paring and use a FIXED PIN CODE. In this way, we let HOST to reply the specific pin code '0000' or '1111'.

boernsen-development, 2022-10-27:
@BetterJincheng Thank you for the good idea!

I indeed got one step further now. Actually, I tried to turn off Secure Simple Pairing from idf.py menuconfig before, but that produces these build errors on the master branch:

```
/home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(esp_hid_gap.c.obj):(.literal.init_bt_gap+0x10): undefined reference to `esp_bt_gap_set_security_param'
/home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(esp_hid_gap.c.obj): in function `init_bt_gap':
/home/boernsen/eclipse-workspace/esp_hid_host_master_20221021_201725_ae30c509ef/build/../main/esp_hid_gap.c:419: undefined reference to `esp_bt_gap_set_security_param'
```

Anyway, with your changes (default sdkconfig) there is a new error now. Here is full log:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7024
load:0x40078000,len:15400
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080648
I (27) boot: ESP-IDF v5.1-dev-1401-gae30c509ef 2nd stage bootloader
I (27) boot: compile time Oct 27 2022 20:14:49
I (29) boot: chip revision: V300
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=25ae0h (154336) map
I (168) esp_image: segment 1: paddr=00035b08 vaddr=3ffbdb60 size=04940h ( 18752) load
I (176) esp_image: segment 2: paddr=0003a450 vaddr=40080000 size=05bc8h ( 23496) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=980b0h (622768) map
I (411) esp_image: segment 4: paddr=000d80d8 vaddr=40085bc8 size=11a88h ( 72328) load
I (441) esp_image: segment 5: paddr=000e9b68 vaddr=50000000 size=00010h (    16) load
I (453) boot: Loaded app from partition at offset 0x10000
I (453) boot: Disabling RNG early entropy source...
I (465) cpu_start: Pro cpu up.
I (465) cpu_start: Starting app cpu, entry point is 0x400812e0
0x400812e0: call_start_cpu1 at /home/boernsen/esp-idf/master/esp-idf/components/esp_system/port/cpu_start.c:146

I (0) cpu_start: App cpu up.
I (481) cpu_start: Pro cpu start user code
I (481) cpu_start: cpu freq: 160000000 Hz
I (481) cpu_start: Application information:
I (486) cpu_start: Project name:     esp_hid_host
I (491) cpu_start: App version:      4cba8c8-dirty
I (497) cpu_start: Compile time:     Oct 27 2022 20:14:40
I (503) cpu_start: ELF file SHA256:  338d334c51055515...
I (509) cpu_start: ESP-IDF:          v5.1-dev-1401-gae30c509ef
I (516) heap_init: Initializing. RAM available for dynamic allocation:
I (523) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (529) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (535) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (541) heap_init: At 3FFC82C0 len 00017D40 (95 KiB): DRAM
I (547) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (553) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (560) heap_init: At 40097650 len 000089B0 (34 KiB): IRAM
I (568) spi_flash: detected chip: generic
[... 27 lines cut ...]
W (11411) BT_SDP: process_service_attr_rsp

W (11721) BT_SDP: process_service_attr_rsp

I (11791) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:0
I (11791) ESP_HID_GAP: Using pin code: 0000
W (12001) BT_APPL: new conn_srvc id:23, app_id:1
I (12011) ESP_HIDH_DEMO: aa:bb:cc:dd:ee:ff OPEN: 
BDA:aa:bb:cc:dd:ee:ff, Status: OK, Connected: YES, Handle: 0, Usage: MOUSE
Name: , Manufacturer: , Serial Number: 
PID: 0x030e, VID: 0x05ac, VERSION: 0x0160
Report Map Length: 116
    VENDOR   INPUT REPORT, ID:  19, Length:   1
    VENDOR FEATURE REPORT, ID:  85, Length:  64
   GENERIC FEATURE REPORT, ID:  71, Length:   1
     MOUSE   INPUT   BOOT, ID:   2, Length:   3
     MOUSE   INPUT REPORT, ID:   2, Length:   3
E (12181) BT_HIDH: Not find report handle: 0 mode: REPORT
E (12181) BT_HIDH: Not find report handle: 0 mode: REPORT
W (12321) BT_HCI: hcif mode change: hdl 0x81, mode 2, intv 18, status 0x0
I (12321) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:2
E (12421) BT_HIDH: Not find report handle: 0 mode: REPORT
```

Do you have an idea how to solve that? (Should I open a new issue?)

boernsen-development, 2022-10-27:
I also don't understand why I got the same original error with the Apple Magic Mouse (which has no PIN). However, with the latest master I realized the Magic Mouse seems to connect. So I tried again with the Trackpad.

After I changed the pin_code from esp_hid_gap.c from originally 1,2,3,4 to 0,0,0,0:
```
static esp_err_t init_bt_gap(void)
{
    esp_err_t ret;
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
    /*
     * Set default parameters for Legacy Pairing
     * Use fixed pin code
     */
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_FIXED;
    esp_bt_pin_code_t pin_code;
    pin_code[0] = '0';
    pin_code[1] = '0';
    pin_code[2] = '0';
    pin_code[3] = '0';
    esp_bt_gap_set_pin(pin_type, 4, pin_code);

    if ((ret = esp_bt_gap_register_callback(bt_gap_event_handler)) != ESP_OK) {
        ESP_LOGE(TAG, "esp_bt_gap_register_callback failed: %d", ret);
        return ret;
    }

    // Allow BT devices to connect back to us
    if ((ret = esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE)) != ESP_OK) {
        ESP_LOGE(TAG, "esp_bt_gap_set_scan_mode failed: %d", ret);
        return ret;
    }
    return ret;
}
```
**It actually connects!!! (at least most of the time)**

BetterJincheng, 2022-10-28:
@boernsen-development I think you can open a new issue to solve the new problem. Thanks!

boernsen-development, 2022-10-28:
@BetterJincheng Thank you very much for all your help!

I'm actually very confused now. With the latest master (2d9e75bf05), it all works without any changes. However, after that I tried release v5.0 and it also works!! I don't get why as it doesn't ask for a PIN and the only PIN I can find in the sources is 1234 whereas the trackpad actually has PIN 0000. Still it connects with both versions no matter if I deactivate Simple Pairing Mode or not... (To build with deactivated Simple Pairing Mode, I still need the workaround from #10069)

I actually felt I was quiet meticulous when testing to avoid overseeing anything. The only explanations I can come up with are:
- I did something different when putting the trackpad into pairing mode. Recently, after using the power button to put the trackpad into pairing mode, I tend to hold down the power button while the LED blinks indicating pairing mode. Maybe that deactivates the PIN for that time or changes the pairing behavior? I don't know as there is no proper documentation... shame on Apple
- The trackpad behaves differently to the ESP32 after it has been successfully paired once.

Quiet mysterious for me...

closed 2022-11-28 by espressif-bot