espressif/esp-idf #10007: WiFi Provisioning Manager current state update?

lawrenceko-intecular, 2022-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

The BLE mechanism WiFi provisioning manager with security 1 is used. We want to know if the device is currently being connected (or is in pairing mode) once the PoP has been entered and is correct. Is there any event_handler event that can be used to detect this?

The console showed the below log when PoP is entered.
```
I (214581) NimBLE: GAP procedure initiated: advertise; 
I (214581) NimBLE: disc_mode=2
I (214581) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (214591) NimBLE: 
```

Thank you

laukik-hase, 2022-10-21:
Hello, @lawrenceko-intecular!

Thank you for the report. If I understand correctly, do you want an event when the BLE host (Android, iOS) is successfully paired and connected with the target device (ESP32)? The PoP is entered after the connection is established and has nothing to do with the transport establishment.

lawrenceko-intecular, 2022-10-21:
Hello @laukik-hase ,

The objective is to identify which device is currently being paired up (e.g. show led blink), and the best way is to have an event notice after PoP connection is established besides checking if BLE host is paired as any device can connect to its BLE host.

Thank you.

laukik-hase, 2022-10-21:
Hello, @law-ko!

Could you please check if the attached patch works for you? I have added events for transport connection establishment and termination. Please verify with the `wifi_prov_mgr` example.
[wifi_prov_event_transport_paired.zip](https://github.com/espressif/esp-idf/files/9839010/wifi_prov_event_transport_paired.zip)

P.S. - Use the command `git am <diff.patch>` to apply the patch over the current master branch @ https://github.com/espressif/esp-idf/commit/6a8810fa397d4696f1aa0650ec106040e9816e64.

law-ko, 2022-10-22:
Hello @laukik-hase ,

I am using v4.4.2 ESP-IDF and I looked into your patch it seems like you added `PROTOCOMM_TRANSPORT_BLE_EVENT` in the event handler, is `PROTOCOMM_TRANSPORT_BLE_EVENT` a new event added in ESP-IDF v5? It shows undeclared when I try to use it.

Thank you

laukik-hase, 2022-10-26:
Hello, @law-ko!

The event `PROTOCOMM_TRANSPORT_BLE_EVENT` has been added to ESP-IDF in this patch.
This patch is just a proposed solution to verify whether the solution works for you. If yes, I'll go forward and get it merged internally.

As mentioned earlier, this patch is applicable to the `master` branch only (@ https://github.com/espressif/esp-idf/commit/6a8810fa397d4696f1aa0650ec106040e9816e64) and not to the `v4.4` release. We generally do not backport features like these to the previous releases (e.g. `v4.4`) but I'll discuss it internally and get back to you.

closed 2022-11-02 by espressif-bot

herculanodavi, 2023-04-11:
Hi, is this feature available in 5.0?