espressif/esp-idf #9996: is  there any way to get a device mac within lan

featherfly, 2022-10-18:
### Is your feature request related to a problem?

is there any way to get a device mac within lan, such as `arp -a 192.168.x.x`

MaxwellAlan, 2024-01-15:
Hi @featherfly 

In response to your request, please clarify the specific usage requirements. 

From our perspective, implementing the 'arp -a' functionality within an IoT chip may not be highly significant. This is because the lwIP protocol stack internally updates and maintains the ARP table regularly. For upper-layer applications, obtaining the MAC address of the counterpart may not be crucial. In most cases, communication at the upper layers is conducted using IP addresses.

MaxwellAlan, 2024-01-15:
PS: you can get the arp table from here: https://github.com/espressif/esp-lwip/blob/f5c43549e5d12ec2105a43c27986416124143e6e/src/core/ipv4/etharp.c#L105

Sherry616, 2024-03-18:
Hi @featherfly, could you please share your latest status of this issue? Thanks.

closed 2024-06-07 by Sherry616