espressif/esp-idf #9999: [ESP-MESH] esp_wifi_scan_start always return -1 on MESH_EVENT_NO_PARENT_FOUND

KonssnoK, 2022-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-381-gc97755c9ae5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

Issue that can be reproduced with the same code as 
https://github.com/espressif/esp-idf/issues/9955

If the mesh fails to find a parent node i expect it to be able to scan for WIFI, especially if not on self-organizing network.

### What is the actual behavior?

```
W (209874) mesh_main: Checking wifi connectivity (disconnect 1)
I (209874) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
W (209874) wifi:I (209874) mesh: [wifi]disconnected reason:106(scan fail), continuous:5/max:12, root, vote(,stopped)<><>
sta_scan: STA is connecting, scan are not allowed!
W (209894) mesh_main: esp_wifi_scan_start -1
```

### Steps to reproduce.

Code from 
https://github.com/KonssnoK/esp-idf/commit/9dacebda0b32c4367551b2fdfc3b8dc991e8f16e
on ESP32S3-DevKitC

### Debug Logs.

```plain
I (178064) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x1, look_for_nwk_count:60
I (179764) mesh: [FIND][ch:0]AP:0, otherID:0, MAP:0, idle:0, candidate:0, root:0[00:00:00:00:00:00]<scan router>
I (179764) mesh: [FIND:60]fail to find a network, channel:0, cfg<channel:0, router:KI, 00:00:00:00:00:00>

I (179774) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:61
I (179784) mesh: <MESH_NWK_LOOK_FOR_NETWORK>no parent found, set ROOTLESS, scan times:60, heap:248756
I (179794) mesh: <MESH_NWK_LOOK_FOR_NETWORK>router<cfg:00:00:00:00:00:00> is allowed to switch, find routers with SSID:KI
I (179794) mesh_main: <MESH_EVENT_NO_PARENT_FOUND>scan times:60
I (179804) mesh: <MESH_NWK_PARENT_DISCONNECTED>non-root, need_reselect:1
W (179814) mesh_main: Perform BOOT FIXED ROOT handover
I (179814) mesh: <MESH_NWK_LOOK_FOR_NETWORK>channel<cfg:0> is allowed to switch, do a full channel scan
W (179824) mesh_main: Triggering FIXED ROOT handover
I (179834) mesh: [IO]disable self-organizing<stop reconnect>
I (179844) wifi:mode : sta (7c:df:a1:e0:74:40) + softAP (7c:df:a1:e0:74:41)
W (179854) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
I (179854) wifi:Total power save buffer number: 16
I (179874) mesh: [IO]disable self-organizing<reconnect>
I (179874) mesh: [CONFIG]connect to router:KI, 00:00:00:00:00:00
W (179874) mesh_main: Modem start
I (179874) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
I (179874) mesh: <WIFI_EVENT_SCAN_DONE>status:fail, num:0, id:188
I (182534) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,stopped)<><>
I (182534) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (185194) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,stopped)<><>
I (185194) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (187854) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (187854) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, root, vote(,stopped)<><>
I (190514) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, root, vote(,stopped)<><>
I (190514) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (193184) mesh: [wifi]disconnected reason:201(), continuous:5/max:12, root, vote(,stopped)<><>
I (193184) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (195844) mesh: [wifi]disconnected reason:201(), continuous:6/max:12, root, vote(,stopped)<><>
I (195844) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (198504) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, root, vote(,stopped)<><>
I (198504) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (201174) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,stopped)<><>
I (201174) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (203834) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,stopped)<><>
I (203834) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (206494) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, root, vote(,stopped)<><>
I (206494) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (209154) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, root, vote(,stopped)<><>
I (209154) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
W (209874) mesh_main: Checking wifi connectivity (disconnect 1)
I (209874) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
W (209874) wifi:I (209874) mesh: [wifi]disconnected reason:106(scan fail), continuous:5/max:12, root, vote(,stopped)<><>
sta_scan: STA is connecting, scan are not allowed!
W (209894) mesh_main: esp_wifi_scan_start -1
I (209874) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:106
I (212554) mesh: [wifi]disconnected reason:201(), continuous:6/max:12, root, vote(,stopped)<><>
I (212554) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (215214) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, root, vote(,stopped)<><>
I (215214) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (217874) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,stopped)<><>
I (217874) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (220534) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,stopped)<><>
I (220534) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (223194) mesh: [wifi]disconnected reason:201(), continuous:3/max:12, root, vote(,stopped)<><>
I (223194) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (225854) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, root, vote(,stopped)<><>
I (225854) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (228514) mesh: [wifi]disconnected reason:201(), continuous:5/max:12, root, vote(,stopped)<><>
I (228514) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (231174) mesh: [wifi]disconnected reason:201(), continuous:6/max:12, root, vote(,stopped)<><>
I (231174) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (233834) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (233834) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, root, vote(,stopped)<><>
I (236504) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (236504) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,stopped)<><>
I (239174) mesh: [wifi]disconnected reason:201(), continuous:2/max:12, root, vote(,stopped)<><>
I (239174) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
W (239874) mesh_main: Checking wifi connectivity (disconnect 1)
I (239874) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
W (239874) wifi:I (239874) mesh: [wifi]disconnected reason:106(scan fail), continuous:3/max:12, root, vote(,stopped)<><>
sta_scan: STA is connecting, scan are not allowed!
W (239894) mesh_main: esp_wifi_scan_start -1
```
```

### More Information.

@shenjun7 might be related to the issue already under investigation

KonssnoK, 2022-10-24:
rebased 
https://github.com/KonssnoK/esp-idf/tree/bug/mesh_wifi
on top of latest 
755ce1077df771ee872871ea03f5ce747ff99bfa

to include latest wifi library

Alvin1Zhang, 2022-11-22:
This seems duplicate https://github.com/espressif/esp-idf/issues/9955, let us close this and track there, thanks.

closed 2022-11-22 by Alvin1Zhang

KonssnoK, 2022-11-29:
Hi @Alvin1Zhang 
it is related to the #9955 but the reported issue is different.
in 9955 the focus is on the system stopping to work correctly, 
in this the focus is on the wifi scanning permanently failing when NO_PARENT_FOUND is triggered!

Please let me know if you have addressed this issue.

Thanks!

zhangyanjiaoesp, 2022-11-29:
This [comment](https://github.com/espressif/esp-idf/issues/9955#issuecomment-1313610839) has solved your problem. 
In your original code, the API is called in the following order：
> esp_mesh_set_self_organized(false,false);
esp_mesh_set_type(MESH_ROOT/IDLE)

This will lead the internal reconnection parameter to be RECONNECT_ENABLE, and the mesh node will reconnect to the same AP, when you call scan in the application layer, the scan will be canceled due to connecting. Thus `W (239894) mesh_main: esp_wifi_scan_start -1`

after changing the code, the internal reconnection parameter can be set to RECONNECT_DISABLE correctly, then the application layer scan can work.

KonssnoK, 2022-11-29:
ah ok, makes sense! thanks @zhangyanjiaoesp, as soon as i have back my hw i will restart my mesh tests!
Thanks again!