espressif/esp-idf #10008: example user_event_loops crashed on ESP32S3

gazam2842, 2022-10-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DevkitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

 log should printed and task running 

I (296) user_event_loops: setting up
I (296) user_event_loops: starting event source
I (296) user_event_loops: starting application task
I (296) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_without_task, iteration 1 out of 10
I (316) user_event_loops: application_task: running application task
I (326) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_without_task, iteration 1
I (826) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_with_task, iteration 2 out of 10
I (826) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_with_task, iteration 2
I (1326) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_without_task, iteration 3 out of 10
I (1326) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_without_task, iteration 3
I (1426) user_event_loops: application_task: running application task
I (1826) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_with_task, iteration 4 out of 10
I (1826) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_with_task, iteration 4
I (2326) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_without_task, iteration 5 out of 10
I (2326) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_without_task, iteration 5
I (2526) user_event_loops: application_task: running application task
I (2826) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_with_task, iteration 6 out of 10
I (2826) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_with_task, iteration 6
I (3326) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_without_task, iteration 7 out of 10
I (3326) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_without_task, iteration 7
I (3626) user_event_loops: application_task: running application task
I (3826) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_with_task, iteration 8 out of 10
I (3826) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_with_task, iteration 8
I (4326) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_without_task, iteration 9 out of 10
I (4326) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_without_task, iteration 9
I (4726) user_event_loops: application_task: running application task
I (4826) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_with_task, iteration 10 out of 10
I (4826) user_event_loops: handling TASK_EVENTS:TASK_ITERATION_EVENT from loop_with_task, iteration 10
I (5826) user_event_loops: application_task: running application task
I (5826) user_event_loops: deleting task event source
I (6926) user_event_loops: application_task: running application task
I (8026) user_event_loops: application_task: running application task
I (9126) user_event_loops: application_task: running application task
...

### What is the actual behavior?

I'm runing user_event_loops example on ESP32S3 devkit 
It crashed with stack overflow message below. 

"***ERROR*** A stack overflow in task application_tas has been detected."

I change application_task stack size from 2048 to 3072 it works. 
but I wonder why this happened. 

I tested same source on ESP32 devkit. it works. 

### Steps to reproduce.

1. build user_events_loop  example in IDF 4.4.2 
2. run 
3. get stack overflow error. 

### Debug Logs.

```plain
I (151) cpu_start: App cpu up.
I (201) cpu_start: Pro cpu start user code
I (201) cpu_start: cpu freq: 160000000
I (201) cpu_start: Application information:
I (204) cpu_start: Project name:     user_event_loops
I (209) cpu_start: App version:      1
I (214) cpu_start: Compile time:     Oct 19 2022 16:22:18
I (220) cpu_start: ELF file SHA256:  3f3b4747afb35ba9...
I (226) cpu_start: ESP-IDF:          v4.4.2-dirty
I (231) heap_init: Initializing. RAM available for dynamic allocation:
I (239) heap_init: At 3FC93AB0 len 0004C550 (305 KiB): D/IRAM
I (245) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (252) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (258) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (265) spi_flash: detected chip: generic
I (269) spi_flash: flash io: dio
I (273) sleep: Configure to isolate all GPIO pins in sleep state
I (280) sleep: Enable automatic switching of GPIO sleep configuration
I (287) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (307) user_event_loops: setting up
I (307) user_event_loops: starting event source
I (307) user_event_loops: starting application task
I (307) user_event_loops: posting TASK_EVENTS:TASK_ITERATION_EVENT to loop_without_task, iteration 1 out of 10
I (327) user_event_loops: application_task: running application task
I (337) user_event_
***ERROR*** A stack overflow in task application_tas has been detected.

Backtrace:0x403759f6:0x3fce1cb00x40378f41:0x3fce1cd0 0x4037be5e:0x3fce1cf0 0x4037ab13:0x3fce1d70 0x40378ff8:0x3fce1d90 0x40378fee:0xa5a5a5a5  |<-CORRUPTED
0x403759f6: panic_abort at C:/Users/gazam/esp/esp-idf-v4.4.2/components/esp_system/panic.c:402

0x40378f41: esp_system_abort at C:/Users/gazam/esp/esp-idf-v4.4.2/components/esp_system/esp_system.c:128

0x4037be5e: vApplicationStackOverflowHook at C:/Users/gazam/esp/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:394

0x4037ab13: vTaskSwitchContext at C:/Users/gazam/esp/esp-idf-v4.4.2/components/freertos/tasks.c:3505

0x40378ff8: _frxt_dispatch at C:/Users/gazam/esp/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:436

0x40378fee: _frxt_int_exit at C:/Users/gazam/esp/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:231
```

ESP-Marius, 2022-10-19:
Hi @gazam2842, thanks for reporting this!

The S3 by default uses a little bit more stack so this probably why you dont see any issues on ESP32.

On master we already increased the default stack size for this example, i'll backport a similar update for v4.4

gazam2842, 2022-10-19:
> Hi @gazam2842, thanks for reporting this!
> 
> The S3 by default uses a little bit more stack so this probably why you dont see any issues on ESP32.
> 
> On master we already increased the default stack size for this example, i'll backport a similar update for v4.4

Thanks @ESP-Marius

closed 2022-10-19 by gazam2842