espressif/esp-idf #10009: Undefined reference error to some functions in LVGL library files

Vaishnavibhavaa, 2022-10-19:
Hi ,

Header and source files are included correctly in CMake and header files are directly included in corresponding files too .

Still facing this undefined reference error . Can anyone help me with this ?

```
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\whirpool.elf.rsp  -o whirpool.elf  && cd ."
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/lvgl/liblvgl.a(lv_hal_disp.c.obj):(.literal.lv_disp_drv_init+0x8): undefined reference to `lv_draw_sw_init_ctx'
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/lvgl/liblvgl.a(lv_hal_disp.c.obj):(.literal.lv_disp_drv_register+0x8): undefined reference to `lv_theme_default_is_inited'
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/lvgl/liblvgl.a(lv_hal_disp.c.obj):(.literal.lv_disp_drv_register+0xc): undefined reference to `lv_theme_default_init'
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/lvgl/liblvgl.a(lv_hal_disp.c.obj):(.literal.lv_disp_drv_register+0x10): undefined reference to `lv_theme_default_get'
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/lvgl/liblvgl.a(lv_hal_disp.c.obj): in function `lv_disp_drv_register':
c:\users\bvv1cob\desktop\8.2oct17\build/../components/lvgl/lvgl/src/hal/lv_hal_disp.c:187: undefined reference to `lv_theme_default_is_inited'
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:\users\bvv1cob\desktop\8.2oct17\build/../components/lvgl/lvgl/src/hal/lv_hal_disp.c:196: undefined reference to `lv_theme_default_init'
c:/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:\users\bvv1cob\desktop\8.2oct17\build/../components/lvgl/lvgl/src/hal/lv_hal_disp.c:196: undefined reference to `lv_theme_default_get'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
ninja failed with exit code 1

kayzen9, 2022-10-19:
Shud file issue at https://github.com/lvgl/lvgl/??

Alvin1Zhang, 2022-10-21:
Thanks for reporting, please help file issue at https://github.com/lvgl/lvgl/, feel free to reopen.

closed 2022-10-21 by Alvin1Zhang