espressif/esp-idf #10011: LCI configuration for FTM responder

yogiD, 2022-10-19:
### Is your feature request related to a problem?

Android API call return the ResponderLocation  as per IEEE P802.11-REVmc/D8.0 spec section 9.4.2.22, under Measurement Report Element.

Not able to find LCI implementation for FTM responder ( AP mode).

### Describe the solution you'd like.

Need to implement Measurement Report Element as per  IEEE P802.11-REVmc/D8.0 spec section 9.4.2.22 in FTM responder AP mode.

yogiD, 2023-02-22:
Any plan to implement or expose the api call so we can push it.

sarveshb14, 2023-02-23:
Hi @yogiD,

Yes, We are planning to add support for this.