espressif/esp-idf #10022: Low Throughput iper on ESP32-C3-DevKitC-02

SelcRGIO, 2022-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are trying to run a stress test to a router. To do so, we need to simulate multiple nodes that are connecting through WiFi. These nodes are ESP32 running iperf. Connected to the router using ethernet there is a laptop running Linux. In the laptop we run the iperf server  and in the ESP32-C3 nodes we run the iperf client from the esp-idf example.
Our problem is: when using a single ESP32 we can achieve speeds of up to 40 Mbps. But if we try to add more nodes, i.e., running 2, 3, 4 ESP32s in parallel, there seems to be a cap in our maximum speed, and in total we never achieve more than 40 Mbps. We can understand that a single ESP32 may not be able to get a higher throughput, but the combined throughput should get higher, given we're not close to the limit of Wi-Fi 4. What's preventing the combined throughput to grow? Is there anyway to fix it or improve it?
Thanks

MaxwellAlan, 2022-10-26:
Hi @SelcRGIO 

There are some related issues maybe you need to check
1. Test environment: in the shield box or in the open air environment?
2. The Router's info, such as bandwith(ht20 or ht40), brand and types, any other special configs etc.
3. ESP-IDF commit and ESP32 or ESP32C3 modules?
4. Test method: only test UDP both TX and RX, then provide the new test result.

SelcRGIO, 2022-11-04:
Hi @MaxwellAlan 
1. The environment is open air, but we have conducted test in different locations (checking with a WiFi analyzer that there wasn't to many networks/traffic interfering) with more or less the same results.
2. The router info is: RT-AX68U ASUS (configured only ht20)
3. Using esp-idf-v4.4.2 on a ESP32-C3-DevKitC-02 module
4. See attachments of results with a single device and 4 devices in parallel  
[image]
[image]

closed 2023-09-05 by Alvin1Zhang