espressif/esp-idf #10023: [BLE Mesh] Example proxy fails after reboot

JeromeLinkio, 2022-10-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After the provisionning (successful), if the ESP32 reboots, the device is not visible as a proxy.

This how I flash the example : 
`cd esp-idf-v4.4.2/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client/
/usr/bin/cmake -S. -Bbuild -G Ninja
/usr/bin/cmake --build build --target flash`

I use nRF Mesh on iOS.

Here are my steps, without reboot : 
- Provisionning
- Request Composition Data : timeout
- In Proxy : select proxy “ESP-BLE-MESH”
- Request Composition Data : OK
- Bind App Key : OK

Here are my steps, WITH reboot : 
- Provisionning
- Request Composition Data : timeout
- Reboot ESP32 (power on - off)
- Request Composition Data : timeout
- In Proxy : select proxy … the device doesn't appear : NOT OK

Notes : 
- in nRF Mesh on Android, the device is not even visible.. but this is another problem.
- in menuconfig : "Store BLE Mesh configuration persistently" is checked.

Am I missing something ? 
Thanks a lot for your help.

forx157, 2022-10-24:
hi, i have tested that, and it works well, could you provide me a log?

JeromeLinkio, 2022-10-24:
Hi, thanks for your interest.
Here is my log when rebooting the ESP32 : 

ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
[0;32mI (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 11:36:04[0m
[0;32mI (27) boot: chip revision: 3[0m
[0;32mI (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (38) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (42) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (47) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (51) boot: Enabling RNG early entropy source...[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (60) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (90) boot: End of partition table[0m
[0;32mI (94) boot_comm: chip revision: 3, min. application chip revision: 0[0m
[0;32mI (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=28fe0h (167904) map[0m
[0;32mI (171) esp_image: segment 1: paddr=00039008 vaddr=3ffbdb60 size=04f9ch ( 20380) load[0m
[0;32mI (179) esp_image: segment 2: paddr=0003dfac vaddr=40080000 size=0206ch (  8300) load[0m
[0;32mI (183) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9e710h (648976) map[0m
[0;32mI (420) esp_image: segment 4: paddr=000de738 vaddr=4008206c size=14f74h ( 85876) load[0m
[0;32mI (455) esp_image: segment 5: paddr=000f36b4 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (467) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (467) boot: Disabling RNG early entropy source...[0m
[0;32mI (479) cpu_start: Pro cpu up.[0m
[0;32mI (479) cpu_start: Starting app cpu, entry point is 0x400811a4[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (495) cpu_start: Pro cpu start user code[0m
[0;32mI (495) cpu_start: cpu freq: 160000000[0m
[0;32mI (495) cpu_start: Application information:[0m
[0;32mI (500) cpu_start: Project name:     onoff_client[0m
[0;32mI (505) cpu_start: App version:      v4.4.2-dirty[0m
[0;32mI (511) cpu_start: Compile time:     Oct 17 2022 11:35:58[0m
[0;32mI (517) cpu_start: ELF file SHA256:  0ed41160cc02a3e2...[0m
[0;32mI (523) cpu_start: ESP-IDF:          v4.4.2-dirty[0m
[0;32mI (528) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (535) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (541) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (547) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (554) heap_init: At 3FFCA040 len 00015FC0 (87 KiB): DRAM[0m
[0;32mI (560) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (566) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (573) heap_init: At 40096FE0 len 00009020 (36 KiB): IRAM[0m
[0;32mI (580) spi_flash: detected chip: gd[0m
[0;32mI (583) spi_flash: flash io: dio[0m
[0;33mW (587) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (601) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (611) EXAMPLE: Initializing...[0m
[0;32mI (611) gpio: GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (621) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (631) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
use esp timer !!!
[0;32mI (641) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (671) BTDM_INIT: BT controller compile version [bf9c077][0m
[0;32mI (671) system_api: Base MAC address is not set[0m
[0;32mI (681) system_api: read default base MAC address from EFUSE[0m
[0;32mI (681) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[0;32mI (1101) EXAMPLE_NVS: Open namespace done, name "mesh_example"[0m
[0;32mI (1181) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0[0m
[0;32mI (1181) EXAMPLE_NVS: Restore, key "onoff_client" not exists[0m
[0;32mI (1181) EXAMPLE: BLE Mesh Node initialized[0m
[0;32mI (1201) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0[0m

forx157, 2022-10-25:
hi, @JeromeLinkio ,That log shows the node didn't have provisioned, and you need to Provisionning first. And there will have the following statements in the log after provisioning & reboot:
```
I (599) EXAMPLE: net_idx: 0x0000, addr: 0x0003 (value maybe different)
I (599) EXAMPLE: flags: 0x00, iv_index: 0x00000000
```
and you can see that device in the "Select Proxy".
If there still have problems, please provide me the log after you reboot, and nRF Mesh's Screenshot。

JeromeLinkio, 2022-10-25:
Hello @forx157 ,
As I said, I did the provisionning first, and then the reboot.
But, after reading the code of the example, I saw that the function `mesh_example_info_store()` is not called after the provisionning, but after the binding of the app key to a model (which I wasn't doing).
Still, after reboot, the node is not provisionned (it's possible to provision it) even though it restores some data from NVS (`EXAMPLE_NVS: Restore, data: 00 00 00 00 00 00`) : 
```
[0;32mI (1101) EXAMPLE_NVS: Open namespace done, name "mesh_example"[0m
[0;32mI (1191) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0[0m
[0;32mI (1191) EXAMPLE_NVS: Restore, key "onoff_client", length 6[0m
[0;32mI (1191) EXAMPLE_NVS: Restore, data: 00 00 00 00 00 00 [0m
[0;32mI (1201) EXAMPLE: Restore, net_idx 0x0000, app_idx 0x0000, onoff 0, tid 0x00[0m
[0;32mI (1201) EXAMPLE: BLE Mesh Node initialized[0m
[0;32mI (1221) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0[0m
[0;32mI (539011) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT[0m
[0;33mW (539401) BLE_MESH: No Health Server context provided[0m
[0;33mW (541811) BLE_MESH: No Health Server context provided[0m
[0;32mI (541841) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT[0m
[0;32mI (541841) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT[0m
[0;32mI (541841) EXAMPLE: net_idx: 0x0000, addr: 0x034b[0m
[0;32mI (541841) EXAMPLE: flags: 0x00, iv_index: 0x00000000[0m
[0;33mW (541891) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13[0m
```

So, when you said you have tested that, what have you tested exactly, because without binding the app key to a model, the example shouldn't work with the steps I described ?

forx157, 2022-10-25:
hi, @JeromeLinkio, 
My test sequence is:
- provisioning
-  waitting Request Composition Data timeout
- reboot node
- select Proxy device, the device will be found

And the basic net info such as the net index,net key will be stored even if you didn't bind the appkey. you can find a function named `bt_mesh_provision` in the file `components/bt/esp_ble_mesh/mesh_core/main.c ` will be triggered after your node was provisioned, in this function will store some basic information of that net.
Detailed call chain is:
```c
bt_mesh_provision -> schedule_store - submit a work-> store_pending -> store_pending_keys & store_pending_net and so on.
```
After you reboot your device, a function ` bt_mesh_settings_init` whill be triggered by the function `bt_mesh_init`, and then the previous net info will be loaded.
Detailed call chain is:
```c
esp_ble_mesh_init -> bt_mesh_init -> bt_mesh_settings_init -> bt_mesh_settings_init_foreach -> ctx->settings_load() (Notice: ctx is settings_ctx[i] so settings_load is settings_core_load) -> settings_core_load -> settings[i].func (will be role_set, net_set, ...)
```
but all of this need you to open "Store BLE Mesh configuration persistently" in the menuconfig.
So, the device still will be in net after you reboot, and proxy service still will be found even if you have not bind app key.
Finally, if there still have any questions, you can change the log level from warning to info by the following path:`Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL ` and `Component config → ESP BLE Mesh Support → BLE Mesh NET BUF DEBUG LOG LEVEL` to get more infomation.

JeromeLinkio, 2022-10-25:
Thanks for your answer @forx157 

Actually, it works by adding (to `sdkconfig.defaults`) the line : 
```
CONFIG_BLE_MESH_SETTINGS=y
```

I guess I had a problem with the menuconfig, because the line "Store BLE Mesh configuration persistently" was checked..

closed 2022-10-25 by JeromeLinkio