espressif/esp-idf #10025: [ESP DS] [Digital Signature] esp_ds_sign() does not support Virtual Fuses ?❗❗ ESP_ERR_HW_CRYPTO_DS_HMAC_FAIL

chipweinberger, 2022-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### **Is there any way to test digital signature code without actually burning fuses?**

I'm trying to get ESP Digital Signature working, but `esp_ds_sign()` returns `ESP_ERR_HW_CRYPTO_DS_HMAC_FAIL`, likely due to `ets_hmac_calculate_downstream()` not supporting virtual fuses.

Is there anything like `esp_test_hmac_calculate_downstream(uint8_t* hmacKey256)` that lets you pass in the hmac key without actually burning the fuses?

The code is quite complicated (reproduced below).

**How can I test this code without buying 100 development boards?**

**generating Y,M,Rb,MPrime:**
```
def burnPrivateKey(st):

    # pem files
    p = getPaths(st)

    print("----------------------------------------------------------------------------")
    print("----------------------------------------------------------------------------")
    print("---------------     [Auth Esp] Burn Private Key      -----------------------")
    print("----------------------------------------------------------------------------")
    print("----------------------------------------------------------------------------")

    pemFile = open(p.devPrivate_Pem, "rb") 
    pemData = pemFile.read()
    pemFile.close()

    private_key = serialization.load_pem_private_key(pemData, password=None, backend=default_backend())

    if not isinstance(private_key, rsa.RSAPrivateKey):
        bigPrint.print_fail()
        print("expected PEM to be RSAPrivateKey")
        exit(2)

    priv_numbers = private_key.private_numbers()
    pub_numbers = private_key.public_key().public_numbers()

    Y = priv_numbers.d
    M = pub_numbers.n

    # this cyrptic code came from esspressif
    # /esp_secure_cert_mgr/blob/main/tools/esp_secure_cert/configure_ds.py
    key_size = private_key.key_size
    rr = 1 << (key_size * 2)
    rinv = rr % pub_numbers.n
    mprime = - rsa._modinv(M, 1 << 32)
    mprime &= 0xFFFFFFFF

    cmdJson = {}
    cmdJson['rsaY'] = hex(Y).replace("0x","")
    cmdJson['rsaM'] = hex(M).replace("0x","")
    cmdJson['rsaRb'] = hex(rinv).replace("0x","")
    cmdJson['rsaMPrime'] = hex(mprime).replace("0x","")

    # Serializing json
    jsonStr = json.dumps(cmdJson, indent=4).replace('\n','').replace(' ','')

    jResp = jflashConsole.invoke_json_cmd(st,"/api/auth/upload-esp-dev-private", jsonStr)

    print(jResp)

    j = '{"challenge256":"1234567890123456789012345678901234567890123456789012345678901234"}'
    jResp = jflashConsole.invoke_json_cmd(st,"/api/auth/esp-challenge", j)

    print(jResp)

```

**uploading private key:**
```
cJSON* pd_api_auth_upload_esp_dev_private(const pd_4096_bit_t* rsaY, 
                                          const pd_4096_bit_t* rsaM, 
                                          const pd_4096_bit_t* rsaRb, 
                                                      uint32_t rsaMPrime, 
                                                  pd_error_t* error)
{
    pd_err_unset(error);
    
    // allcations
    cJSON* jRoot = NULL;
    esp_ds_p_data_t* params = NULL;
    esp_ds_data_t* encrypted = NULL;

    {
    // set params
    params = (esp_ds_p_data_t*) calloc_safe2(MALLOC_CAP_DMA, sizeof(esp_ds_p_data_t), 1, HERE);
    params->length = ESP_DS_RSA_4096;
    memcpy(params->Y, rsaY->bytes, 4096/32);
    memcpy(params->M, rsaM->bytes, 4096/32);
    memcpy(params->Rb, rsaRb->bytes, 4096/32);
    params->M_prime = rsaMPrime;

    // allcate
    encrypted = (esp_ds_data_t*) calloc_safe2(MALLOC_CAP_DMA, sizeof(esp_ds_data_t), 1, HERE);

    // iv
    uint8_t iv[16];
    esp_fill_random(iv, 16);

    // hmac
    uint8_t hmac[32];
    esp_fill_random(hmac, 32);

    // aes encrypt RSA params with IV & HMAC keys
    esp_err_t err = esp_ds_encrypt_params(encrypted, iv, params, hmac);
    if (err != ESP_OK){
        pd_err_fail(error, TAG, "esp_ds_encrypt_params() failed. %s", esp_err_to_name(err));
        goto end;
    }

    // store encrypted RSA key into nvs
    pd_nvs_esp_ds_cipher_set(encrypted->c, ESP_DS_C_LEN);
    pd_nvs_esp_ds_iv_set(iv, ESP_DS_IV_LEN);

    // burn hmac key into fuses
    err = esp_efuse_write_key(EFUSE_BLK_KEY0, 
        ESP_EFUSE_KEY_PURPOSE_HMAC_DOWN_DIGITAL_SIGNATURE, hmac, 32);
    if (err != ESP_OK){
        pd_err_fail(error, TAG, "esp_efuse_write_key() failed. %s", esp_err_to_name(err));
        goto end;
    }

    jRoot = cJSON_CreateObject();
    cJSON_AddBoolToObject(jRoot, "burned", true);

    pd_err_success(error);
    }
end:
    free_safe(params);
    free_safe(encrypted);

    if (is_fail(*error)){
        cJSON_Delete(jRoot);
        return NULL;
    }

    return jRoot;
}
```

**challenge:**
```
cJSON* pd_api_auth_esp_challenge(const pd_256_bit_t* challenge, pd_error_t* error)
{
 pd_err_unset(error);

    // Allocations
    cJSON* jRoot = NULL;
    char* sigHex = NULL;
    uint8_t* cipher = NULL;
    uint8_t* iv = NULL;
    uint8_t* signature = NULL;

    {
    // get cipher
    uint32_t len = 0;
    uint8_t* cipher = pd_nvs_esp_ds_cipher_get(&len);
    if (cipher == NULL) {
        pd_err_fail(error, TAG, "cipher is not set");
        goto end;
    }

    if (len != ESP_DS_C_LEN) {
        pd_err_fail(error, TAG, "cipher len (%u) should be %u", len, ESP_DS_C_LEN);
        goto end;
    }

    // get iv
    len = 0;
    uint8_t* iv = pd_nvs_esp_ds_iv_get(&len);
    if (iv == NULL) {
        pd_err_fail(error, TAG, "iv is not set");
        goto end;
    }

    if (len != ESP_DS_IV_LEN) {
        pd_err_fail(error, TAG, "iv len (%u) should be %u", len, ESP_DS_IV_LEN);
        goto end;
    }

    // create ds data
    esp_ds_data_t data;
    data.rsa_length = ESP_DS_RSA_4096;
    memcpy(data.iv, iv, ESP_DS_IV_LEN);
    memcpy(data.c, cipher, ESP_DS_C_LEN);

    // signature
    uint32_t sigLen = (data.rsa_length+1) * 4;
    signature = (uint8_t*) calloc_safe(sigLen, 1, HERE);

    // ds sign
    esp_err_t err = esp_ds_sign(challenge->bytes, &data, HMAC_KEY0, signature);
    if (err != ESP_OK){
        pd_err_fail(error, TAG, "esp_ds_sign() failed. %s", esp_err_to_name(err));
        goto end;
    }

    // hex encode
    sigHex = hexEncode_alloc(signature, sigLen);

    jRoot = cJSON_CreateObject();
    cJSON_AddStringToObject(jRoot, "signature", sigHex);

    pd_err_success(error);
    }
end:
    free_safe(sigHex);
    free_safe(cipher);
    free_safe(iv);
    free_safe(signature);

    if (is_fail(*error)){
        cJSON_Delete(jRoot);
        return NULL;
    }

    return jRoot;
}
```

chipweinberger, 2022-10-22:
Each device has 5 Key slots, so... I can test 5 times per board?

chipweinberger, 2022-10-22:
And while on the topic, I cannot determine from the documentation which padding scheme the DS peripheral uses.

Does the DS peripheral's RSA implementation correspond to this python code?

```
    signature = private_key.sign(
        bytearray(32), # sign a message with all zeros
        padding.PSS(
            mgf=padding.MGF1(hashes.SHA256()),
            salt_length=padding.PSS.MAX_LENGTH
        ),
        utils.Prehashed(hashes.SHA256())
    )
```

Everywhere I've read suggests that not using padding is very insecure, even for signatures, but I don't see an RSA Padding scheme mentioned anywhere.

chipweinberger, 2022-10-22:
And while on the topic of RSA, which padding scheme the DS peripheral uses? PSS, PKCS1.5? Nothing? The docs don't mentions one, so I assume nothing.

chipweinberger, 2022-10-22:
@mahavirj 

Could you take a look at the code and see if there is anything obviously wrong? The RSA signatures I am generating on the device do not match the signature I'm generating in python.

I think my usage of `esp_ds_encrypt_params()` or  `esp_ds_sign()` might be wrong.

chipweinberger, 2022-10-23:
Closing this issue in favor of a new one.

https://github.com/espressif/esp-idf/issues/10028

closed 2022-10-23 by chipweinberger

chipweinberger, 2022-10-23:
And for the record, virtually fuses are not supported. 

They cause ESP_ERR_HW_CRYPTO_DS_HMAC_FAIL

KonstantinKondrashov, 2022-10-26:
Hi @chipweinberger!
Sorry, It is not possible to support the Virtual eFuses because the DS module uses hardware to get the result.

In case you use the Virtual eFuses with DS you will get ESP_ERR_HW_CRYPTO_DS_HMAC_FAIL because the DS hardware reads a given eFuse key block but it is empty, eFuses are stored in RAM, not in eFuse registers, hardware can not reach it. There is no way you need to use the real eFuse mode.

chipweinberger, 2022-10-27:
Understood.

it would be great if a test function simulated the DS hardware with similar API, using virtual fuses.

But that it just a request. 

With careful planning, I was able to get the DS hardware working with 5 attempts, and 5 burned keys.

mahavirj, 2022-10-27:
@chipweinberger 

> it would be great if a test function simulated the DS hardware with similar API, using virtual fuses.

What might help here is to have more targets (includes ESP32-S2/C2/S3) supported in our qemu fork [here](https://github.com/espressif/qemu], that way it would be possible to easily try out security features under emulator first and then move to the real physical target. Currently we have only ESP32 support in this but it does not feature DS peripheral. We will try to prioritize support for other targets here.

Virtual efuse feature is mostly for our internal testing purpose, but it definitely can not help in scenarios where peripheral is actually expecting credentials/data from actual efuse block/s. We will add note regarding this in our docs. Thank you for the feedback.

chipweinberger, 2022-10-27:
Yes that would be useful. Not all users will be able to use (easily) the simulator due to other hardware constraints.

What I am proposing it to completely *reimplement* the DS hardware in *software*. It might be more trouble than it is worth, but the software could read the virtual fuses. 

I'm not an expert, the the RSA algorithm is not *that* complex. The software might be reasonable to maintain.