espressif/esp-idf #10026: New project crashes on Adafruit Feather ESP32-S3 No PSRAM (v4.4.2 and v5.0)

DamarXCV, 2022-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2 and v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Building a new project, for example [blink](https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink) or a new PlatformIO project (using espidf-framework) and uploading it should not result in permanent crashing the program.

### What is the actual behavior?

After uploading a binary file to my adafruit_feather_esp32s3_nopsram, which is build with the espidf framework, via PlatformIO or ESP-IDF Explorer the ESP32 crashes permanently.

### Steps to reproduce.

### Variant 1 using PlatformIO
1. Create a new project using PlatformIO (Board is [Adafruit Feather ESP32-S3 No PSRAM](https://docs.platformio.org/en/latest/boards/espressif32/adafruit_feather_esp32s3_nopsram.html#id1)[¶](https://docs.platformio.org/en/latest/boards/espressif32/adafruit_feather_esp32s3_nopsram.html#adafruit-feather-esp32-s3-no-psram); framework is [espidf](https://docs.platformio.org/en/latest/frameworks/espidf.html#espressif-iot-development-framework))
2. Build and upload via esptool
3. Adafruit Feather ESP32-S3 No PSRAM crashes

### Variant 2 using ESP-IDF Explorer build and PlatformIO upload
1. Create a new project using ESP-IDF Explorer (Board is ESP32-S3 chip)
2. Build it and copy the binary into a PlatformIO project (also rename it appropriately)
3. upload it via pio cli with `pio run -t nobuild -t upload`
4. Adafruit Feather ESP32-S3 No PSRAM crashes

### Build or installation Logs.

```plain
### Variant 1
Executing task: C:\Users\*****\.platformio\penv\Scripts\platformio.exe run --target upload --environment adafruit_feather_esp32s3_nopsram 

Processing adafruit_feather_esp32s3_nopsram (platform: espressif32; board: adafruit_feather_esp32s3_nopsram; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32s3_nopsram.html
PLATFORM: Espressif 32 (5.2.0) > Adafruit Feather ESP32-S3 No PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Checking size .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.4% (used 20956 bytes from 327680 bytes)
Flash: [==        ]  19.2% (used 251857 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, 
tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM4
Forcing reset using 1200bps open/close on port COM4
Waiting for the new upload port...
Uploading .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v4.2.1
Serial port COM4
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:5c:9f:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x002d0000 to 0x002f8fff...
Flash will be erased from 0x00010000 to 0x00047fff...
Compressed 15040 bytes to 10362...
Writing at 0x00000000... (100 %)
Wrote 15040 bytes (10362 compressed) at 0x00000000 in 0.3 seconds (effective 347.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 223.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 438.7 kbit/s)...
Hash of data verified.
Compressed 165536 bytes to 106746...
Writing at 0x002d0000... (14 %)
Writing at 0x002d8667... (28 %)
Writing at 0x002ddbce... (42 %)
Writing at 0x002e459a... (57 %)
Writing at 0x002ea572... (71 %)
Writing at 0x002efeab... (85 %)
Writing at 0x002f52c0... (100 %)
Wrote 165536 bytes (106746 compressed) at 0x002d0000 in 2.0 seconds (effective 668.6 kbit/s)...
Hash of data verified.
Compressed 227984 bytes to 119474...
Writing at 0x00010000... (12 %)
Writing at 0x0001d17d... (25 %)
Writing at 0x00022e50... (37 %)
Writing at 0x000290c3... (50 %)
Writing at 0x0002efec... (62 %)
Writing at 0x00037e48... (75 %)
Writing at 0x0003fd6f... (87 %)
Writing at 0x00045a39... (100 %)
Wrote 227984 bytes (119474 compressed) at 0x00010000 in 2.5 seconds (effective 733.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
================================================================================ [SUCCESS] Took 19.95 seconds ================================================================================

### Variant 2
pio  run -t nobuild -t upload
Processing adafruit_feather_esp32s3_nopsram (platform: espressif32; board: adafruit_feather_esp32s3_nopsram; framework: arduino)
----------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM4
Forcing reset using 1200bps open/close on port COM4
Waiting for the new upload port...
Uploading .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v4.2.1
Serial port COM4
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:5c:9f:d4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00010000 to 0x00047fff...
Compressed 227984 bytes to 119474...
Writing at 0x00010000... (12 %)
Writing at 0x0001d17d... (25 %)
Writing at 0x00022e50... (37 %)
Writing at 0x000290c3... (50 %)
Writing at 0x0002efec... (62 %)
Writing at 0x00037e48... (75 %)
Writing at 0x0003fd6f... (87 %)
Writing at 0x00045a39... (100 %)
Wrote 227984 bytes (119474 compressed) at 0x00010000 in 2.5 seconds (effective 732.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=========== [SUCCESS] Took 12.45 seconds ===========

### Serial log of esp

21:12:26.685 -> ESP-ROM:esp32s3-20210327
21:12:26.685 -> ESP-ROM:esp32s3-20210327
21:12:26.685 -> Build:Mar 27 2021
21:12:26.685 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
21:12:26.685 -> Saved PC:0x400454d5
21:12:26.685 -> SPIWP:0xee
21:12:26.685 -> mode:QIO, clock div:1
21:12:26.685 -> load:0x3fce3808,len:0x164c
21:12:26.685 -> ets_loader.c 78
```

### More Information.

- Tried variant 1 on 3 devices, 2 had a fresh vs code installation.
- Tested with 2 feather esp32-s3 boards.
- Building a project in PlatformIO with the arduino-framework works, with the espidf its doesn't.

DamarXCV, 2022-10-22:
I guess i found the problem!
I have 2 USB-devices:

- COM3 with Hardware-ID `USB\VID_239A&PID_8113&REV_0100&MI_00` and `USB\VID_239A&PID_8113&MI_00`
- COM4 with Hardware-ID  `USB\VID_303A&PID_1001&REV_0101&MI_00` and `USB\VID_303A&PID_1001&MI_00
`
Using PlatformIO, uploading a project with the **arduino-framework**, PlatformOI disconnects COM3 or COM4 and reconnects it as COM4 uploads the binary. After pressing the hardware reset-button on the board the board reboots and reconnects as COM3.

Using PlatformIO, uploading a project with the **espidf-framework**, PlatformOI disconnects COM3 or COM4 and reconnects it as COM4 uploads the binary. But now the board continuously disconnects and reconnects (or reboots?) as COM4. Even if i press the  hardware reset-button.

DamarXCV, 2022-10-22:
I have a workaround for now...

I compared the Upload comand of both frameworks and merged them.

I take the
- bootloader.bin (".pio\build\adafruit_feather_esp32s3_nopsram\bootloader.bin")

from a arduino-framework project
- boot_app0.bin ("C:\Users\{USERNAME}\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin")
- tinyuf2.bin ("C:\Users\{USERNAME}\.platformio\packages\framework-arduinoespressif32\variants\adafruit_feather_esp32s3_nopsram\tinyuf2.bin")

from the arduino-framework and the
- partitions.bin (.pio\build\adafruit_feather_esp32s3_nopsram\partitions.bin)
- firmware.bin (.pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin)

from the espidf-framework project.

Resulting in: `C:\Users\{USERNAME}\.platformio\penv\Scripts\python.exe C:\Users\{USERNAME}\.platformio\packages\tool-esptoolpy\esptool.py --chip esp32s3 --port "COM4" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0000 custom_boot\bootloader.bin 0x8000 .pio\build\adafruit_feather_esp32s3_nopsram\partitions.bin 0xe000 custom_boot\boot_app0.bin 0x2d0000 custom_boot\tinyuf2.bin 0x10000 .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin`

ESP-Marius, 2022-11-03:
Hi @DamarXCV ,

I think you might be better of asking this question at https://github.com/platformio/ as far as I can see it doesnt seem directly related to esp-idf and unfortunately my knowledge of platform IO is limited :smile: 

Just from your log my initial suspicion is that fails due to trying to boot with QIO as the flash mode (`mode:QIO, clock div:1`) when building/flashing with platformIO's esp-idf plugin, while it probably boots with `DIO` if using Arduino. Not all boards support `QIO`, more about flash modes can be read [here](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/spi-flash-modes.html).

Closing this one for now, but feel free to re-open if the issue turns out to be `esp-idf` related after all.

closed 2022-11-03 by ESP-Marius

DamarXCV, 2022-11-03:
Hi @ESP-Marius,

i decided to open the issue here because the two files required to make the esp working are located in the arduino-framework repo and not in the platformio repo. My guess is that those files are missing in esp-idf to make it work.

My default upload script with espidf looks like this: `"C:\Users\{USERNAME}\.platformio\penv\Scripts\python.exe" "C:\Users\{USERNAME}\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32s3 --port "COM4" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 {project-path}\.pio\build\adafruit_feather_esp32s3_nopsram\bootloader.bin 0x8000 {project-path}\.pio\build\adafruit_feather_esp32s3_nopsram\partitions.bin 0x10000 .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin`.
So it should also use `dio` with esp-idf.

ESP-Marius, 2022-11-04:
`--flash_mode dio` here only decide the mode you flash the binary with. When compiling the binary the mode which to boot the bootloader with is written into the binary image header, so probably when taking the `bootloader.bin` from Arduino framework this header is `DIO`, while from IDF it is `QIO`. 

But both of these are configs decided when compiling the framework, and set by platform IOs build script/configs, so I still think you'll need their assistance.

chegewara, 2022-11-04:
Its not exactly the answer, but it should help to see how important is to select correct params. Also it may give you a hint if you need to open issue in platformio.

https://github.com/platformio/platform-espressif32/issues/837

Like @ESP-Marius said, the problem is not related to arduino or esp-idf, because platformio is using own clone/copy of esp-idf, i think.