espressif/esp-idf #10028: [Documentation] [Question] Cannot figure out how to use esp_ds_sign() and esp_ds_encrypt_params()

chipweinberger, 2022-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been banging my head for 2 days trying to get esp_ds_sign() to work. 

I've read the docs, and the technical pdf, but there is not enough information about what `esp_ds_sign()` actually does, and the format that `esp_ds_encrypt_params()` expects.

 The big questions:

1. what byte order does `esp_ds_encrypt_params()` expect are the arguments to be in? Little endian U32's? Big Endian U32's? A raw byte array from big to little bytes?  A raw byte array from little to big bytes? 
2. does `esp_ds_sign()` expect the `message` to already be SHA256 hashed? or does it perform the SHA256 hash itself?
3. does `esp_ds_sign()` add any RSA padding? Such as PKCS1v15? Or Are we expected to add it ourself to the message before we sign?

I've taken my best guess at everything, and come up with this code, but the signatures still do not match!

```
 === Invalid Signature ===
signature: 3e98886589402cebd.....
signature expected: a4b5bdf2388336e400.....
```

# Generating Y,M,Rb,MPrime & Burning & Challenging: 
```
def burnPrivateKey(st):

    p = getPaths(st)

    print("----------------------------------------------------------------------------")
    print("----------------------------------------------------------------------------")
    print("---------------     [Auth Esp] Burn Private Key      -----------------------")
    print("----------------------------------------------------------------------------")
    print("----------------------------------------------------------------------------")

    pemFile = open(p.devPrivate_Pem, "rb") 
    pemData = pemFile.read()
    pemFile.close()

    private_key = serialization.load_pem_private_key(pemData, password=None, backend=default_backend())

    if not isinstance(private_key, rsa.RSAPrivateKey):
        bigPrint.print_fail()
        print("expected PEM to be RSAPrivateKey")
        exit(2)

    print("signing...\n")

    priv_numbers = private_key.private_numbers()
    pub_numbers = private_key.public_key().public_numbers()

    Y = priv_numbers.d # RSA Exponent
    M = pub_numbers.n  # RSA modulus

    # this cryptic code came from esspressif
    # /esp_secure_cert_mgr/blob/main/tools/esp_secure_cert/configure_ds.py
    # The ESP RSA accelerator is based on Montgomery multiplication.
    # Therefore, aside from the X, Y , and M arguments, 
    # two additional ones are needed — rinv and mprime
    # Which we need to calculate in advance
    key_size = private_key.key_size

    # calculate rinv == Rb
    rr = 1 << (key_size * 2)
    rinv = rr % pub_numbers.n # RSA r inverse operand

    # calculate MPrime
[... 418 lines cut ...]
    ds_r = esp_ds_start_sign((const void *)signature, s_ds_data, hmacBlock, &esp_ds_ctx);
    if (ds_r != ESP_OK) {
        ESP_LOGE(TAG, "Error in esp_ds_start_sign, returned %d ", ds_r);
        free_safe(signature);
        return -1;
    }

    ds_r = esp_ds_finish_sign((void *)signature, esp_ds_ctx);
    if (ds_r != ESP_OK) {
        ESP_LOGE(TAG, "Error in esp_ds_finish sign, returned %d ", ds_r);
        free_safe(signature);
        return -1;
    }

    printf("done\n");

    for (unsigned int i = 0; i < (s_ds_data->rsa_length + 1); i++) {
        ((uint32_t *)sig)[i] = SWAP_INT32(((uint32_t *)signature)[(s_ds_data->rsa_length + 1) - (i + 1)]);
    }

    free_safe(signature);
    return 0;
}

```

chipweinberger, 2022-10-23:
Looking at the docs more closely, **SHA256** is *not applied* in `esp_ds_sign()`.

And `esp_ds_sign()` expects 'message' to be in *uint4096_t* little-endian format.

So, this is the void* message I am giving `esp_ds_sign()`, with PKCS1v15 padding.

          > input message:
          > 0bb90fb25b098113078a0ddac3ad6d3297bef630347d6c2d5fddcbd7d89e43512004000501020403650148866009060d30313000ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff0100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

and I've changed the python to not apply SHA256 either `utils.Prehashed(hashes.SHA256())`

```
    challengeStr  = "51439ed8d7cbdd5f2d6c7d3430f6be97326dadc3da0d8a071381095bb20fb90b"
    challengeStr  += "0000000000000000000000000000000000000000000000000000000000000000"
    challengeStr  +=  "000...
    .... repeat zeros until 4096 bit number....

    # expected signature
    sigExpected_PKCS1v15 = private_key.sign(
        bytes.fromhex(challengeStr[0:64]),
        padding.PKCS1v15(),
        utils.Prehashed(hashes.SHA256())
        #hashes.SHA256()
    )
```

Still, the signatures return completely different results....starting to run out of ideas.

chipweinberger, 2022-10-23:
Another mistake I think I'm making....`esp_ds_encrypt_params()` should probably be set up using uint4096_t little endian as well. 

```
    // convert from big-endian to native-endian
    for(int i = 0; i < 4096/32; i++){
        params->Y[i] = SWAP_INT32(((uint32_t *)rsaY->bytes)[(4096/32) - (i + 1)]);
        params->M[i] = SWAP_INT32(((uint32_t *)rsaM->bytes)[(4096/32) - (i + 1)]);
        params->Rb[i] = SWAP_INT32(((uint32_t *)rsaRb->bytes)[(4096/32) - (i + 1)]);
    }
    params->M_prime = rsaMPrime;
```

chipweinberger, 2022-10-23:
Wow!!! That was it!!!! It works!!!! 

**PLEASE** add more documentation to `esp_ds_sign()` and `esp_ds_encrypt_params()`.

Specify that they all use `uint4096_t` little endian numbers. i.e. for Y,M,Rb, the void* message.

**Specify that `esp_ds_sign()` does *not* apply SHA256 to the message.** It seems stupid, but it wasn't obvious to me. Lots of other RSA libraries, such as python's crypto library *do* automatically apply SHA256 to the message. And SHA256 is talked about a lot in the documentation, so it would be easy to assume it might be applied.

**Specify that RSA padding, such as PKCS, etc, is not applied unless you apply it yourself to the message**. It's better to just be explicit! This shit is way to complicated to figure out. There are a ton of variables, and theres no way to tell what went wrong when it fails.

chipweinberger, 2022-10-23:
I left feedback on the documentation page.

AdityaHPatwardhan, 2022-10-27:
Hi @chipweinberger  Thank you for raising the issue. Looks like you have figured out the stuff.
I will update the documentation as well for making it easier for other users.
Thanks for your suggestions.

closed 2022-10-29 by chipweinberger