espressif/esp-idf #10029: assert failed: sys_timeout IDF

devtn97, 2022-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-WROOM-Kit

### Power Supply used.

USB

### What is the expected behavior?

Pairing success

### What is the actual behavior?

None

### Steps to reproduce.

- Build step to step : https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html
- When I use Chiptool connect with matter device 
- Scan QR 
- Input SSID and Password 
- Chiptool connected BLE and start  pairing...

### Debug Logs.

```plain
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:14788
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (29) boot: compile time 19:20:44
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
[... 178 lines cut ...]
0x40081cfa: panic_abort at /home/trungnghia/esp-idf/components/esp_system/panic.c:402

0x40090c45: esp_system_abort at /home/trungnghia/esp-idf/components/esp_system/esp_system.c:128

0x40096ee1: __assert_func at /home/trungnghia/esp-idf/components/newlib/assert.c:85

0x4016641c: sys_timeout at /home/trungnghia/esp-idf/components/lwip/lwip/src/core/timeouts.c:316
 (inlined by) sys_timeout at /home/trungnghia/esp-idf/components/lwip/lwip/src/core/timeouts.c:309

0x401548fd: esp_route_table_add_route_entry at /home/trungnghia/esp-matter/components/route_hook/src/esp_route_table.c:81

0x40154692: ra_recv_handler at /home/trungnghia/esp-matter/components/route_hook/src/esp_route_hook.c:103
 (inlined by) icmp6_raw_recv_handler at /home/trungnghia/esp-matter/components/route_hook/src/esp_route_hook.c:147

0x40161ff3: raw_input at /home/trungnghia/esp-idf/components/lwip/lwip/src/core/raw.c:178

0x4016a8d0: ip6_input at /home/trungnghia/esp-idf/components/lwip/lwip/src/core/ipv6/ip6.c:1063

0x4016d755: ethernet_input at /home/trungnghia/esp-idf/components/lwip/lwip/src/netif/ethernet.c:229

0x401605a9: tcpip_thread_handle_msg at /home/trungnghia/esp-idf/components/lwip/lwip/src/api/tcpip.c:180
 (inlined by) tcpip_thread at /home/trungnghia/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

0x40093b35: vPortTaskWrapper at /home/trungnghia/esp-idf/components/freertos/port/xtensa/port.c:131
```

devtn97, 2022-10-27:
I change my router => fixed

devtn97, 2022-10-27:
Fixed

closed 2022-10-27 by devtn97

HomeACcessoryKid, 2023-02-12:
would you spare some details on which router was changed to which
And maybe a hint of what you think made the difference??
TIA, HacK

HomeACcessoryKid, 2023-02-13:
I have just found the real reason.
see [Incorrect call to sys_timeout for real-life IPv6 lifetime of 2 weeks #217](https://github.com/espressif/esp-matter/issues/217)
In your case the lifetime is 30 days...
The good news is that it is possible to fix the code in case you don't want to change your router ;-)

devtn97, 2023-02-17:
@HomeACcessoryKid  When you run GHA example => router must support and router standand not map IP address. If lauch with hub router it no problem

HomeACcessoryKid, 2023-02-17:
The CHIP dev team has just applied the FIX needed to solve it. see [#217](https://github.com/espressif/esp-matter/issues/217) for a link to their fix.
Changing the router is merely avoiding the problem.