espressif/esp-idf #10030: How to close Blufi when I don't need Blufi?

casitelwithdean, 2022-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,recently i am using Blufi to config my wifi,and  when I finish configing, I want to close the Blufi to save power,but I don't know what APIs  I should use?
Thanks!

rahult-github, 2022-10-25:
Hi @casitelwithdean , 

Recently , a [commit](https://github.com/espressif/esp-idf/commit/aa03ffdc1beb1b650453f3adfe4a89aab2e24c82) was made to add an API to stop Bluetooth/Blufi.  Can you please refer the [esp_blufi_host_deinit](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/blufi/main/blufi_init.c#L195) API  ? 

Thanks,
 Rahul

casitelwithdean, 2022-10-25:
@rahult-github  Hi,this solved my problem ,i am really appreciated.Thanks very much!

closed 2022-10-25 by rahult-github