espressif/esp-idf #10033: esp_wifi_set_protocol can only be set to WIFI_PROTOCOL_11B

podaen, 2022-10-24:
### Board

ESP32

### Device Description

DevKit v3

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

1.8.19

### Operating System

Windows 7

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I discovered that the protocol can only be set  to` const uint8_t protocol = WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G || WIFI_PROTOCOL_11N;` OR `const uint8_t protocol = WIFI_PROTOCOL_11B` OR `const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;` anything else fails. 

It doesn't look the esp32 supports anything else than the very old protocol that only gives me a connection speed of 10Mbs. Any explanation on that?

### Sketch

```cpp
#include "esp_wifi.h"
#include "esp_log.h"

esp_err_t err;

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	err = esp_wifi_init(&cfg);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set wifi power mode!");
		return false;
	}

	err = esp_wifi_set_mode(WIFI_MODE_STA);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set sta!");
		return false;
	}

	//const uint8_t protocol = WIFI_PROTOCOL_11G;
	//const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;
	const uint8_t protocol = WIFI_PROTOCOL_11B;

	err = esp_wifi_set_protocol(WIFI_IF_STA, protocol);
	if (err != ESP_OK) {
		//258 = ESP_ERR_INVALID_ARG
		Serial.println(err);
		Serial.println("Could not set protocol!");
		return false;
	}

	uint8_t getprotocol;
	//esp_err_t err;
	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
	if (err != ESP_OK) {
		Serial.println("Could not get protocol!");
		//log_e("Could not get protocol! %d", err);
		return false;
	}
	if (getprotocol & WIFI_PROTOCOL_11N) {
		Serial.println("WiFi_Protocol_11n");
	}
	if (getprotocol & WIFI_PROTOCOL_11G) {
		Serial.println("WiFi_Protocol_11g");
	}
	if (getprotocol & WIFI_PROTOCOL_11B) {//worst, but dificuld to set else
		Serial.println("WiFi_Protocol_11b");
	}
```

### Debug Message

```plain
No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zhangyanjiaoesp, 2022-10-24:
@podaen  Does this can be closed also？Is same to #9949 ?

podaen, 2022-10-24:
No, Because the protocol is not been set by ESP Arduino and therefore not showed in the log. Still it is some important information about the network.

podaen, 2022-10-27:
@VojtaBartoska Do you agrea?

zhangyanjiaoesp, 2022-10-31:
@VojtechBartoska I have checked the Arduino project, I didn't find a interface for user to call `esp_wifi_set_protocol()` to set the 11B/G/N phy mode, maybe need to add one.
@podaen If the user doesn't change the phy mode, the default phy mode is `11bgn`,  the log you provided doesn't include the phy mode info.

zhangyanjiaoesp, 2022-11-02:
> To set the n protocol we need to set bgn and because it has been set after the start of the WiFi I think it can change over time depending on the connectivity of the device(s).

Yes，you need to set bgn for 11n protocol. And the STA will change the negotiated protocol according to the connected AP. Take a look at : https://github.com/espressif/esp-idf/issues/9949#issuecomment-1288513162

> Q: The bandwide is restored from nvs, so will it stay WIFI_BW_HT20 when it had some power issues at startup?

Yes

podaen, 2022-11-03:
I have implement [WiFiProv ](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFiProv/examples/WiFiProv)and it connects eminently, because my credentials where allready been flashed in to the nvs. Now I need to entirely `nvs_flash_erase()` to see how the provisioning works. It would be useful if there was an `esp_err_t esp_wifi_erase(void)` in the `esp_wifi.h` like the [restore](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_restorev) where the credentials are located.

podaen, 2022-11-07:
Can find WiFi multi in IDF...  `esp_wifi.h` api's can not be used as pointer because of the storage in the nvs. Would in not be better to adapted the storage of the wifi settings to make it more suitable for other uses like get, erase and multi wifi.

Q: For what is the manager.h for wifi_provisioning be used for? Can't find any documentation about that.

podaen, 2022-11-07:
I found it. It was in the same documentation of provisioning

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html#_CPPv418wifi_prov_mgr_init22wifi_prov_mgr_config_t

VojtechBartoska, 2022-11-07:
Hi @podaen, excuse delayed reply, I was on vacation.

Is there anything else what is not clear?

podaen, 2022-11-07:
This topic seems finished.

closed 2022-11-07 by podaen