espressif/esp-idf #10034: esp-idf/components/mbedtls/mbedtls/include' is not a  directory.

YitianDevelop, 2022-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 release

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

i just run idf.py set-target esp32c3  in hello_world dir

### What is the actual behavior?

even hello_world can't build successfully .

### Steps to reproduce.

 idf.py set-target esp32c3 

### Build or installation Logs.

```plain
CMake Error at /root/esp32c3/esp-idf/tools/cmake/component.cmake:312 (message):
  Include directory
  '/root/esp32c3/esp-idf/components/mbedtls/mbedtls/include' is not a
  directory.
Call Stack (most recent call first):
  /root/esp32c3/esp-idf/tools/cmake/component.cmake:486 (__component_add_include_dirs)
  /root/esp32c3/esp-idf/components/mbedtls/CMakeLists.txt:18 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/root/esp32c3/hello_world/build/CMakeFiles/CMakeOutput.log".
```

### More Information.

i failed to build on ArchLinux but the same problem also occurred in Ubuntu . 
i have run git submodule   --init --recursive   , and all repository are up to date

ESP-Marius, 2022-10-24:
I'd recommend doing `git submodule update --init --recursive` again to make sure the submodules are update. Because this error is typically what you see when the submodules havent been pulled.

What is the content of `components/mbedtls/mbedtls`? If this folder doesnt exist/is empty then the submodule is indeed missing.

YitianDevelop, 2022-10-24:
clone

closed 2022-10-24 by YitianDevelop