espressif/esp-idf #10035: JTAG Debugging on ESP32-S2 using ESP-Prog: Timed out when waiting for target halted

larryqiann, 2022-10-24:
Hi,

I am attempting to follow the debugging tutorial on the Espressif web page. I am using an ESP32-S2 (LOLIN S2 mini) and the ESP-PROG JTAG board. I have soldered some thin copper wires to the four JTAG pins on the chip, and connected to the adapter. However, I am experiencing some problems with OpenOCD. It seems like the CPU is detected correctly but GDB cannot be started. The following is what is output: I am on Apple M1.

`Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
[OpenOCD]
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
[OpenOCD]
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : esp32s2: Debug controller was reset.
[OpenOCD]
Info : esp32s2: Core was reset.
[OpenOCD]
Info : starting gdb server for esp32s2 on 3333
[OpenOCD]
Info : Listening on port 3333 for gdb connections
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333
[OpenOCD]
Info : Halt timed out, wake up GDB.
[OpenOCD]
❌ Error: timed out while waiting for target halted
Error executing event gdb-attach on target esp32s2:

Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected`

chipweinberger, 2022-10-24:
There's much more useful documentation in the espressif vs code repo

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

erhankur, 2025-03-12:
"❌ Error: timed out while waiting for target halted"  usually seen with a bad low quality USB cable or poor connections to the JTAG pins. Let us know if you still need help to start debugging.

closed 2025-03-12 by erhankur