espressif/esp-idf #10039: PCNT std::atomic_exchange like function (read current counter value and reset to 0 atomically)

imwhocodes, 2022-10-24:
### Is your feature request related to a problem?

Given the available function on esp-idf it is only possible to read the counter and reset using two separate functions

This imply that while doing something like this you can loose some pulse in a race condition like:

```cpp
pcnt_unit_handle_t pcnt_unit = NULL;
// Init stuff

long int total_pulse = 0;

while(true){
   int pulse_count = 0;
   
   ESP_ERROR_CHECK(pcnt_unit_get_count(pcnt_unit, &pulse_count));
   
   // !!! NEW PULSE HAPPEN HERE, ACTUAL VALUE LOST FOREVER AT CLEAR!!!
   
   ESP_ERROR_CHECK(pcnt_unit_clear_count(pcnt_unit));
   
   total_pulse += pulse_count;
   
   // Do other stuff
}
```

### Describe the solution you'd like.

It would be very helpfull to have a function that read and reset the counter atomically, something like:

```cpp
esp_err_t pcnt_unit_get_reset_count(pcnt_unit_handle_t unit, int *value)
```

Such function should atomically retrive the counter value and set it to 0
Something like https://en.cppreference.com/w/cpp/atomic/atomic_exchange

If at return of such function the value of the counter already increased it is not a problem, but it is important that the load and store of the counter value happen atomically

Is this possible?

suda-morris, 2022-10-25:
Hi @imwhocodes this is a good question, but I'm afraid the hardware doesn't have the ability for an "atomic read and clear" operation. We still have to do them in two steps, even if we write them in a signal function. 

But we will provide this feature request to our digital design team, and let you know when there's progress.

imwhocodes, 2022-10-25:
Hi

My primary concern was that on [esp-idf v5 documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/pcnt.html) say:

> The returned count value is a signed integer, where the sign can be used to reflect the direction. The internal counter will overflow when it reaches high or low limit, but this function doesn’t compensate for that loss.

And for `int32_t` overflow it means that il will go from max positive value to minimum negative value (actually c++ definition for overflow is undefined behaviour), so i wanted a way to not have this overflow behaviour (like keep reseting it to zero, see example on first message)

But on [esp-idf v4.4.2 documentation](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/peripherals/pcnt.html) there is no mention of that overflow and counter value is returned as a `int16_t`

Also when checking official [esp32 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) both for functional diagram and text explanation it say that when max (or minimum) value is reached counter reset to 0

> The pulse counters have five watchpoints that share one interrupt. Interrupt generation can be enabled or disabled
> for each individual watchpoint. The watchpoints are:
> • Maximum count value: Triggered when PULSE_CNT >= PCNT_CNT_H_LIM_Un. Additionally, this will reset
> the counter to 0. PCNT_CNT_H_LIM_Un should be a positive number.
> • Minimum count value: Triggered when PULSE_CNT <= PCNT_CNT_L_LIM_Un. Additionally, this will reset

[image]

And `Register 17.4. PCNT_Un_CNT_REG (n: 0­7) (0x28+0x0C*n)` (the register with the counter value) is 32bit wide (but with 16 bit reserved)

So now my question become what is the real behaviour of the counter? how it act on max or min value? and what is the definition of overflow in this context?

Thank you,
Luca

suda-morris, 2022-10-26:
Good question @imwhocodes Sorry for the misleading in our API doc. Your understanding in the peripheral is correct. I can confirm the "overflow" here means "reset to zero". We should update the doc!

imwhocodes, 2022-10-26:
Thank you

So you confirm that the behaviour (going back to `0`) is the same on both esp-idf versions:
- V5.0 -> returning `int32_t`
- V4.x.x -> returning `int16_t`

closed 2022-10-27 by espressif-bot