espressif/esp-idf #10041: [ESP-MESH] MESH_EVENT_TODS_REACHABLE is always zero

KonssnoK, 2022-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-381-gc97755c9ae5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

From documentation

> toDS state, devices shall check this state firstly before trying to send packets to external IP network. This state indicates right now whether the root is capable of sending packets out. If not, devices had better to wait until this state changes to be MESH_TODS_REACHABLE. 

https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32s3/api-reference/network/esp-wifi-mesh.html

```
/**
 * @brief The reachability of the root to a DS (distribute system)
 */
typedef enum {
    MESH_TODS_UNREACHABLE,    /**< the root isn't able to access external IP network */
    MESH_TODS_REACHABLE,      /**< the root is able to access external IP network */
} mesh_event_toDS_state_t;
```

I expect the event MESH_EVENT_TODS_REACHABLE to be generated each time the root loses/gains the ability to reach the external network (Internet).
I don't know how this is checked internally, since having an IP is not enough to assess access to an external network.

Moreover, 
i expect MESH_EVENT_TODS_REACHABLE  to be propagated to the whole reachable network, so that nodes can interrupt sending data to the root.

### What is the actual behavior?

MESH_EVENT_TODS_REACHABLE is called once on connection with Reachable = 0.
```
I (13313) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (13313) wifi:BcnInt:102400, DTIM:2
I (13323) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (13343) mesh: <nvs>write layer:1
I (13343) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (13343) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:0-->1, parent:d6:92:7d:c3:02:1e<ROOT>, ID:77:77:77:77:77:76
I (13343) mesh: <flush>root
I (13353) mesh_netif: Already wifi station, no need to do anything
I (13363) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
W (13373) mesh_main: <MESH_EVENT_TODS_REACHABLE>reachable:0
I (13373) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (13383) mesh_main: <MESH_EVENT_ROOT_ADDRESS>root address:7c:df:a1:e0:74:41
I (14563) esp_netif_handlers: sta ip: 192.168.43.144, mask: 255.255.255.0, gw: 192.168.43.23
I (14563) mesh_main: <IP_EVENT_STA_GOT_IP>IP:192.168.43.144
I (14563) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14563) mesh_main: Tried to publish layer:1 IP:192.168.43.144
I (14563) mesh_mqtt: Other event id:7
W (16843) wifi:<ba-add>idx:0 (ifx:0, d6:92:7d:c3:02:1e), tid:0, ssn:3, winSize:64
E (26853) esp-tls: [sock=54] select() timeout
E (26853) TRANSPORT_BASE: Failed to open a new connection: 32774
E (26853) MQTT_CLIENT: Error transport connect
I (26853) mesh_mqtt: MQTT_EVENT_ERROR

```

### Steps to reproduce.

- take code from https://github.com/KonssnoK/esp-idf/tree/bug/mesh_wifi
- connect to any hotspot

You will have connectivity to external network but no event triggered.

zhangyanjiaoesp, 2022-12-26:
@KonssnoK The root node can broadcast its external IP network connection status to all other nodes by calling [esp_mesh_post_toDS_state()](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-reference/network/esp-wifi-mesh.html#_CPPv424esp_mesh_post_toDS_stateb).
The event MESH_EVENT_TODS_REACHABLE can't be generated each time the root loses/gains the ability to reach the external network (Internet).

KonssnoK, 2022-12-26:
@zhangyanjiaoesp 
let me understand if i got it right:
TODS state is a manual event that needs to be triggered with [esp_mesh_post_toDS_state()](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-reference/network/esp-wifi-mesh.html#_CPPv424esp_mesh_post_toDS_stateb).
Correct?

zhangyanjiaoesp, 2022-12-26:
Yes

KonssnoK, 2022-12-27:
ok thanks, please put it somewhere in the docs :)

zhangyanjiaoesp, 2022-12-28:
ok, will update the docs

KonssnoK, 2023-01-30:
@zhangyanjiaoesp while we wait for the other ticket,
i tried to implement toDS propagation on disconnection from root to children

Please see commit
6bc5028311671c43e06e69749bbcb8691b0fc6b7

The purpose is quite simple:
- if root disconnects, post toDS state with FALSE
- if child receives toDS FALSE -> disconnect MQTT client

Currently what i see is that the toDS state is never received by the child on root disconnection

Any idea?
Please consider the other ticket as higher priority than this one.

KonssnoK, 2023-01-30:
EDIT: Apparently the toDS status is propagated only on change.

By adding the code in 
c40b03605b4fbce46d6b59febbecffaf5783e1f3

the children are now receiving the status change!

closed 2023-02-06 by KonssnoK