espressif/esp-idf #10043: Cannot find "xtensa-esp32-elf-gcc" VS on Windows

ANQUN12, 2022-10-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

**I have a problem with any code that uses xtensa-esp32-gcc**
[image]

### What is the actual behavior?

**I added IDF_PATH, xtensa-esp32-elf
However it gives the same error**

[image]
[image]

### Steps to reproduce.

1. idf.py set-target esp32       // **ok**
2. idf.py menuconfig       // **ok**
3. idf.py build     // **error**
...

closed 2022-10-28 by ANQUN12

Hairy-Lee, 2022-11-09:
What was the resolution to this issue? I have 4.4.3 installed and I'm experiencing the same issue

brianignacio5, 2023-05-22:
You need to specify the full compiler path (C:\esp\Espressif\tools\xtensa-esp32-elf\...\bin\xtensa-esp32-elf-gcc) instead of just the binary name. If you are using the ESP-IDF extension for Visual Studio Code, use the `ESP-IDF: Set Espressif device target` so the extension will update this compiler path automatically. 

Based on what you shared, seems you are running `idf.py` which doesn't update this c_cpp_properties.json file and you need to update it manually. I also don't t see the `xtensa-esp32-elf`

pixelprizm, 2023-06-03:
I had a similar issue (I'm using Rust but sharing in case it's helpful), and it was fixed when I installed MinGW using the instructions here: https://github.com/esp-rs/rust-build#prerequisites-x86_64-gnu.  Note that the `7z e` command should be `7z x` to maintain directory structure.  (Just the MinGW step -- for everything else I just used `cargo install espup`, and `espup install`.)