espressif/esp-idf #10045: how th work with session on http

hzkincony, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# IDF version : v4.3.3-dirty
<br>
My usage is saving login status by using session.The problem is when I refresh browser several times ,the esp32 can not get the session string.It,The session is't free.
the code is :
```
static void free_sess_ctx(void *ctx)
{
    ESP_LOGI(TAG, "Free Context function called");
    free(ctx);
}
if(checkAdmin(username,password))
{
	if (! req->sess_ctx) {
		ESP_LOGI(TAG, "allocating new session");
		req->sess_ctx = malloc(10);
		req->free_ctx = free_sess_ctx;
		memset(req->sess_ctx,0,10);
		sprintf(req->sess_ctx,"admin");
	}
}

static esp_err_t setting_html_get_handler(httpd_req_t *req)
{
	extern const unsigned char setting_html_start[] asm("_binary_setting_html_start");
	extern const unsigned char setting_html_end[]   asm("_binary_setting_html_end");
	const size_t file_len = (setting_html_end - setting_html_start);

	if((req->sess_ctx == NULL)||(strcmp(req->sess_ctx,"admin")!=0))
	{
		return move_302(req,"/login.html");
	}

	httpd_resp_set_type(req, "text/html");
	httpd_resp_send(req, (const char *)setting_html_start, file_len);

	return ESP_OK;
}
```

# When I login success.Esp32 creat session success too.When I refresh several time ,esp32 can not get the session string,and send 031 move.
But the session is't  delete now .When I close  the browser ,delete the session.

[image]
[image]

hzkincony, 2022-10-25:
print the sess_ctx value,first time is admin,the second is NULL

```
I (535200) web_server: allocating new session
I (537440) web_server: sess_ctx:admin
I (538970) web_server: sess_ctx:NULL
```

hmalpani, 2022-10-27:
Hello @hzkincony 
Could you also share debug logs?

hmalpani, 2022-11-07:
Hello @hzkincony
You can try [persistent sockets](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/persistent_sockets) example in esp-idf.
Also it will be helpful if you can share debug logs for your run. To enable debug logs, please enable `LOG_DEFAULT_LEVEL_VERBOSE` and `LWIP_SOCKETS_DEBUG` in menuconfig.

Alvin1Zhang, 2023-01-12:
Thanks for reporting, will close, feel free to reopen with more updates.

closed 2023-01-12 by Alvin1Zhang