espressif/esp-idf #10047: ip_napt forward not work for eth?

Bei-Ji-Quan, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Does ip_napt_enable only works for WIFI STA and AP?
 I use WIFI STA and WIFI AP with ip_napt_enable(), esp32 can work as an router. mobile phone can connect to internet  connetted while connected to ap. But using ETH and wifi AP,  mobile phone can't connect to internet while  connetted to ap.

david-cermak, 2022-10-26:
@Bei-Ji-Quan This should work also for Ethernet interface. Which IDF version are you using? Note that the default ETH netif didn't allow for *updating* packets in IP layer in versions ~ `v4.3`. It was possible to create a *non-default* Ethernet, as demonstrated in this example https://github.com/david-cermak/eth-ap-nat/tree/main/eth-ap-idf4.3.
(but should work normally with the default ETH on `v5.0` and later)

Bei-Ji-Quan, 2022-10-26:
thank you. I will have a try

Bei-Ji-Quan, 2022-10-28:
this example https://github.com/david-cermak/eth-ap-nat/tree/main/eth-ap-idf4.3 It works. thank you

closed 2022-10-28 by Bei-Ji-Quan