espressif/esp-idf #10048: When I use BLUFI esp_blufi_profile_deinit() is different from what I expected

WangShuoran, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am a new BLE user. My test program is modified on the blufi example. After BLUFI initialization, I add an event which is key event to execute esp_blufi_profile_deinit(), I feel that BLUFI still works, which I think is abnormal.

1. I understand esp_blufi_profile_init() is the registration of GATT, right?

2. That's in esp_blufi_profile_deinit() is the unregistration of GATT, is that right?

3. When I execute esp_blufi_profile_init() first time, the log printed is as follows:
[image]
But the esp_blufi_profile_init() return ESP_OK, I don't know why.

4. Who triggers ESP_BLUFI_EVENT_DEINIT_FINISH which register to `esp_blufi_callbacks_t`? When I use esp_blufi_profile_deinit(), the event can not be triggered. I do not know what will cause this event, and the document does not explain.

5. Is it impossible to unregister GATT in BLE after GATT registration?

WangShuoran, 2022-10-25:
This is my test code. I am band key to esp32c3's GPIO9:
[blufi_esp32c3.zip](https://github.com/espressif/esp-idf/files/9857637/blufi_esp32c3.zip)

xulongzhuang, 2022-10-26:
If you want to turn off bluetooth completely after using blufi, you can use the following sequence to deinitialize:
esp_blufi_profile_deinit();
esp_bluedroid_disable();
esp_bluedroid_deinit();
esp_bt_controller_disable();
esp_bt_controller_deinit();

Of course, the above is the order of de-initialization using bluedroid. If you are using nimble host, you need to explore further.

rahult-github, 2022-10-26:
Hi @WangShuoran ,

Recently , a [commit](https://github.com/espressif/esp-idf/commit/aa03ffdc1beb1b650453f3adfe4a89aab2e24c82) was made to add an API to stop Bluetooth/Blufi. Can you please refer the implementation of  [esp_blufi_host_deinit](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/blufi/main/blufi_init.c#L195) API to get the needed information ?

Thanks,
Rahul

WangShuoran, 2022-10-27:
@xulongzhuang Thank you for your quick reply. I am use Nimble host, and I upload my test code in this:

> This is my test code. I am band key to esp32c3's GPIO9: [blufi_esp32c3.zip](https://github.com/espressif/esp-idf/files/9857637/blufi_esp32c3.zip)

@rahult-github Thank you for your quick reply.

Sorry for my lots of same question, and I clear my question is: 

When I want blufi to start(broadcast its services), the blufi broadcast itself(I need other ble services cannot be affected), when I want to stop blufi(close broadcast its services), I need other ble services cannot be affected too.

Does [esp_blufi_host_deinit](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/blufi/main/blufi_init.c#L195) is my upper question answer?

WangShuoran, 2022-10-27:
I think [esp_blufi_host_deinit](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/blufi/main/blufi_init.c#L195) can not triggers ESP_BLUFI_EVENT_DEINIT_FINISH.
Beacuse the esp_blufi_host_deinit behavior in Nimble is:
```c
    esp_err_t ret = ESP_OK;

    ret = esp_blufi_profile_deinit();
    if(ret != ESP_OK) {
        return ret;
    }

    esp_blufi_btc_deinit();

    ret = nimble_port_stop();
    if (ret == 0) {
        nimble_port_deinit();
    }

    return ret;
```

This behavior is same which my upload test code at last. Which is my short sample code:

```c
            esp_blufi_profile_init();
            esp_err_t ret = esp_blufi_profile_deinit();
            if (ret != ESP_OK) {
                printf("error in esp_blufi_profile_deinit\n");
            }
            status = 0;
            ret = nimble_port_stop();
            if (ret != 0) {
                printf("error in nimble_port_stop %d\n", ret);
            }
```

And I think nimble_port_deinit() is cannot triggers ESP_BLUFI_EVENT_DEINIT_FINISH.

rahult-github, 2022-10-27:
Hi @WangShuoran 

> When I want blufi to start(broadcast its services), the blufi broadcast itself(I need other ble services cannot be affected), when I want to stop blufi(close broadcast its services), I need other ble services cannot be affected too.

Blufi for ESP32  is  more of a Wi-Fi network configuration function via Bluetooth channel. What you attempting above is more of extra implementation and not just blufi's implementation.  My suggestion is as below : 

1. Use blufi and do BLE related tasks.
2. If use case is to STOP blufi, then use the existing APIs to close blufi . But current implementation also closes the stack. 
3. If use case is to also use other services ( and don't care about blufi service ), then i think, it should be ok to not stop blufi. 
4. If use case is indeed to now use a different custom service ( after stopping blufi) , then the application will have to redo nimble init , and add the custom gatt server service again. 

As i see, in Nimble, there doesn't exist an API to run time remove a service from the gatt server. i.e. once a service is registered, it remains in the database unless the stack is stopped.

WangShuoran, 2022-10-28:
@rahult-github Thank you for your quick reply.

> What you attempting above is more of extra implementation and not just blufi's implementation.

Yes, I try to use nimble to let blufi and ble_spp or other ble gatt running in same time. So I need to register and unregister blufi at any time I want. 

As you say, I try to use bluedroid, and it will trigger ESP_BLUFI_EVENT_DEINIT_FINISH, so your say:

> As i see, in Nimble, there doesn't exist an API to run time remove a service from the gatt server. i.e. once a service is registered, it remains in the database unless the stack is stopped.

I think so, but why in nimble `esp_blufi_profile_deinit()` will return 0, I think if can not remove blufi gatt, return -1(error) is more consistent with constants, because my beginning question is esp_blufi_profile_deinit() is ok, but can not triggers ESP_BLUFI_EVENT_DEINIT_FINISH.

rahult-github, 2022-10-28:
Hi @WangShuoran , 

 The API [esp_blufi_host_deinit](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/blufi/main/blufi_init.c#L195) was recently added to give user one single function that can be invoked which handles profile/ stack cleanup. 
 
 For your query, i think, functionality is working correctly. It is just event is not posted . Please check if attached patch helps resolves your issue. 
[post_event.txt](https://github.com/espressif/esp-idf/files/9887603/post_event.txt)

Thanks,
 Rahul

WangShuoran, 2022-10-29:
I am try to understand the patch, you are use `btc_blufi_cb_to_app(ESP_BLUFI_EVENT_DEINIT_FINISH, &param);` instead of `btc_transfer_context(&msg, &param, sizeof(esp_blufi_cb_param_t), NULL);`, I am read the API name, I think the btc_blufi_cb_to_app is only send ESP_BLUFI_EVENT_DEINIT_FINISH, and any other behavior not trigger, That more like to trap, but I like it.
So do you think a beginner want the nimble and bluedroid behavior agreement? or in nimble esp_blufi_profile_deinit return -1 tell user in nimble can not unregister gatt, or return 0 and trigger the ESP_BLUFI_EVENT_DEINIT_FINISH?

rahult-github, 2022-10-31:
Hi @WangShuoran 

> That more like to trap, but I like it.

If you observe, the esp_blufi_deinit API logically does the reverse of what esp_blufi_init does. Hence, this implementation specific to Nimble .

> So do you think a beginner want the nimble and bluedroid behavior agreement? 

One point to note that both bluedroid and nimble are completely different stacks and hence comparing them codewise is not possible. The intention in IDF is to give as same API as possible and make an abstraction layer such that end application should not be concerned about the underlying stack. If a particular implementation is absent in one stack OR is handled in some other way, then giving two different APIs will not be a good solution.  

Having said, will still look at the feedback you provided and add necessaty changes, if any need to be done.

WangShuoran, 2022-10-31:
@rahult-github Thank you for your quick reply.

I think this issue is solve my problem perfectly. I think I will close this issue after comment this reply.

> One point to note that both bluedroid and nimble are completely different stacks and hence comparing them codewise is not possible. The intention in IDF is to give as same API as possible and make [...]

That is espressif's choose, I only put your question and my any suggest. But this suggest is only for me, not other. I recognize your preciseness.

And I think this topic is not very clear to give another any answer. So I will rewrite this topic as "blufi use nimble can not trigger ESP_BLUFI_EVENT_DEINIT_FINISH", I think it more help for other people, but U can modfiy come back.

>When I use BLUFI esp_blufi_profile_deinit() is different from what I expected

closed 2022-10-31 by WangShuoran