espressif/esp-idf #10049: Can ESP32C3 have 2 UART available while log via usb serial enabled?

xixixihaha, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32C3 has the capability to redirect the log via usb serial. I want to know if the 2 uart controller are available to user code under this situation. Or the default UART0 is still occupied. 
I can't find detail info in the documentation. 
Thanks!

nopnop2002, 2022-10-25:
This might help.

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_async_rxtxtasks

chipweinberger, 2022-10-25:
yes.

xixixihaha, 2022-10-25:
I want to use both UART0 and UART1 to connect to peripherals at different baud rate while using esp_log via usb. 
Is it possible?

nopnop2002, 2022-10-25:
>Is it possible?

UART_NUM_MAX of ESP32C3 is 2.

This shows that two UARTs are available.

By default, the UART0 is used to flash firmware and print logs.
UART0 and UART1 can be used to connect peripherals by changing the output destination of Logging.

nopnop2002, 2022-10-25:
What is logging via CDC？

xixixihaha, 2022-10-25:
> 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html

nopnop2002, 2022-10-25:
>I want to know if ESP32C3's logging via CDC conflict with UART0. 

The output to the CDC port (GPIO18&GPIO19) is also output to UART0.

nopnop2002, 2022-10-25:
Try this on your ESP32C3.
This example uses /dev/uart/0.

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_select

You can use CDC to flash the firmware, but logging will still output to UART0.

FabITA1, 2022-10-26:
@xixixihaha I'm working on a similar project. I'm using a ESP32C3 device connected on a 4G modem on Uart1 and on a host mcu on Uart0. Trough the usb port I'm able to debug via OpenOcd and monitor the log uart at the same time. 
The only problem is that on the boot phase some log messages comes out from Uart0, but when the application starts all log are redirect on the usb port.

nopnop2002, 2022-10-26:
I was able to separate Logging and UART0 communication using Logging's redirect function.

https://github.com/nopnop2002/esp-idf-net-logging/tree/main/uart_async_rxtxtasks

nopnop2002, 2022-10-26:
@xixixihaha 

>The only problem is that on the boot phase some log messages comes out from Uart0

Solve by using UART0 other than GPIO1/GPIO3.

```
#define TXD_PIN (GPIO_NUM_4)
#define RXD_PIN (GPIO_NUM_5)

uart_set_pin(UART_NUM_0, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
```

nopnop2002, 2022-10-26:
@igrr 

>It's also possible to disable the ROM bootloader output using eFuses.

I can’t find in menuconfig.

xixixihaha, 2022-10-26:
> 

@FabITA1 Thanks so much. I'll give it a try.

ESP-Marius, 2022-11-02:
It should also be possible to disable the ROM bootloader from menuconfig now with the `BOOT_ROM_LOG_SCHEME` option from the `esp_rom` component.

xixixihaha, 2022-11-14:
@FabITA1 I've tried to run the uart sample code on uart0. The ESP_LOGI messages also went out onto the uart0 port. Please check the hw log on analyzer.
Have you set any specific log setting?
Thanks!

> @xixixihaha I'm working on a similar project. I'm using a ESP32C3 device connected on a 4G modem on Uart1 and on a host mcu on Uart0. Trough the usb port I'm able to debug via OpenOcd and monitor [...]

[image]

FabITA1, 2022-11-14:
@xixixihaha have you change in menuconfig the log preferencies?:
* First for "Channel for console output" you need to select "USB Serial/JTAG Controller"
* Then for "Channel for console secondary output" you need to select "No secondary console"

xixixihaha, 2022-11-14:
> @xixixihaha have you change in menuconfig the log preferencies?:
> 
>     * First for "Channel for console output" you need to select "USB Serial/JTAG Controller"
> 
>     * Then for "Channel for console secondary output" you need to select "No secondary console"

@FabITA1 Thanks so much! I've changed the setting from uart0 to usb serial. Now there is no log on uart0 now. Really appreciate your support!

ESP-Marius, 2022-11-15:
@xixixihaha Sounds like your issue has been resolved then so ill close this ticket.

Feel free to reopen it if there is still any outstanding issues.

closed 2022-11-15 by ESP-Marius