espressif/esp-idf #10051: [esp32 s3] Allow .bss segment placed in external memory

Mbded, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello 

I have some project which was compiled for first version chip esp32 with vesrion esp-idf 4.3. And in this project option
**CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY** 
was enabled and was used.
My new task is compile this same project but under esp32**s3.** Firstly I wanted to use IDF form master baranch but I received to much error so I decided to change IDF version to 4.4.2 (latest stable) My project was compiled but with error with exhausted memory

`Used static IRAM:   66750 bytes ( 295490 remain, 18.4% used)
      .text size:   65723 bytes
   .vectors size:    1027 bytes
**Used stat D/IRAM:  204889 bytes ( -31961 remain, 118.5% used) Overflow detected!**
      .data size:   11321 bytes
      .bss  size:  193568 bytes
Used Flash size : 1680543 bytes
      .text     : 1213455 bytes
      .rodata   :  466832 bytes
Total image size: 1758614 bytes (.bin may be padded larger)` 

A quick check revealed that I need also to enable **Allow .bss segment placed in external memory** . Unfortunately under IDF v4.4.2 it is impossible. 
Can I find some fix for S3 for IDF v4.4.2 ?

franz-ms-muc, 2022-12-07:
I think this is solved by ESP-IDF 5.0

AxelLin, 2023-05-05:
I think f8249550f82262d3edd3fa2c0a5e29feed8cc0a7 is not available in v4.4 branch.

GaryChan-work, 2023-05-16:
Hello,

We have the same problem here on IDF v4.4.4, could you provide the fix on the coming release e.g. v4.4.5?