espressif/esp-idf #10054: intermittent network failure with ESP32/rtl8201: update link duplex speed failed

chrisovergaauw, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### Development Kit.

Custom PCB with ESP32-WROOM-32UE (8MB)

### Power Supply used.

External 5V

### What is the expected behavior?

open a socket connection and be able to read/write as required.

### What is the actual behavior?

sockets become unresponsive when this is printed:
```
E (1516180) esp.emac: emac_esp32_write_phy_reg(85): phy is busy
E (1516180) rtl8201: rtl8201_page_select(75): write PSR failed
E (1516180) rtl8201: rtl8201_update_link_duplex_speed(91): select page 0 failed
E (1516190) rtl8201: rtl8201_get_link(144): update link duplex speed failed
```

### Steps to reproduce.

I need help trying to reproduce this without the use of MIcropython.
According to a micropython developer this information might be relevant (to show that it is likely the issue is with esp-idf and not with micropython)

> MicroPython only provides the very minimum layer on top of the lwip/sockets layer and configuration of the NIC. If necessary, you can point them to https://github.com/micropython/micropython/blob/master/ports/esp32/network_lan.c and https://github.com/micropython/micropython/blob/master/ports/esp32/modsocket.c
Importantly, once the NIC is configured, we only use the lwip/sockets API.

I would like to try the iperf example but run into the following issue:
it boot loops without being able to initialize.
```
E (819) esp_eth: esp_eth_driver_install(214): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x40086c38
0x40086c38: _esp_error_check_failed at /Users/chris/somatics/git/fridge-smart-controller/esp-idf/components/esp_system/esp_err.c:42

file: "IDF/examples/ethernet/iperf/main/cmd_ethernet.c" line 319
func: register_ethernet
expression: esp_eth_driver_install(&config, &eth_handle)

abort() was called at PC 0x40086c3b on core 0
0x40086c3b: _esp_error_check_failed at /Users/chris/somatics/git/fridge-smart-controller/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x40081bfa:0x3ffb66200x40086c45:0x3ffb6640 0x4008f8c6:0x3ffb6660 0x40086c3b:0x3ffb66d0 0x400d6c79:0x3ffb66f0 0x400d6881:0x3ffb67a0 0x4010d31a:0x3ffb67f0 0x4008a4b9:0x3ffb6810
```

### Debug Logs.

```plain
E (1516180) esp.emac: emac_esp32_write_phy_reg(85): phy is busy
E (1516180) rtl8201: rtl8201_page_select(75): write PSR failed
E (1516180) rtl8201: rtl8201_update_link_duplex_speed(91): select page 0 failed
E (1516190) rtl8201: rtl8201_get_link(144): update link duplex speed failed
```
```

### More Information.

I have found a similar issue with a different PHY and it is two years old:
https://github.com/espressif/esp-idf/issues/4747

chrisovergaauw, 2022-10-26:
I've ran into a variation of the error today:
```
E (1255320) esp.emac: emac_esp32_read_phy_reg(107): phy is busy
E (1255320) rtl8201: rtl8201_update_link_duplex_speed(92): read BMSR failed
E (1255320) rtl8201: rtl8201_get_link(144): update link duplex speed failed
```

It still results in a an unresponsive socket

chrisovergaauw, 2022-10-26:
I'm wondering if similiar errata's like these could still be applied to the rtl8201 driver:
https://github.com/espressif/esp-idf/commit/ae2da1e2352a15867d47008d0b8029882f35d5ff

chrisovergaauw, 2022-10-28:
context: trying to get Iperf to work for a reproducable scenario.

I've not been able to get around the ` insufficient TX buffer size` yet.
I've added `ESP_LOGW` statements to compare the working uPy setup vs the non-working iperf.
They are both identical.

So the mac intializing in the iperf example is identical to the initialization in uPy:
```
// examples/ethernet/iperf/main/cmd_ethernet.c
esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
```
results in:
```
W (2935) mac_debug: sw_reset_timeout_ms: 100
W (2935) mac_debug: sw_reset_timeout_ms: 100
W (2945) mac_debug: rx_task_stack_size: 2048
W (2945) mac_debug: rx_task_prio: 15
W (2945) mac_debug: smi_mdc_gpio_num: 16
W (2955) mac_debug: smi_mdio_gpio_num: 17
W (2955) mac_debug: flags: 0
W (2965) mac_debug: interface: 0
W (2965) mac_debug: clock_mode: 0
W (2965) mac_debug: clock_gpio: 0
```

and 
```
// ports/esp32/network_lan.c
esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
```
results in:
```
W (2930) mac_debug: sw_reset_timeout_ms: 100
W (2930) mac_debug: sw_reset_timeout_ms: 100
W (2930) mac_debug: rx_task_stack_size: 2048
W (2930) mac_debug: rx_task_prio: 15
W (2940) mac_debug: smi_mdc_gpio_num: 16
W (2940) mac_debug: smi_mdio_gpio_num: 17
W (2950) mac_debug: flags: 0
W (2950) mac_debug: interface: 0
W (2950) mac_debug: clock_mode: 0
W (2960) mac_debug: clock_gpio: 0
```

same for phy_config:
```
// ports/esp32/network_lan.c
esp_eth_phy_t *phy = esp_eth_phy_new_rtl8201(&phy_config);
```
results in:
```
W (2975) phy_debug: phy_addr: 0
W (2975) phy_debug: reset_timeout_ms: 100
W (2985) phy_debug: autonego_timeout_ms: 4000
W (2985) phy_debug: reset_gpio_num: -1
```

and:
```
// ports/esp32/network_lan.c
self->phy = esp_eth_phy_new_rtl8201(&phy_config);
```
results in:
```
W (2960) phy_debug: phy_addr: 0
W (2970) phy_debug: reset_timeout_ms: 100
W (2970) phy_debug: autonego_timeout_ms: 4000
W (2980) phy_debug: reset_gpio_num: -1
```

kostaond, 2022-11-22:
Sorry for late response.

> I've not been able to get around the insufficient TX buffer size yet.

This is caused by fact, that you are trying to write more data than EMAC is able to handle. I.e. in ideal case, maximum throughput is 100 Mbit/s but you are trying to transmit more. Practically it is a little less (you can try 90 or 80 Mbits/s).
The way to limit Tx bandwidth in iperf is by using `-b` option:
`iperf -u -c PC_IP -i 3 -t 30 -b 80`

Regarding
`E (5295270) esp.emac: emac_esp32_read_phy_reg(107): phy is busy`
issue, thanks for reporting I'll investigate it but we need to find a way how to reproduce it.

chrisovergaauw, 2022-11-24:
Thank you, I didn't spot the '-b' flag in the README.md.
I ran the UDP example at 80 Mbits/s for 2 hours straight and it worked well:
```
[  1] 0.00-7200.00 sec  70.3 GBytes  83.9 Mbits/sec
```

Only the last iteration/closing is not handled properly:
```
# esp32 as server
I (7225170) iperf: Udp socket server is closed.
I (7225170) iperf: iperf exit
```

```
# macbook as client
[  1] 0.00-7200.00 sec  70.3 GBytes  83.9 Mbits/sec
[  1] Sent 51358829 datagrams
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
read failed: Connection refused
[  4] WARNING: did not receive ack of last datagram after 10 tries.
``` 
I will try TCP now as that's what we're using in our real world application

kostaond, 2022-11-29:
> Only the last iteration/closing is not handled properly

I think this is normal...

Were you able to reproduce the original issue?

chrisovergaauw, 2022-11-29:
No, unfortunately I haven't.
I'll reopen whenever I succeed.
Thanks for your help so far.

closed 2022-11-29 by chrisovergaauw

kostaond, 2022-11-29:
> edit (just for clarity, it's still an issue in my project, but I understand that you cannot do anything if I cannot reproduce it)

You can try to create simple app with Micropython based on your project. And if you were able to reproduce with it, you could share it and we could have a look into it in details.