espressif/esp-idf #10055: ESP32-S2 stalls for 4.28ms every 2 secs

dmitrykez, 2022-10-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an issue with ESP32-S2-WROOM which stalls for 4.28ms every 2 seconds. 
I tries several modules (same P/N) and all of them behave similar. 

To reproduce the problem I used a very simple code which constantly toggles GPIO in a main loop. 
`void loop() 
{
    debug_gpio = !debug_gpio;
    digitalWrite(MY_GPIO, debug_gpio);
}`

The chip remains with a default setting as I don't configure anything on boot. 
Only SPIFFS is being configured, but it doesn't change the behavior. 

Pause every 2 seconds:
[image]

Pause length is 4.27ms:
[image]

Anyone familiar with the phenomena?

PepeTheFroggie, 2024-03-28:
No solution for this?
This is VERY> annoying if you try to fly a quadcopter with an esp32s2 !

PepeTheFroggie, 2024-03-29:
In the main Arduino loop, above loop() function, in the file cores/esp32/main.cpp we see yieldIfNecessary() function. It works on ESP32-S2 for example as it only has one core. Function adds a 5ms delay every two seconds. Under some specific circumstances, this causes problem with loop() execution time.

https://github.com/espressif/arduino-esp32/issues/6119