espressif/esp-idf #10056: ESP32-S2: Very slow startup time compared to ESP32

ProjectProcrastinator, 2022-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I use an ESP32-S2-Solo module on a battery powered custom PCB to read sensor data and send the data with ESP-NOW. Therefore I desire a fast boot and code execution time to preserve battery life. My project works but the boot time of the ESP32-S2 seems really slow. I measured both the total time for code execution including startup and the time just for the execution of my code using `millis()`. On my initial try using Arduino-framework with PlatformIO I got ~510ms total, ~60ms for code execution. So 450ms just for startup from deep sleep. Seems very long for me.
So I tried to improve it and therefore migrated the code to ESP-IDF using Arduino as a component with PlatformIO. I used all suggested options by Espressif in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html to speed up startup and overall performance. Without edits in menuconfig the startup time was even higher. With the suggested edits I got the same time. Still slow. 
So I wrote a simple sketch just to test startup speed:

```
#include <Arduino.h>
 
void setup() {
    int start = millis();
    Serial.begin(115200);
    Serial.println("Starting");
    esp_sleep_enable_timer_wakeup(10 * 1000000);
    Serial.println((String)"Time from boot = " + millis() + (String)"ms");
    int stop = millis();
    int time = stop - start;
    Serial.println((String)"Time for code execution = " + time + (String)"ms");
    Serial.println();
    esp_deep_sleep_start();
}
 
void loop()  {}
```

Results for ESP32-S2 without menuconfig edits: 678ms total, 1ms for code execution
Results for ESP32-S2 with menuconfig edits: 443ms total, 1ms for code execution

Then I used the same code with an ESP32-Module, but with [board = esp32doit-devkit-v1] in platformio.ini instead of [board = esp32-s2-saola-1]. That was the only difference.

Results with ESP32 without menuconfig edits: 47ms total, 1ms for code execution
Results with ESP32 with menuconfig edits: 27ms total, 1ms for code execution. 

So it seems to me that ESP32 has a way faster startup than ESP32-S2. 

Later I tried some more things. I ditched the Arduino-code and just used ESP-IDF with that code:

```
#include <stdio.h>
#include "esp_timer.h"
#include "esp_sleep.h"

void app_main(void)
{
    printf("\n");
    printf("Time since boot: %lld us\n", esp_timer_get_time());

    esp_sleep_enable_timer_wakeup(10 * 1000000);    
    esp_deep_sleep_start(); 
}
```

I tried that code with PlatformIO and with ESP-IDF VSCode Extension. With the latter I tried various versions of ESP-IDF, including the current stable version, some earlier versions, version 5.0 and the current master. Startup time varied from version to version but it was never better than above, rather worse, like 600-700ms.

The ESP32-S2-Solo Module was bought from Digikey and the PCB uses the exact layout and peripheral components as suggested in the official datasheet. I don't see a reason why ESP32-S2 should be that much slower than ESP32. Am I doing something wrong or might that be a bug of some sort?

chipweinberger, 2022-10-29:
the bootloader logs what it is doing during boot. 

you should compare the two devices logs

ProjectProcrastinator, 2022-10-29:
Thanks for the advice.
I have used this code:
```
#include <Arduino.h>

void setup() {
    int start = millis();
    Serial.begin(115200);
    Serial.println("Starting");
    esp_sleep_enable_timer_wakeup(10 * 1000000);
    Serial.println((String)"Time from boot = " + millis() + (String)"ms");
    int stop = millis();
    int time = stop - start;
    Serial.println((String)"Time for code execution = " + time + (String)"ms");
    Serial.println();
    esp_deep_sleep_start(); 
}

void loop()  {}
```
for ESP32 and ESP32-S2. 
In menuconfig i have set both to: 
Bootloader log verbosity = Verbose 
and 
Default log verbosity = Verbose.
Boot ROM output is permanently disabled for ESP32-S2.

Result for ESP32:
```
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:10544
load:0x40078000,len:18836
ho 0 tail 12 room 4
load:0x40080400,len:5228
entry 0x40080720
␛[0;32mI (29) boot: ESP-IDF 4.4.2 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 13:40:21␛[0m
D (29) bootloader_flash: non-XMC chip detected by SFDP Read (00), skip.␛[0m
D (35) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)␛[0m
␛[0;32mI (42) boot: chip revision: 1␛[0m
␛[0;32mI (45) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
D (52) qio_mode: Probing for QIO mode enable...␛[0m
D (57) qio_mode: Raw SPI flash chip id 0x5e4016␛[0m
D (62) qio_mode: Manufacturer ID 0x5e chip ID 0x4016␛[0m
␛[0;32mI (67) qio_mode: Enabling default flash chip QIO␛[0m
D (72) qio_mode: Initial flash chip status 0x2␛[0m
D (76) qio_mode: QIO mode already enabled in flash␛[0m
D (81) qio_mode: Enabling QIO mode...␛[0m
D (85) boot.esp32: magic e9␛[0m
D (88) boot.esp32: segments 03␛[0m
D (91) boot.esp32: spi_mode 02␛[0m
D (94) boot.esp32: spi_speed 0f␛[0m
D (97) boot.esp32: spi_size 02␛[0m
␛[0;32mI (101) boot.esp32: SPI Speed      : 80MHz␛[0m
␛[0;32mI (105) boot.esp32: SPI Mode       : QIO␛[0m
␛[0;32mI (110) boot.esp32: SPI Flash Size : 4MB␛[0m
D (114) boot: Enabling RTCWDT(9000 ms)␛[0m
␛[0;32mI (118) boot: Enabling RNG early entropy source...␛[0m
D (124) esp_image: reading image header @ 0x20000␛[0m
D (129) bootloader_flash: mmu set block paddr=0x00020000 (was 0x00000000)␛[0m
V (135) esp_image: loading segment header 0 at offset 0x20018␛[0m
V (141) esp_image: segment data length 0xae30 data starts 0x20020␛[0m
V (147) esp_image: segment 0 map_segment 1 segment_data_offs 0x20020 load_addr 0x3f400020␛[0m
␛[0;32mI (156) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=0ae30h ( 44592) map␛[0m
D (164) esp_image: free data page_count 0x00000032␛[0m
[... 434 lines cut ...]
D (1113) intr_alloc: Connected src 73 to int 10 (cpu 0)␛[0m
V (1118) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args␛[0m
V (1125) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E␛[0m
D (1133) intr_alloc: Connected src 28 to int 2 (cpu 0)␛[0m
␛[0;32mI (1138) cpu_start: Starting scheduler on PRO CPU.␛[0m
D (1144) heap_init: New heap initialised at 0x3fffc000␛[0m
V (1149) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args␛[0m
V (1149) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE␛[0m
D (1149) intr_alloc: Connected src 17 to int 3 (cpu 0)␛[0m
D (1169) partition: Loading the partition table␛[0m
V (1173) calculated md5: 0x3ffc5478   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|␛[0m
V (1183) stored md5: 0x3f018070   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|␛[0m
D (1192) partition: Partition table MD5 verified␛[0m
␛[0;32mI (1201) uart: queue free spaces: 20␛[0m
V (1202) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args␛[0m
V (1207) intr_alloc: espStarting
Time from boot = 1625ms
Time for code execution = 19ms

D (1221) sleep: RTC_PERIPH: AUTO(OFF)␛[0m
D (1221) sleep: RTC_SLOW_MEM: ON␛[0m
D (1221) sleep: RTC_FAST_MEM: ON␛[0m
```

To be honest, I don't understand most of this, so help would be greatly appreciated. Notable is the huge difference in time needed. 269ms (ESP32) vs 1625ms (ESP32-S2). The additional logs seem to have a much greater effect on ESP32-S2 compared to ESP32.

ESP-Marius, 2024-08-15:
@ProjectProcrastinator, Sorry for the super slow response.

Are you still facing issues here? In our internal CI i'm seeing similar numbers for ESP32 vs ESP32S2 when running our [startup time example](https://github.com/espressif/esp-idf/tree/master/examples/system/startup_time)

So as far as I can see there doesnt seem to be anything wrong here, so ill mark this as closed.

If you are still facing issues, even when applying the configs shown in the example, then feel free to re-open this ticket and ill help take a look.

closed 2024-08-15 by ESP-Marius