espressif/esp-idf #10058: ESP32的ADC连续读取模式为什么测量误差这么大

kurau1, 2022-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

开发板模组：ESP32-WROOM-32
操作系统：WIN10
开发软件：VSCODE
ESP-IDF版本：V4.4.2

我使用了ESP32-S2（80k采样率）以及ESP32（250k采样率）分别对一个5Khz的正弦波信号进行了测量（100mv-1000mv）

下面是ESP32的测试结果图：

[image]
[image]

下面是ESP32-S2的测试结果图：

[image]
[image]

可以看出无论是时域还是频域，ESP32得到的结果都比较糟糕，我有在网上听说ESP32的ADC测不太准，但是他这个不准得很有规律。从频域来看就像叠加了120KHz的高频。我想知道这是硬件问题还是软件问题，如果是硬件问题我就要重新选择adc采样方案了。

还有一个困惑我很久的问题，ESP32文件里写明了，其拥有2Mhz的采样率。
```
#define SOC_ADC_SAMPLE_FREQ_THRES_HIGH          (2*1000*1000)
#define SOC_ADC_SAMPLE_FREQ_THRES_LOW           (20*1000)
```
但实际上我只能将其设置到250khz，如果我将其设置为`.sample_freq_hz = 500*1000,`，我得到的结果里，每相邻两个结果会是一摸一样的，当我设置为`.sample_freq_hz = 1000*1000,`，每相邻四个结果也是一样的。所以等同下来还是250KHz的实际采样率啊。不知道我有没有表达清楚。我想知道使用官方`idf\esp-idf_4.4,2\esp-idf\examples\peripherals\adc\dma_read`此例程到底能不能达到2Mhz的采样率啊T T。

以上是我的两个问题，谢谢大家！

eos1d3, 2022-10-30:
以dma讀adc，ESP32及其它新型號都是經interrupt，而ISR 是驚人的複雜及很大代碼。250Kz我都感覺cpu耗損死了，之前版本還用Queue來儲數據，這樣CPU基本沒其它事可以做，而只能做ADC運算。

他們完全不懂analog需要，這個ADC API根本沒用也不能用於即時處理。

我們做了2個設計，一個是用STM32 + esp32c3，另一個是只用ESP32-S3, 結果選了前者，因為後者要over-sampling, 99% CPU都在做ADC。而STM32 DMA ADC加oversampling, 全硬體，完全不用cpu。

這是ESP32一個嚴重問題。

kurau1, 2022-10-31:
> 以dma讀adc，ESP32及其它新型號都是經interrupt收集，而ISR 是驚人的複雜及很大代碼。250Kz我都感覺cpu耗損死了，之前版本還用Queue來儲數據，這樣CPU基本沒其它事可以做，而只能做ADC運算。
> 
> 他們完全不懂analog需要，這個ADC API根本沒用也不能用於即時處理。
> 
> 我們做了2個設計，一個是用STM32 + esp32c3，另一個是只用ESP32-S3, 結果選了前者，因為後者要over-sampling, 99% CPU都在做ADC。而STM32 DMA ADC加oversampling, 全硬體，完全不用cpu。
> 
> 這是ESP32一個嚴重問題。

@eos1d3 你好，非常感谢你回复我的问题。
根据你的建议，我使用vTaskGetRunTimeStats（）；查询了一下系统的任务占用，系统打开250khz采样率一直采样并计算采样率、系统每3秒打印一次任务占用情况。我不知道是不是我测试方法有问题，我得到的占用情况表示系统大部分时间都处于空闲，结果如下：
```
--------------- heap:173728 ---------------------
main            1904374         <1%
speed_test      19027868                9%
IDLE            180307002               88%
IDLE            202407328               99%
count_func      55304           <1%
tiT             129326          <1%
ipc0            14143           <1%
Tmr Svc         21              <1%
esp_timer       239411          <1%
wifi            1190081         <1%
sys_evt         2332            <1%
ipc1            66074           <1%
----------------------------------------------
fps = 244736
fps = 246400
fps = 246272
```
IDLE是FreeRTOS创建的空闲任务，它总是占用90%左右。`speed_test  `  和  `count_func  `    是我的任务，前者是ad采样任务，后者是每一秒输出一次采样率的任务。他们的占用情况并不高，我不知道是不是我测量方法的问题。如果你注意到我这个问题的话，希望能给我些意见，谢谢！

eos1d3, 2022-10-31:
大家的取樣不太一樣。

說回最早時，大家都發現ESP32 ADC的noise是超級恐怖。對比大部分mcu來說，直流訊號adc一般也就+-1個單位。
而ESP32至ESP32-S3, C3都不止這麼一點點。ESPxx一般都可以突然跳個50多出來。如開了WiFi, DC訊號可以跳個100多出來。
看這裡：https://github.com/espressif/esp-idf/issues/164

於是ESP公司又搞出無聊花樣，叫大家加filter capacitors, 這可以。但又要多次取樣(oversampling)再算平均數。我試了讀一個通道至少要128次才可以，比如5個通道，就是要再加快5 x 128倍。這對於普通單片機的處理來說，簡直就是個笑話。我用ST的 STM8, STM32都從不用CPU大量處理ADC。特別是STM32, 128次取樣及算平均數，都是mcu的ADC自己來，CPU沒任何負擔。ESP就是完全不懂ADC。

你可以看看他們官方文件，看了你還敢用嗎？
https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/peripherals/adc.html#minimizing-noise

用了大量處理, 結果比一般單片機一次取樣還比不上。我測試都是用espressif的官方開發板。他們adc api的文件也有說這個noise問題。

還有是他們的ADC API, 讀取的數據如程序沒及時處理，之後讀的的全是以前的數據。那問題來了，比如要做過流保護，這些數據可以是久以前的。你的線路可能一早就炸了。

所以對ADC有要求，只可以用別的。我目前一個項目就是用STM32+C3，C3只做無線通訊。
而ESP32的ADC, 是所有ESP 中最差最有缺陷的。但新的也不好到那裡去。
硬件先天有缺陷，ADC API library也一樣差到極點。

kurau1, 2022-10-31:
> 大家的取樣不太一樣。
>
> 說回最早時，大家都發現ESP32 ADC的noise是超級恐怖。對比大部分mcu來說，直流訊號adc一般也就+-1個單位。 而ESP32至ESP32-S3, C3都不止這麼一點點。ESPxx一般都可以突然跳個50多出來。如開了WiFi, DC訊號可以跳個100多出來。 看這裡：#164
>
> 於是ESP公司又搞出無聊花樣，叫大家加filter capacitors, [...]

感谢你的回答

不论是噪声还是你提到的 `读取数据是以前的 `都解决了我之前遇到的疑惑。我之前用的时候也发现 每次使用`adc_digi_read_bytes`读取，前`max_store_buf_size`的数据都是从前的数据的。不读取就不会更新。

根据你的建议我也需要更换adc采样方案，我打算采用ESP32+外部ADC的方案，ESP32作为主控与通讯，通过ESP32的SPI读取外部ADC采集的值，ESP32的SPI还未学，不知能否和STM32的一样好用，能不能直接使用dma存储，这都值得我去探索。

最后，真的很感谢你解开我的疑惑。
谢谢！

dreamtrek, 2022-11-01:
> > 大家的取樣不太一樣。
> > 說回最早時，大家都發現ESP32 ADC的noise是超級恐怖。對比大部分mcu來說，直流訊號adc一般也就+-1個單位。 而ESP32至ESP32-S3, C3都不止這麼一點點。ESPxx一般都可以突然跳個50多出來。如開了WiFi, DC訊號可以跳個100多出來。 看這裡：#164
> > 於是ESP公司又搞出無聊花樣，叫大家加filter capacitors, 這可以。但又要多次取樣(oversampling)再算平均數。我試了讀一個通道至少要128次才可以，比如5個通道，就是要再加快5 x 128倍。這對於普通單片機的處理來說，簡直就是個笑話。我用ST的 STM8, STM32都從不用CPU大量處理ADC。特別是STM32, 128次取樣及算平均數，都是mcu的ADC自己來，CPU沒任何負擔。ESP就是完全不懂ADC。
> > 你可以看看他們官方文件，看了你還敢用嗎？ https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/peripherals/adc.html#minimizing-noise
> > 用了大量處理, 結果比一般單片機一次取樣還比不上。我測試都是用espressif的官方開發板。他們adc api的文件也有說這個noise問題。
> > 還有是他們的ADC API, 讀取的數據如程序沒及時處理，之後讀的的全是以前的數據。那問題來了，比如要做過流保護，這些數據可以是久以前的。你的線路可能一早就炸了。
> > 所以對ADC有要求，只可以用別的。我目前一個項目就是用STM32+C3，C3只做無線通訊。 而ESP32的ADC, 是所有ESP 中最差最有缺陷的。但新的也不好到那裡去。 硬件先天有缺陷，ADC API library也一樣差到極點。
> 
> 感谢你的回答
> 
> 不论是噪声还是你提到的 `之後讀的的全是以前的數據 `都解决了我之前遇到的疑惑。我之前用的时候也发现 每次使用`adc_digi_read_bytes`读取，前`max_store_buf_size`的数据都是从前的数据的。不读取就不会更新。
> 
> 根据你的建议我也需要更换adc采样方案，我打算采用ESP32+外部ADC的方案，ESP32作为主控与通讯，通过ESP32的SPI读取外部ADC采集的值，ESP32的SPI还未学，不知能否和STM32的一样好用，能不能直接使用dma存储，这都值得我去探索。
> 
> 最后，真的很感谢你解开我的疑惑。 谢谢！

SPI应该也有类似的问题，如果你要频繁调用SPI传输小数据的话，要注意ESP32的SPI启动比一般单片机SPI的启动耗费的时间更长，也许无法达到要求

kurau1, 2022-11-01:
你好， @dreamtrek 
我的确需要频繁调用SPI传输数据，因为我打算采用500ksps的adc进行采样，然后量化后立即通过SPI传给ESP32。

我将乐鑫编程指南关于spi的部分看了一遍，感觉还是似懂非懂，至于能不能达到要求，也必须硬着头皮去尝试了 T T。

eos1d3, 2022-11-01:
我認為ESP32 SPI 沒有什麼啟動問題, 只要啟動後就可以用, 反正SPI 長時間在工作, 不會有延遲的。

不過SPI 接收後, 要polling或interrupt收集及整理, 這又是費時的事。就是用DMA也一樣。不像STM32那樣有個ring buffer不斷重寫而程序什麼都不用理。

這個引申到另外問題, 用外加的ADC, 成本很高, 比mcu還貴。

espressif這個API 問題是他們只懂用I2S收集音頻, 所以每個ADC讀數都要存, 但這種做法, 基本上ADC都不是這樣用。他們完全不懂, 我公司有聯繫他們技術人員, 但說了還是不懂。做數字訊號的人, 真的不懂analog。有ADC要求的, 不要用ESP做主控。

dreamtrek, 2022-11-01:
> 

是的，我上面的表述有问题，应该说从调用传输函数到真正开始数据传输之间有个较长的间隔，官方的说法好像是20us。之前有个应用采用Polling方式每次发送8字节数据，结果发现就算开个线程不断调用函数发送数据，速度依旧达不到要求，最后只能增加单次传输的数据量

kurau1, 2022-11-10:
> > 
> 
> 是的，我上面的表述有问题，应该说从调用传输函数到真正开始数据传输之间有个较长的间隔，官方的说法好像是20us。之前有个应用采用Polling方式每次发送8字节数据，结果发现就算开个线程不断调用函数发送数据，速度依旧达不到要求，最后只能增加单次传输的数据量

是的，每次发送间隔实测有10-20us的一个延时，但spi通信的adc转换触发信号有时就取决于cs信号线的拉高，拉高之后采样量化编码，拉低之后传输数据，间隔这么高的话，根本无法实现高采样率。
[image]

eos1d3, 2022-11-10:
我之前就說了，ESP只能用於通訊，要ADC我只考慮另一個MCU。就是SPI沒延遲也要用中斷不斷高速處理ADC數據。這是很不合理的做法。

我試了ESP32, 其ADC是最差的一個。而C3及S3只改善了esp32的嚴重缺陷，但基本問題完全沒法解決。

你的發現更確認了我的看法。

kurau1, 2022-11-10:
可我如果采用STM32进行adc采样，我还是需要使用wifi发送出来，这样STM32和ESP32之间还是会进行数据高速传输呀（我看了串口透传根本不够，我一秒里面可以有好几百KB的数据）。这个困境有办法能解决吗？

eos1d3, 2022-11-10:
不會吧？ 我的做法是用STM32處理所有的事情，連ADC都不給ESP32, 當需要上傳時才把數據發給ESP32。
而ESP32大部分時間就是睡眠。

你需要不斷發送ADC上網？如要連續轉發ADC數據，ESP32估計不能用。看看RTL8720這類行不行？ 我沒用過。

另用SPI的CS腳，是否可以永久pull low? 即永遠連著ADC IC，這就沒了delay, 但有的SPI 下要求CS low -> high以代表完成一個命令。

你如果是公司開發，可以直接找espressif，他們對公司有支援。

kurau1, 2022-11-10:
是的，我是学生做课题。需要对多个振动信号采样（需要多通道），随后在pc上位机上复现，类似于一个无线示波器，所以需要高速的数据传输。

感谢你提供新的解决方案RTL8720，在我的印象里多数wifi模块都是做成串口透传的模式，使用AT指令控制，这种模式操作方便，但是受限于串口的波特率，速度表现无法达到我的需求。我准备研究ESP32使用SPI快速接收STM32数据，看看速度是否能达到需求，或者找找SPI传输的wifi模块。

对于SPI的CS脚，我读了TI一款比较新的ADC芯片手册AD7067，得出的结论是无法永久pull low，因为cs的拉高对adc来讲代表着命令接收结束 开始采样 或是 切换采样通道采样，没有cs的拉高adc就无法正常工作。
[image]

eos1d3, 2022-11-10:
cs腳永pull low, 是不一定可以，比如SPI EEPROM我就發現不可以。

如不是做產品，不如考慮Raspberry Pi Pico W 。W是無線版本。

Raspberry Pi Pico 有官方 API, 寫得很不錯的。開發也簡單。

kurau1, 2022-11-10:
我去看看Raspberry Pi Pico W，谢谢你！

kurau1, 2022-11-23:
最近逛论坛，我找到了关于该问题的答案，以下回答的原话：
`大家好：按照ESP_houwenxiang提供的模板，我已搞定android wifi wave capture-display, 2Msps 4channel & 2Msps 2channel action perfectly, 但是 2Msps single channel 有小幅锯齿波噪声叠加 :mrgreen: ，不久前我在Arduino IDE中以i2s模式进行AD单通道采样时------.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN), 其sample rate 仅能达到78.125Ksps------也遭遇到小幅锯齿波噪声叠加，疑似只要进行AD单通道采样时就有小幅锯齿波噪声叠加，与采样速率无关，多通道scan就正常。我怀疑单通道AD采样值被i2s模块切成左右声道，输出时前后顺序错位，i2s改为单声道可能就OK了。另外小幅锯齿波噪声只有用信号发生器输入正弦波时才能看到，直流信号无锯齿，这更加说明AD数据输出前后顺序可能错位。虽然我对目前的结果已十分满意，毕竟双通道1Msps/channel 的速率省掉了两片AD7476的开销，但还是希望乐鑫能够核查并解决此bug, 圆我一个2Msps/channel的小梦想！！`
原文链接：https://www.esp32.com/viewtopic.php?f=25&t=8335&sid=aa79cd7e15e2730069eebb98d59349e2&start=10

为何ESP32采集正弦波会叠加一个高频噪声，这是因为ESP32采用I2S采集ADC数据，`单通道AD采样值被i2s模块切成左右声道，输出时前后顺序错位，i2s改为单声道可能就OK了`  我就是采用的ADC单通道采样，所以遇到了该问题。

通过将数据前后位置调换，我得到了最新ESP32采样波形图：
[image]
[image]

可以看到 以前的高频噪声消失了。

closed 2025-01-14 by Alvin1Zhang