espressif/esp-idf #10059: Target is already running an algorithm

vdm97, 2022-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2, v5.0-beta1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

A successful flash via OpebOCD/JTAG.

### What is the actual behavior?

My company uses a custom board where the ESP32-WROOM-32 with 16MB flash (ESP32) is used as a WiFi chip. For that purpose it works very well, so thank you very much for providing the chip and the good tools! We use the [VS Code ESP extension](https://github.com/espressif/vscode-esp-idf-extension) in conjunction with the ESP-Prog Board (JTAG) which works (most of the time) quite good. Furthermore the ESP32 is connected to a STM32 and UART is used for communication. The STM32 can also be used as a last hope to [flash](https://github.com/espressif/esp-serial-flasher) the ESP32.

Now about the problem. We wanted to expand the usage of the ESP32 by adding Bluetooth Low Energy (BLE) functionality. For that we activated Bluetooth in the menuconfig and selected the NimBLE stack. This did not work with the error described below. After some searching we found [this ticket](https://github.com/espressif/openocd-esp32/issues/223) with a similar error in the OpenOCD log and updated our ESP-IDF from v4.4.1 to v4.4.2 and then to v5.0-beta1 (After realising that the fix is not part of v4.4.2.). After this BLE worked fine on its own and with WiFi coexistence but the WiFi stopped working with timeout issues (SPIFFS could not be served). To fix the issue we tried different task priorities for WiFi and BLE and varying BLE window and interval settings but without success. In the end we opened the menuconfig and changed the BLE stack to Bluedroid. With this stack BLE and WiFi works in coexistence but our initial error appeared again. The OpenOCD log where the error appears is attached below. 

We do not know what the real cause of the error is but it appears on ESP-IDF v4.4.1 and v4.4.2 when we activate Bluetooth and select the NimBLE or Bluedroid stack. After we switched to ESP-IDF v5.0-beta1 the error only appears when we select the Bluedroid stack but does not appear when we select the NimBLE stack (We are ignoring the NimBLE/WiFi coexistence error/misconfiguration? here.). To fit a BLE stack on the ESP we had to deactivate all three WiFi IRAM optimizations, so maybe this plays a role in the error. Furthermore the IRAM and DRAM of the ESP32 is quite full so maybe this plays also a role.

Interestingly the ESP32 can always be flashed with the same binary (BLE enabled with Bluedroid stack) when the STM32 is used for serial flashing and the code is working fine if we look at the monitor and connect via WiFi/BLE but the same binary can only be be flashed the first time when OpenOCD/JTAG is used for flashing. On the second flashing the error in OpenOCD/JTAG appears, so BLE/Bluedroid code has to be present on the ESP32 for the error to appear.

All help is appreciated and we are happy to provide more information if needed. 

### Steps to reproduce.

I cannot provide our closed source firmware of the ESP32 and building an example is not easy because we have a lot of parts and i am not really sure where the error is coming from. I will take the time to do that if it will not be possible to get help otherwise.

### Build or installation Logs.

```plain
<details>
<summary>Image size</summary>
<br>
Total sizes:
Used static DRAM:   90236 bytes (  34344 remain, 72.4% used)
      .data size:   23316 bytes
      .bss  size:   66920 bytes
Used static IRAM:  121086 bytes (   9986 remain, 92.4% used)
      .text size:  120059 bytes
   .vectors size:    1027 bytes
Used Flash size : 1407143 bytes
      .text     : 1126587 bytes
      .rodata   :  280300 bytes
Total image size: 1551545 bytes (.bin may be padded larger) 
</details>

<details>
<summary>OpenOCD (The VS Code extension is responsible for the strange formatting...)</summary>
<br>
`
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
adapter speed: 20000 kHz

[OpenOCD]
Info : Listening on port 6666 for tcl connections
[OpenOCD]
Info : Listening on port 4444 for telnet connections
[OpenOCD]
Info : ftdi: if[OpenOCD]
 you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
[OpenOCD]
Info : JT[OpenOCD]
AG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : [esp32.cpu0] [OpenOCD]
Target halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : start[OpenOCD]
ing gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
[OpenOCD]
Flashing c:/svn/P[OpenOCD]
_ARM_CASSY2/ESP32/build/bootloader/bootloader.bin at 0x1000
[OpenOCD]
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e[OpenOCD]
5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : [esp32.cpu0] requesting target halt and executing a soft reset
[OpenOCD]
Info : [esp32.cpu0] Deb[OpenOCD]
ug controller was reset.
[OpenOCD]
Info : [esp3[OpenOCD]
2.cpu0] Core was reset.
[OpenOCD]
Info : [esp32.cpu0] Targ[OpenOCD]
et halted, PC=0x500000CF, debug_reason=00000000
[OpenOCD]
Info : Set GDB target to 'es[OpenOCD]
p32.cpu0'
[OpenOCD]
Info : [esp32.cpu1] [OpenOCD]
requesting target halt and executing a soft reset
[OpenOCD]
Info : [esp32.cpu0] Core was reset.
[OpenOCD]
Info : [esp32.cpu0] Target[OpenOCD]
 halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : [[OpenOCD]
esp32.cpu1] Debug controller was reset.
[OpenOCD]
Info : [es[OpenOCD]
p32.cpu1] Core was reset.
[OpenOCD]
Info : [es[OpenOCD]
p32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : [esp32.cpu[OpenOCD]
0] Target halted, PC=0x40092612, debug_reason=00000001
[OpenOCD]
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 266 [OpenOCD]
KB
Info : Flash mapping 1: 0x60020 -> 0x400d0020, 1105 KB
[OpenOCD]
Info : [esp3[OpenOCD]
2.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
[OpenOCD]
Info : Auto-[OpenOCD]
detected flash bank 'esp32.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32.cpu0.flash' size 16384 KB
[OpenOCD]
Info : [esp3[OpenOCD]
2.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
[OpenOCD]
Info : Flash mapping 0: 0x10020 -> 0x3f40002[OpenOCD]
0, 266 KB
Info : Flash mapping 1: 0x60020 -> 0x400d0020, 1105 KB
[OpenOCD]
Info : [esp32.cpu[OpenOCD]
0] Target halted, PC=0x40092612, debug_reason=00000001
[OpenOCD]
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 16384 KB
Info : Using[OpenOCD]
 flash bank 'esp32.cpu1.flash' size 16384 KB
[OpenOCD]
** Progr[OpenOCD]
amming Started **
[OpenOCD]
❌ Error: timed out [OpenOCD]
while waiting for target halted
[OpenOCD]
Info : [esp32.cpu0] Target halted, PC[OpenOCD]
=0x40082343, debug_reason=00000000
[OpenOCD]
❌ Error: xtensa_wait_algorithm: not halted [OpenOCD]
❌ 0, pc 0x40082343, ps 0x60423
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
[OpenOCD]
❌ Error: Ta[OpenOCD]
❌ rget is already running an algorithm
Error: Failed to start algorithm (-4)!
[OpenOCD]
Info : PROF: Erased [OpenOCD]
28672 bytes in 66.962 ms
[OpenOCD]
Err[OpenOCD]
❌ or: Target is already running an algorithm
Error: Failed to start algorithm (-4)!
[OpenOCD]
Info : PROF: Wrote 286[OpenOCD]
72 bytes in 73.221 ms (data transfer time included)
[OpenOCD]
** Program[OpenOCD]
ming Finished in 40669 ms **
[OpenOCD]
** Verify[OpenOCD]
 Started **
[OpenOCD]
❌ Error: Target [OpenOCD]
❌ is already running an algorithm
Error: Failed to start algorithm (-4)!
[OpenOCD]
Info : PROF: Flash verified in 70[OpenOCD]
❌ .972 ms 
Error: **** Verification failure! ****
Error: target_hash 000...000
Error: file_hash: 0d6ef...722da1
[OpenOCD]
** Flashing[OpenOCD]
 Failed **
-1
`
</details>
```

vdm97, 2022-10-26:
Body was too long...
<details>
<summary>sdkconfig</summary>
<br>
`
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=3
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_CLASSIC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_RTC_SLOW_CLOCK_SUPPORT_8MD256=y
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_SIZE=64
CONFIG_SOC_DAC_PERIPH_NUM=2
CONFIG_SOC_DAC_RESOLUTION=8
CONFIG_SOC_GPIO_PORT=1
CONFIG_SOC_GPIO_PIN_COUNT=40
CONFIG_SOC_GPIO_VALID_GPIO_MASK=0xFFFFFFFFFF
CONFIG_SOC_GPIO_SUPPORT_SLP_SWITCH=y
[... 2028 lines cut ...]
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
`
</details>

chipweinberger, 2022-10-29:
there is a good debugging doc in the esp-idf vscode extension repo, if you haven't found it.

isthereafreeusername, 2022-11-02:
Hi there,
i think i got the same problem

I already posted at the ESP32.com forum:
https://www.esp32.com/viewtopic.php?f=14&t=30217

vdm97, 2022-11-02:
> Hi there, i think i got the same problem
> 
> I already posted at the ESP32.com forum: https://www.esp32.com/viewtopic.php?f=14&t=30217

Hi, the specific error code is different (yours is `esp_algorithm.c:225 algorithm_run(): Failed to wait algorithm (-302)!`, mine is `esp_algorithm.c:203 algorithm_run(): Failed to start algorithm (-4)!`) but the general error image is very similar indeed. The error appears when we add extra code and the error breaks JTAG.

> there is a good debugging doc in the esp-idf vscode extension repo, if you haven't found it.

I investigated the error with more detailed debug output, but i lack the neccessary JTAG/OpenOCD knowlegde to find a solution. I added the debug output of ESP-IDFv5.0-beta1 for someone who has the neccessary knowledge.

<details>
<summary>ESP-IDFv5.0-beta1 OpenOCD log</summary>
<br>

```
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
User : 3 6 opt[OpenOCD]
ions.c:63 configuration_output_handler(): debug_level: 3
User : 4 6 options.c:63 configuration_output_handler(): 
Debug: 5 6 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 7 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../op[OpenOCD]
enocd-esp32/share/openocd
Debug: 7 7 options.c:246 add_default_dirs(): exepath=c:/Users/xxx/esp/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
Debug: 8 7 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 9 7 configuration.c:44 add_script_search_dir(): adding c:\Users\xxx\esp\esp-idf-tools\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts
Debug: 10 8 configuration.c:44 add_script_search_dir():[OpenOCD]
 adding C:/Users/xxx/AppData/Roaming/OpenOCD
Debug: 11 8 configuration.c:44 add_script_search_dir(): adding C:\Users\xxx/.config/openocd
Debug: 12 8 configuration.c:44 add_script_search_dir(): adding C:\Users\xxx/.openocd
Debug: 13 8 configuration.c:44 add_script_search_dir(): adding c:/Users/xxx/esp/es[OpenOCD]
p-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/site
Debug: 14 9 configuration.c:44 add_script_search_dir(): adding c:/Users/vde[OpenOCD]
derer/esp/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin/../share/openocd/scripts
Debu[OpenOCD]
g: 15 9 command.c:166 script_debug(): command - ocd_find interface/ftdi/esp32_devkitj_v1.cfg
[OpenOCD]
Debug: 16 10 configuration.c:99 find_file(): found c:\Users\xxx\esp\esp-idf-tools\tools\openocd-esp32\v0.11.0-e[OpenOCD]
sp32-20220706/openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg
[OpenOCD]
Debug: 17[OpenOCD]
 12 command.c:166 script_debug(): command - adapter driver ftdi
Debug: 18 13 command.c:166 script_debug(): command - ftdi vid_pid 0x0403 0x6010 [OpenOCD]
0x0403 0x6014
Debug: 19 13 command.c:166 script_debug(): command - ftdi channel 0
Debug: 20 13 command.c:166 script_debug(): command - ftdi layout[OpenOCD]
_init 0x0008 0xf00b
Debug: 21 14 command.c:166 script_debug(): command - ftdi layout_signal LED -data 0x1000
Debug: 22 14 command.c:166 script_debug(): command - ftdi layout_signal LED2 -data 0x2000
Debug: 23 14 c[OpenOCD]
ommand.c:166 script_debug(): command - ftdi layout_signal LED3 -data 0x4000
Debug: 24 14 command.c:166 script_debug(): command - ftdi layout_signal LED4 -data 0x8000
Debug: 25 14 command.c:166 script_debug(): comman[OpenOCD]
d - adapter speed 20000
Debug: 26 15 adapter.c:180 adapter_config_khz(): han[OpenOCD]
dle adapter khz
Debug: 27 15 adapter.c:144 adapter_khz_to_speed(): convert khz to adapter specific spee[OpenOCD]
d value
Debug: 28 15 adapter.c:144 adapter_khz_to_speed(): convert khz to adapter specific speed value
User : 29 15 options.c:63 configuration_output_handler(): adapter [OpenOCD]
speed: 20000 kHz
User : 30 15 options.c:63 configuration_output_handler(): 
Debug: 31 15 command.c:166 script_debug(): command - ocd_find target/esp32.cfg
Debug: 32 15 configuration.c:99[OpenOCD]
 find_file(): found c:\Users\xxx\esp\esp-idf-tools\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts/target/esp32.cfg
[OpenOCD]
[... 2976 lines cut ...]
Debug: 1904 42215 xtensa.c:833 xtensa_core_status_check(): [esp32.cpu0] DSR (8080CC11)
Debug: 1905 42215 target.c:2471 target_write_buffer(): writing buffer of 512 byte at 0x40090e00
[OpenOCD]
Debug: 1906[OpenOCD]
 42217 xtensa.c:833 xtensa_core_status_check(): [esp32.cpu0] DSR (8080CC11)
Debug: 1907 42218 target.c:2471 target_write_buffer(): writing buffer of 512 byte at 0x40091000
[OpenOCD]
Debug: 1908 [OpenOCD]
42223 xtensa.c:833 xtensa_core_status_check(): [esp32.cpu0] DSR (8080CC11)
Debug: 1909 42223 target.c:2471 target_write_buffer(): writing buffer of 512 byte at 0x40091200
[OpenOCD]
Debug: 191[OpenOCD]
0 42226 xtensa.c:833 xtensa_core_status_check(): [esp32.cpu0] DSR (8080CC11)
Debug: 1911 42226 target.c:2471 target_write_buffer(): writing buffer of 512 byte at 0x40091400
[OpenOCD]
Debug: [OpenOCD]
1912 42229 xtensa.c:833 xtensa_core_status_check(): [esp32.cpu0] DSR (8080CC11)
Debug: 1913 42230 target.c:2471 target_write_buffer(): writing buffer of 512 byte at 0x40091600
[OpenOCD]
Debug: 1914 42233[OpenOCD]
 xtensa.c:833 xtensa_core_status_check(): [esp32.cpu0] DSR (8080CC11)
Debug: 1915 42233 target.c:2471 target_write_buffer(): writing buffer of 512 byte at 0x40091800
[OpenOCD]
Debug: 1916 42237 xt[OpenOCD]
ensa.c:833 xtensa_core_status

chipweinberger, 2022-11-02:
here is the doc I was referring to:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

vdm97, 2022-11-03:
> here is the doc I was referring to:
> 
> https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

Thank you for the link but i do not get how this should help me. The error is not present when Bluetooth is deactivated and only appears when i activate Bluetooth so i do not think that it is a VSCode ESP IDF extension / OpenOCD misconficuration. I could be wrong of course..

gojimmypi, 2023-01-29:
Yesterday I encountered a [similar JTAG problem](https://github.com/golioth/golioth-firmware-sdk/issues/77#issuecomment-1407563157). 

Does anyone else think it might be related to the `CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y` setting? Given that I _also_ removed the Bluetooth settings, I suppose it could also be a memory issue? 

I'm not sure I've ever encountered a low memory issue causing JTAG programming problems. Is there a device RAM overhead for JTAG programming and/or single-step debugging?

@vdm97 I'd be interested in seeing your `sdkconfig` settings if possible.

vdm97, 2023-01-30:
Hi @gojimmypi,

my sdkconfig is posted above:

> Body was too long...
> [sdkconfig](https://github.com/espressif/esp-idf/issues/10059#issuecomment-1291744454)

PS: I do not use the `CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE `setting.

vdm97, 2023-05-24:
Today i encountered this error again. After updating from IDF version 4.4.3 to version 5.0.2 my IRAM usage was at 92,7 % and i could not flash my ESP32-WROOM-32 a second time anymore. So i started to search for a solution and found this [issue](https://github.com/espressif/openocd-esp32/issues/289) but updating OpenOCD to the newest release did not help. Therefore i removed the last configurable piece of code from the IRAM (CONFIG_LWIP_IRAM_OPTIMIZATION) which reduced the IRAM usage to 82.2 %. After a manual flash via serial flashing i could flash again with the ESP-IDF VS Code Extension like before. So unfortunately the error is still present...

chconnor, 2023-12-09:
Thanks for your investigations. You mentioned the heap size... could it possibly be related to written flash size? (I'm a total beginner, so I'm just brainstorming here.)

I'm seeing similar -- my procedure is to switch to a smaller project and do a non-debug upload. Then I can go back to the "large" project and do a debug upload and it works, but only one time. Then the board is "broken" for debugging until I repeat the procedure.

I posted at the PIO forum with some more details: https://community.platformio.org/t/interesting-results-re-chronic-esp-prog-problems/37168

Relevant quote:

> You have to switch to a “small” project, and you have to do a non-debug upload to fix it. Trying to do a non-debug upload of the “large” project will not fix it. Trying to do a debug upload of a “small” project will not work, and will not fix it.
> 
> My “large” project uses about 79% of the flash. Successful debug logs show that it’s uploading 1044480 bytes. Again, this only works one time, then it is “broken” until the procedure above is followed.
> 
> A “small” project can be debugged over and over without issue.
> 

It's interesting to me that your flash usage is right around where mine is... I've been incrementally building this project, and it just started to fail, so if in fact there is a firm threshold of image size that triggers the issue, I would expect it to be near the size I'm currently failing at (1044480.)

erhankur, 2023-12-09:
Sorry, I just saw this long-time-opened issue. 

OpenOCD uses the IRAM memory `0x3FFE0000` to `0x3FFF8000` for the working area meaning that it is a region in the SRAM for the stub flasher code and data.
https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp32.cfg#L32
This region is in use by OpenOCD, when you want to flash applications or use flash breakpoints. Before running the flasher stub, OpenOCD will read this region, save it, run the flasher stub application, and restore it.

As I understand, when your application IRAM usage is high, you can not update your application from the JTAG. Correct? 

I can try to create a test app, to reproduce the issue by allocating a high chunk of memory from the heap. Or maybe someone can do it by simulating the application memory usage? Also, are any other settings needed to see the failure?

I am sorry to hear that this issue is called a 'chronic' in the community. I will do my best to find a solution.

chconnor, 2023-12-09:
Thanks! I have been working this morning to narrow things down. Unfortunately I am a beginner and barely know C or microcontrollers (e.g. I barely understand what IRAM even is) but I will help if I can!

Here is what I found today (doing this on an ESP32-DevKitC-VIE which uses WROVER):

Suspecting that the issue was written flash size, I made a program that #include's "bigfile.h" where bigfile is like this:

```
const long arrsize = 260000;
const int bigarray[260000] = {
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
...
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, };

```
(they are ints and I mistakenly assigned bytes, but I assume that's irrelevant)

And main.cpp was:

```
#include <Arduino.h>
#include <bigfile.cpp>

void setup()
{
  Serial.begin(115200);
  while (!Serial) delay(10);
  srand(time(NULL));
  Serial.println(bigarray[rand()%arrsize]);
}

void loop() {;}

```
...even though flash was 99.5% used, this debugged fine.

So instead I used a python script to generate ../include/bigfile.cpp which is like this:

```
#pragma GCC push_options
#pragma GCC optimize("O0")
static volatile int i = 0;
void func0(int newv);
void func0(int newv) { volatile int a = 0; a += 1; i = a; }
void func1(int newv);
void func1(int newv) { volatile int a = 0; a += 1; i = a; func0(a); }
void func2(int newv);
void func2(int newv) { volatile int a = 0; a += 1; i = a; func1(a); }
void func3(int newv);
void func3(int newv) { volatile int a = 0; a += 1; i = a; func2(a); }
...
void func124(int newv) { volatile int a = 0; a += 1; i = a; func123(a); }
#pragma GCC pop_options

```
And main is like this:

```
#include <Arduino.h>
#include <bigfile.cpp>

void setup()
{
  Serial.begin(115200);
  while (!Serial) delay(10);
  func124(0);
}

void loop() {;}
```

[... 700 lines cut ...]
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu0 examination failed
Error: Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu1 examination failed
Info : starting gdb server for esp32.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Error: Target not examined yet
Error executing event gdb-attach on target esp32.cpu0:

Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
.pioinit:11: Error in sourced command file:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
</details>