espressif/esp-idf #10060: ESP32S3, double flash setup hangs in esp_flash_erase_region(), function rom_spiflash_api_funcs->start(chip)

marbalon, 2022-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-427-g4532e6e0b2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Continue erasing with external flash. 

### What is the actual behavior?

I have a project with ESP32S3 + LCD 800x480 over RGB + 8MByte PSRAM. ESP-IDF 5.0. 

Because I have a problem with refreshing LCD described here https://github.com/espressif/esp-idf/issues/10010

I've decided to add a second flash to my setup and it is W25Q128JVPIM connected to SPI3 - so different than main SPI chip and PSRAM. The problem I'm having now is random hang in the function esp_flash_erase_region() in lines:

```
esp_rom_printf("==> esp_flash_erase_region start: %d\n\r", __LINE__);
err = rom_spiflash_api_funcs->start(chip); //somtimes this function never return
esp_rom_printf("==> esp_flash_erase_region: %d\n\r", __LINE__);        
if (err != ESP_OK) {
      break;
}
bus_acquired = true;
```

My application does not use any WiFi/Bluetooth functions, Just UART communication and LCD. What is strange is that it hangs in one task that is kind of a garbage collector for my custom file system, but another task can still read images from this flash.

I saw this issue https://github.com/espressif/esp-idf/issues/9332 but seems to not be related.

### Steps to reproduce.

Nothing special here.

marbalon, 2022-10-26:
One more question, because I haven't found the answer to this.

If I'm reading/writing to the same partition from two tasks should I use any mutex/semaphore for this? Or it is sorted out by the partition/SPI flash API ?

I'm asking because I've just found that my task also hangs on function esp_partition_read().

Strange thing is that it was working on the main flash without the problem, the problem is only when we have two chips configuration.

weilian1977, 2023-01-26:
关注一下