espressif/esp-idf #10062: esp_ota_ops: OTA image has invalid magic byte -->  Guru Meditation Error:

sergioaiello, 2022-10-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-CAM

### Power Supply used.

USB

### What is the expected behavior?

ESP32-CAM hosts the STA and HTTP client
ESP32 v1 hosts AP and HTTP server

ESP32-CAM requests an OTA download. If HTTP server sends a bin standard file, then the OTA update completes successfully.
Viceversa, if the server sends a different than bin std file, then the OTA procedure fails 

### What is the actual behavior?

ESP32-CAM requests an OTA download. If HTTP server sends a bin standard file, then the OTA update completes successfully.
Viceversa, if the server sends a different than bin std file, then the OTA procedure fails (and that is correct) but this also triggers a **crash of the ESP32-CAM**.
Two questions:
- Is this problem already known? (apologies, since I haven't found a similar case)
- Did someone verify the scenario described above? In that case, what was the result of the test?

### Steps to reproduce.

- HTTP server configured via HTML for "STA/HTTP client" OTA procedure
- as HTTP client polls the server it receives the OTA file to download
- HTTP client initiates the OTA downloading the file and trying the update

### Debug Logs.

```plain
****************** from ESP32-CAM monitor window *******************************
E (56821) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x23)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40178453  PS      : 0x00060730  A0      : 0x8010d874  A1      : 0x3ffe5bd0  
0x40178453: ssl_load_six_bytes at C:/esp/esp-idf-44/components/mbedtls/mbedtls/library/ssl_tls.c:3870
 (inlined by) mbedtls_ssl_dtls_replay_check at C:/esp/esp-idf-44/components/mbedtls/mbedtls/library/ssl_tls.c:3883

A2      : 0xaf01a734  A3      : 0x00000000  A4      : 0x00000030  A5      : 0x3ffcc334
A6      : 0x3ffe58d0  A7      : 0x00000003  A8      : 0x3ffe5d10  A9      : 0x3ffe5cf0
A10     : 0x3ffbe188  A11     : 0x3ffbdd1c  A12     : 0x3ffbe188  A13     : 0x00000000  
A14     : 0x3ffe5cf0  A15     : 0x0000000c  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xaf01a7a8  LBEG    : 0x4008b722  LEND    : 0x4008b72d  LCOUNT  : 0x00000000  
0x4008b722: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x4008b72d: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x40178450:0x3ffe5bd00x4010d871:0x3ffe5bf0 0x4010d261:0x3ffe5c10 0x4015fd4c:0x3ffe5c30 0x4017a7b9:0x3ffe5c80 0x401605ec:0x3ffe5ca0 0x4017a775:0x3ffe5cd0 0x4010e55a:0x3ffe5cf0 0x4010fe0b:0x3ffe5d10 
0x400dbc9e:0x3ffe5d30 0x400dc20f:0x3ffe6b10 0x40091af1:0x3ffe6b30
0x40178450: ssl_load_six_bytes at C:/esp/esp-idf-44/components/mbedtls/mbedtls/library/ssl_tls.c:3870
 (inlined by) mbedtls_ssl_dtls_replay_check at C:/esp/esp-idf-44/components/mbedtls/mbedtls/library/ssl_tls.c:3883

0x4010d871: esp_mbedtls_verify_certificate at C:/esp/esp-idf-44/components/esp-tls/esp_tls_mbedtls.c:269 (discriminator 13)

0x4010d261: tcp_connect at C:/esp/esp-idf-44/components/esp-tls/esp_tls.c:332
 (inlined by) esp_tls_plain_tcp_connect at C:/esp/esp-idf-44/components/esp-tls/esp_tls.c:446

0x4015fd4c: esp_transport_ssl_init at C:/esp/esp-idf-44/components/tcp_transport/transport_ssl.c:440

0x4017a7b9: http_parse_host_char at C:/esp/esp-idf-44/components/nghttp/port/http_parser.c:2261

0x401605ec: ssl_get_context_data at C:/esp/esp-idf-44/components/tcp_transport/transport_ssl.c:64
 (inlined by) esp_tls_connect_async at C:/esp/esp-idf-44/components/tcp_transport/transport_ssl.c:71

0x4017a775: http_parse_host_char at C:/esp/esp-idf-44/components/nghttp/port/http_parser.c:2217

0x4010e55a: http_client_prepare_first_line at C:/esp/esp-idf-44/components/esp_http_client/esp_http_client.c:1222

0x4010fe0b: esp_http_check_response at C:/esp/esp-idf-44/components/esp_http_client/esp_http_client.c:787 (discriminator 3)

0x400dbc9e: http_rest_with_url at C:\ToDoDelay\build/../main/take_picture.c:750 (discriminator 13)

0x400dc20f: http_rest_with_url at C:\ToDoDelay\build/../main/take_picture.c:1026 (discriminator 3)

0x40091af1: vPortTaskWrapper at C:/esp/esp-idf-44/components/freertos/port/xtensa/port.c:130
```

sergioaiello, 2022-10-27:
a solution that works for me:
- removed the line "esp_http_client_cleanup(evt->client);" from "chunked file receiving loop"
In this way the HTTP client completes the file receiving (although the "OTA image has invalid magic byte") and closes gracefully the HTTP connection
According to this workaround the issue is not related to OTA procedure and it seems to be related to HTTP FSM

hmalpani, 2022-10-27:
Hello @sergioaiello 
Can you check [Native OTA example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/native_ota_example). 

It do not crash after error: `esp_ota_ops: OTA image has invalid magic bytes`

sergioaiello, 2022-10-27:
Hi hmalpani,
thank you for the hint.
I started from that example, indeed.
After, I needed to modify the task.
Therefore I needed to remove the line "task_fatal_error();" and I imagine that step is critical for the mattering issue. Isn't it?

hmalpani, 2022-10-27:
task_fatal_error() keeps the task in infinite loop and does nothing. I think because you removed `task_fatal_error()`, you might still be reading from the server or using some variable which was freed and that is causing the `Guru Meditation Error`.

sergioaiello, 2022-10-27:
Hi hmalpani,
yes, the task continues, indeed.
The ESP32 client is supposed to send (to the file server) the OTA procedure result.
i think of the following two options:
- remove the client cleanup line and also the esp_ota_write (considering the routine esp_ota_set_boot_partition will warn anyway about the wrong format of the file)
- optimize the code adding a task as in the example
In both cases there is nothing related to a ESP-IDF bug and I propose to close this thread. Right?

hmalpani, 2022-10-28:
I suggest to close the client connection and exit the ota process. You must be trying to read using a closed connection.

> In both cases there is nothing related to a ESP-IDF bug and I propose to close this thread. Right?

Yes you can close the issue as it does not seem to be a bug in ESP-IDF.

closed 2022-10-30 by sergioaiello