espressif/esp-idf #10063: Console Example

keenanjohnson, 2022-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to run the [ESP-IDF console example here.](https://github.com/espressif/esp-idf/tree/v4.4.1/examples/system/console/basic)

However, when I run the example, I get the following output via idf.py monitor.

I can see any results from my input and the problem seems to be described via the message "Your Terminal application does not support escape sequences."

I'm now running OSX Ventura 13.0.

I'm using this esp32s3 board from adafruit. https://www.adafruit.com/product/5477. The board is connected directly to my macbook via a USBC Cable. 

```
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
```

keenanjohnson, 2022-10-27:
I tried enabling the CONFIG_ESP_CONSOLE_USB_CDC as described in this [changeset](https://github.com/espressif/esp-idf/commit/3390d2a2d18c289b0b1e1cf121bf6670099641db), but when I do, I get the following compilation error: 

```
-- Found Git: /usr/bin/git (found version "2.37.0 (Apple Git-136)") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/keenan/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/keenan/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/keenan/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /Users/keenan/esp/esp-idf/examples/system/console/basic/sdkconfig
Loading defaults file /Users/keenan/esp/esp-idf/examples/system/console/basic/sdkconfig.defaults...
-- Found Python3: /opt/homebrew/Frameworks/Python.framework/Versions/3.10/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "console" version: v4.4.1
-- Adding linker script /Users/keenan/esp/esp-idf/examples/system/console/basic/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
-- Adding linker script /Users/keenan/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmd_nvs cmd_system cmock coap console cxx driver efuse esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/keenan/esp/esp-idf/components/app_trace /Users/keenan/esp/esp-idf/components/app_update /Users/keenan/esp/esp-idf/components/asio /Users/keenan/esp/esp-idf/components/bootloader /Users/keenan/esp/esp-idf/components/bootloader_support /Users/keenan/esp/esp-idf/components/bt /Users/keenan/esp/esp-idf/components/cbor /Users/keenan/esp/esp-idf/examples/system/console/advanced/components/cmd_nvs /Users/keenan/esp/esp-idf/examples/system/console/advanced/components/cmd_system /Users/keenan/esp/esp-idf/components/cmock /Users/keenan/esp/esp-idf/components/coap /Users/keenan/esp/esp-idf/components/console /Users/keenan/esp/esp-idf/components/cxx /Users/keenan/esp/esp-idf/components/driver /Users/keenan/esp/esp-idf/components/efuse /Users/keenan/esp/esp-idf/components/esp-tls /Users/keenan/esp/esp-idf/components/esp32s3 /Users/keenan/esp/esp-idf/components/esp_adc_cal /Users/keenan/esp/esp-idf/components/esp_common /Users/keenan/esp/esp-idf/components/esp_eth /Users/keenan/esp/esp-idf/components/esp_event /Users/keenan/esp/esp-idf/components/esp_gdbstub /Users/keenan/esp/esp-idf/components/esp_hid /Users/keenan/esp/esp-idf/components/esp_http_client /Users/keenan/esp/esp-idf/components/esp_http_server /Users/keenan/esp/esp-idf/components/esp_https_ota /Users/keenan/esp/esp-idf/components/esp_https_server /Users/keenan/esp/esp-idf/components/esp_hw_support /Users/keenan/esp/esp-idf/components/esp_ipc /Users/keenan/esp/esp-idf/components/esp_lcd /Users/keenan/esp/esp-idf/components/esp_local_ctrl /Users/keenan/esp/esp-idf/components/esp_netif /Users/keenan/esp/esp-idf/components/esp_phy /Users/keenan/esp/esp-idf/components/esp_pm /Users/keenan/esp/esp-idf/components/esp_ringbuf /Users/keenan/esp/esp-idf/components/esp_rom /Users/keenan/esp/esp-idf/components/esp_serial_slave_link /Users/keenan/esp/esp-idf/components/esp_system /Users/keenan/esp/esp-idf/components/esp_timer /Users/keenan/esp/esp-idf/components/esp_websocket_client /Users/keenan/esp/esp-idf/components/esp_wifi /U [... cut ...]
[101/101] cd /Users/keenan/esp/esp-idf/examples/system/console/basic/build/bootloader/esp-idf/esptool_py && /Users/keenan/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/keenan/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/keenan/esp/esp-idf/examples/system/console/basic/build/bootloader/bootloader.bin
Bootloader binary size 0x49e0 bytes. 0x3620 bytes (42%) free.
ninja: build stopped: subcommand failed.

 *  The terminal process "/bin/zsh '-c', 'ninja '" terminated with exit code: 1. 
```

chipweinberger, 2022-10-29:
does it work in visual studio code or other terminals?

beriberikix, 2022-10-29:
Running into the same issue. I'm on an x86 macOS Monterey 12.6, running bash. I've tried `idf.py monitor`, `minicom` and `screen`. I don't typically use macOS and have used the shell + `idf.py monitor` many times on Linux.

keenanjohnson, 2022-10-30:
@chipweinberger it does not on any terminal on my machine (zsh, visual studio code, etc)

beriberikix, 2022-10-30:
I've now tested with with v4.4.1, v4.4.2, release/v5.0 & master. Originally I tested on Monterey but I upgrade to Ventura to match @keenanjohnson.

On master/x86/Ventura here's my test results:
* ESP32-S3 Feather - has the "Your terminal application does not support escape sequences."
* ESP32-C3-DevKitC-02 - works correctly
* ESP32-DevKitC - works correctly

So either there's an issue with ESP32-S3s (maybe protocol related?,) HW-related to the ESP32-S3 Feather design, my device is malfunctioning or user error. My intuition says it's S3 more broadly but I unfortunately don't have another S3 (Feather or otherwise) to test.

beriberikix, 2022-10-30:
That is indeed the case for the [ESP32-S3 Feather](https://learn.adafruit.com/adafruit-esp32-s3-feather/downloads) and it looks like the USB Serial JTAG is the fix! Will do more testing but looks promising.

chipweinberger, 2022-10-30:
@igrr 

This issue is a good example why the console should just auto-detect USB Serial/JTAG, and do the right thing by default. 

My PRs attempt to do that, but more work is needed.

The USB Serial/JTAG Console should be "idiot-proof", heh.

keenanjohnson, 2022-10-31:
Thanks all! This is very helpful!

I understand how to set the config option via the SDK File, but is there a guide for configuring that setting in VSCode using the ESP-IDF extension? In the "Flash Method" setting, would I select the JTAG option>? I'm trying to set up our project to be as easy as possible for new people to get started with, so just trying to fill out our docs in the best way I can. I can ask in the vscode idf project directly if this isn't the correct place.

[image]

beriberikix, 2022-11-02:
There's a [GUI configurator](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/FEATURES.md#SDK-Configuration-editor), similar to `idf.py menuconfig` where these values can be set. They can also be set to a default value in `sdkconfig.defaults`, which is what we do in our [examples](https://github.com/golioth/golioth-esp-idf-sdk/blob/main/examples/golioth_basics/sdkconfig.defaults).

[image]

keenanjohnson, 2022-11-02:
Thanks @beriberikix !

ESP-Marius, 2023-02-27:
Closing this as the original issue seems to have been resolved. Feel free to reopen if there is still an outstanding issue.

closed 2023-02-27 by ESP-Marius