espressif/esp-idf #10064: Any full example about Ble Mesh Sub/Pub

jiabuda, 2022-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, we are making a smart toy that could battle among five devices with ble mesh network. So we decide five board with two button on each on them. 

The game host press host button to enable provisioner mode and host the game, the guest peoples press the guest button to enable node mode. The guest peoples finish the game then upload the game score to the host board, the host board will rank the 5 score and mark winner, then send back the rank to all the device. The host never know when the guest finish the game, so host could not call GET_STATUS, but just wait for the guest board use `esp_ble_mesh_server_model_send_msg` to send the score, and this is already work. But we fail at publish the final rank to all the guest board. 

Provisioner definition:
```
static esp_ble_mesh_model_t provisioner_model[] = {
    // server model context
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_CFG_CLI(&config_client),
    ESP_BLE_MESH_MODEL_GEN_PROPERTY_CLI(&property_pub_0, &property_client),
};
```

Node definition:
```
static esp_ble_mesh_model_t node_model[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_GEN_ADMIN_PROP_SRV(&property_pub_1, &admin_property_server_0),
};
```

Now we found code about `esp_ble_mesh_model_subscribe_group_addr` in aligenie example, and we found the code about `esp_ble_mesh_model_publish` in ble mesh console example, yet we could not fully understand how to publish and subscribe to a group address. Those two example focus on two different environment.

So if it is any example about the full process of sub/pub usage? How can a client model publish data and which kind of event will be triggered on guest board's server model? Hope you could help us.

forx157, 2022-10-27:
Hi, @jiabuda 
There are some tips may help you.
-  opcode `ESP_BLE_MESH_MODEL_OP_MODEL_SUB_ADD` is used a config client model to set a config server's subscribe address (that address must be a valid group address)
-  `esp_ble_mesh_config_client_set_state` used to send a set msg to a config server node. For specific use, please refer to ble mesh console's `ble_mesh_reg_cfg_client_cmd.c`
- if client set success, there will be a ESP_BLE_MESH_CFG_CLIENT_SET_STATE_EVT event and opcode is `ESP_BLE_MESH_MODEL_OP_MODEL_SUB_ADD`. For specific use, please refer to ble mesh console's `ble_mesh_reg_cfg_client_cmd.c`

If every thing goes well. your host node could send rank msg to that group address as a general message which destination address is a unicast address.
Recv that msg also like a general message.

Hope it can help you.

jiabuda, 2022-10-28:
@forx157 worked, thanks a lot!

closed 2022-10-28 by jiabuda