espressif/esp-idf #10065: Espressif-IDE JTAG debugging (no device found)

krupis, 2022-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I have purchased ESP32-S3 Devkit (https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html ) and I am trying to setup debugging. 

I have setup Espressif-IDE and read some instructions on the internet (it seemed quite simple on how to get it debugger running).

I have imported a "hello_world" project, created a new Configuration:

Most settings were auto filled so I left them as it is just for now.

[image]

[image]

I plug the micro USB cable to USB port (not the UART since I want to use built in JTAG debugger on the ESP32-S3)

When I try to start debugging session, I get the following error:
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Flashing C:/Espressif/frameworks/esp-idf-v4.4.2/workspace/hello_world/build/partition_table/partition-table.bin at 0x8000
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'
** OpenOCD init failed **
shutdown command invoked
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Assertion failed!

Program: C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0

```

I have a feeling that the issue might be due to my Debugger configration. 
The selected Board is : 
```
ESP32-S3 chip (via ESP-PROG) 
```
but I dont think that is correct since I do not use ESP-PROG board. I know that the ESP32-S3 have built in jtag debugging capabilites and should not need ESP-PROG.
I read that here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#jtag-debugging-configuring-target

In my Config options, the following options are set:
```
-s ${openocd_path}/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32s3.cfg
```
I also think the above is not correct. Documentation ( https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#jtag-debugging-configuring-target ) suggests to set the target to:
```
openocd -f board/esp32s3-builtin.cfg
```
hence I have changed my Config options to :
```
-s ${openocd_path}/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32s3.cfg
```
Trying to debug with updated config option I get different error:
```
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/interface/esp_usb_jtag.cfg:7: Error: invalid command name "espusbjtag"
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/board/esp32s3-builtin.cfg", line 9
at file "C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts/interface/esp_usb_jtag.cfg", line 7
```

Could someone point me in the right direction and help me understand how to setup debugging for the ESP32-S3 devkit?

krupis, 2022-10-27:
**UPDATE**

I have solved the issue by changing the Config options:
```
-s ${openocd_path}/share/openocd/scripts -f interface/esp_usb_jtag.cfg -f board/esp32s3-builtin.cfg
```

I must selected the esp_usb_jtag.cfg and esp32s3-builtin.cfg . The dafault settings assume you are using ESP-PROG

beckmx, 2022-12-15:
@krupis did you reboot the module? is there any sequence on rebooting?

krupis, 2022-12-15:
> @krupis did you reboot the module? is there any sequence on rebooting?

Hey. Are you having issues yourself or trying to help me ? :) Since I have already solved this issue, I assume you are having some issues?

closed 2025-03-12 by Alvin1Zhang