espressif/esp-idf #10066: SPI master driver memory free heap assert failed

wishinlife, 2022-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.42

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3-N8R2

### Power Supply used.

USB

### What is the expected behavior?

I user WK2124 chip external uart, and start tow task, the first task control onboard uart, the second task control external uart on WK2124. The WK2124 is link on SPI3. The first round of sending and receiving data was normal,  and the chip was abort during the second round. WK2124 needs to be read multiple times in each poll. And there's another interesting thing, if I add some printf and some log to the function, it will work just fine.

### What is the actual behavior?

report memory free assert failed.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
assert failed: tlsf_free heap_tlsf.c:872 (!block_is_free(block) && "block already marked as free")

Backtrace:0x40375d6a:0x3fc9d6500x4037c139:0x3fc9d670 0x40383155:0x3fc9d690 0x4038246f:0x3fc9d7b0 0x40382d86:0x3fc9d7d0 0x40375fca:0x3fc9d7f0 0x40383185:0x3fc9d810 0x40377882:0x3fc9d830 0x40377d08:0x3fc9d850 0x40377d55:0x3fc9d880 0x420095cf:0x3fc9d8a0 0x4200977e:0x3fc9d8d0 0x42009ccf:0x3fc9d920 0x42009588:0x3fc9d950 0x42007d8e:0x3fc9d970 0x42008db4:0x3fc9d990 0x42007dc8:0x3fc9d9d0 0x420083cd:0x3fc9d9f0 0x420078c8:0x3fc9da10 0x4037f265:0x3fc9db40 
0x40375d6a: panic_abort at /home/zhuxy/esp/esp-idf/components/esp_system/panic.c:402
0x4037c139: esp_system_abort at /home/zhuxy/esp/esp-idf/components/esp_system/esp_system.c:128
0x40383155: __assert_func at /home/zhuxy/esp/esp-idf/components/newlib/assert.c:85
0x4038246f: tlsf_free at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:872 (discriminator 1)
0x40382d86: multi_heap_free_impl at /home/zhuxy/esp/esp-idf/components/heap/multi_heap.c:222
0x40375fca: heap_caps_free at /home/zhuxy/esp/esp-idf/components/heap/heap_caps.c:340
0x40383185: free at /home/zhuxy/esp/esp-idf/components/newlib/heap.c:39
0x40377882: uninstall_priv_desc at /home/zhuxy/esp/esp-idf/components/driver/spi_master.c:736
0x40377d08: spi_device_polling_end at /home/zhuxy/esp/esp-idf/components/driver/spi_master.c:983 (discriminator 15)
0x40377d55: spi_device_polling_transmit at /home/zhuxy/esp/esp-idf/components/driver/spi_master.c:1003
0x420095cf: wk_device_transmit at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/spi_wk2xxx.c:62
0x4200977e: wk_read_SFifo at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/spi_wk2xxx.c:195
0x42009ccf: wk_read_bytes at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/spi_wk2xxx.c:593
0x42009588: uart_port_read_bytes at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/uart_port.c:296
0x42007d8e: _modbus_rtu_recv at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus-rtu.c:204
0x42008db4: _modbus_receive_msg at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus.c:412
0x42007dc8: _modbus_rtu_receive at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus-rtu.c:191
0x420083cd: modbus_receive at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus.c:504
0x420078c8: modbus_rtu_task at /home/zhuxy/esp-projects/IoT-Station/main/src/modbus_slave.c:122
0x4037f265: vPortTaskWrapper at /home/zhuxy/esp/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 4be6f973b94b35ce

CPU halted.
```

### More Information.

**if swap internal uart and external wk2124 uart in task, the same first read is normal, the second times are abort, the log is as follows:**
`Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40382734  PS      : 0x00060833  A0      : 0x80382d89  A1      : 0x3fc9f8a0  
0x40382734: insert_free_block at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:243
 (inlined by) block_insert at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:269
 (inlined by) tlsf_free at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:876

A2      : 0x3fc9974c  A3      : 0xa5a5a5a5  A4      : 0x3fc97a94  A5      : 0x00060823  
A6      : 0x00060820  A7      : 0x00000001  A8      : 0x3fc9fad4  A9      : 0x00000017  
A10     : 0x0000001f  A11     : 0x3fc9fadc  A12     : 0x3fc9fad4  A13     : 0x3fc9f9e8  
A14     : 0x02ce0988  A15     : 0x00ffffff  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa5a5a5b1  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x40382731:0x3fc9f8a00x40382d86:0x3fc9f8c0 0x40375fca:0x3fc9f8e0 0x40383185:0x3fc9f900 0x40377882:0x3fc9f920 0x40377d08:0x3fc9f940 0x40377d55:0x3fc9f970 0x420095cf:0x3fc9f990 0x4200977e:0x3fc9f9c0 0x42009ccf:0x3fc9fa10 0x42009588:0x3fc9fa40 0x42007d8e:0x3fc9fa60 0x42008db4:0x3fc9fa80 0x42008f49:0x3fc9fac0 0x42008ff1:0x3fc9fc00 0x420071f5:0x3fc9fc30 0x4037f265:0x3fc9fca0 
0x40382731: insert_free_block at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:242
 (inlined by) block_insert at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:269
 (inlined by) tlsf_free at /home/zhuxy/esp/esp-idf/components/heap/heap_tlsf.c:876
0x40382d86: multi_heap_free_impl at /home/zhuxy/esp/esp-idf/components/heap/multi_heap.c:222
0x40375fca: heap_caps_free at /home/zhuxy/esp/esp-idf/components/heap/heap_caps.c:340
0x40383185: free at /home/zhuxy/esp/esp-idf/components/newlib/heap.c:39
0x40377882: uninstall_priv_desc at /home/zhuxy/esp/esp-idf/components/driver/spi_master.c:736
0x40377d08: spi_device_polling_end at /home/zhuxy/esp/esp-idf/components/driver/spi_master.c:983 (discriminator 15)
0x40377d55: spi_device_polling_transmit at /home/zhuxy/esp/esp-idf/components/driver/spi_master.c:1003
0x420095cf: wk_device_transmit at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/spi_wk2xxx.c:62
0x4200977e: wk_read_SFifo at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/spi_wk2xxx.c:195
0x42009ccf: wk_read_bytes at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/spi_wk2xxx.c:593
0x42009588: uart_port_read_bytes at /home/zhuxy/esp-projects/IoT-Station/components/drivers/src/uart_port.c:296
0x42007d8e: _modbus_rtu_recv at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus-rtu.c:204
0x42008db4: _modbus_receive_msg at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus.c:412
0x42008f49: read_registers at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus.c:1202
0x42008ff1: modbus_read_registers at /home/zhuxy/esp-projects/IoT-Station/components/libmodbus/src/modbus.c:1243
0x420071f5: modbusTask at /home/zhuxy/esp-projects/IoT-Station/main/main.c:84
0x4037f265: vPortTaskWrapper at /home/zhuxy/esp/esp-idf/components/freertos/port/xtensa/port.c:131`

chipweinberger, 2022-10-29:
I'd recommend you turn on comprehensive heap corruption detection in the config settings and try again.

This is likely corruption.

wishinlife, 2022-11-02:
I'm manual malloc tx and rx buffer:
`buff = heap_caps_malloc(256, MALLOC_CAP_DMA | MALLOC_CAP_32BIT);`
It works fine.
But I think it's not my code, it's the temporary DMA buffer malloc and free problem.

finger563, 2023-12-04:
I've been encountering this error as well (even on latest ESP-IDF master) and I believe the error occurs when `uninstall_priv_desc` is called as part of the `clean_up` after an ESP_ERR_TIMEOUT. I've added some logs to my code and I'm seeing that it's trying to free `buffer_to_send` even though it did not allocate it (since the buffer I'm providing is already DMA-capable and aligned).