espressif/esp-idf #10068: cJSON_Print memory leak

law-ko, 2022-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

cJSON library is being utilized at many places in ESP-IDF. However, it seems like `cJSON_Print()` is allocating memory to a pointer but freeing the char pointer does not fully freed the memory that `cJSON_Print()` has allocated.

The [issue](https://github.com/DaveGamble/cJSON/issues/715) has been created on the cJSON repo. However, this seemed to be an ESP-IDF issue as after reading some issues at cJSON repo the memory leak does not occur when doing `cJSON_Print()`.

The code in short will be:
```
char *json_string = cJSON_Print(jsonObject)
free(json_string)
```
This code will cause a memory leak regardless of the char pointer is freed.

Thank you.

KaeLL, 2022-10-27:
Try [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks) method instead before jumping to conclusions.

law-ko, 2022-10-27:
The console output for leap tracing:
```
BEFORE cJSON PRINT: free heap size = 294096       min_free_heap_size = 294096 
4 allocations trace (10 entry buffer)
16 bytes (@ 0x3ffb6f1c) allocated CPU 0 ccount 0x039035c0 caller 0x400e22d9:0x400da639
0x400e22d9: _dtoa_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/dtoa.c:234 (discriminator 1)

0x400da639: cvt at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1877
 (inlined by) _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1314

132 bytes (@ 0x3ffb6f30) allocated CPU 0 ccount 0x03905c00 caller 0x4008a60f:0x400e37fc
0x4008a60f: _calloc_r at /home/espuser/esp/esp-idf/components/newlib/heap.c:65

0x400e37fc: _Balloc at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:106

28 bytes (@ 0x3ffb6fb8) allocated CPU 0 ccount 0x039069d4 caller 0x4008a60f:0x400e3828
0x4008a60f: _calloc_r at /home/espuser/esp/esp-idf/components/newlib/heap.c:65

0x400e3828: _Balloc at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:127

24 bytes (@ 0x3ffb6fd8) allocated CPU 0 ccount 0x0390c534 caller 0x4008a60f:0x400e3828
0x4008a60f: _calloc_r at /home/espuser/esp/esp-idf/components/newlib/heap.c:65

0x400e3828: _Balloc at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:127

200 bytes 'leaked' in trace (4 allocations)
total allocations 6 total frees 2

AFTER cJSON PRINT: free heap size = 293880        min_free_heap_size = 293620
```

Portion of the code that tracks `cJSON_Print and free()` heap:
```
    printf("\nBEFORE cJSON PRINT: free heap size = %d \t  min_free_heap_size = %d \n",esp_get_free_heap_size(),esp_get_minimum_free_heap_size());
    heap_trace_start(HEAP_TRACE_LEAKS);
    string = cJSON_Print(monitor);
    free(string);
    heap_trace_stop();
    heap_trace_dump();
```

It shows 200 bytes are leaked in trace

ESP-Marius, 2022-10-28:
I think this is as expected. `_dtoa_r` (used e.g. while printing floats) will lazy allocate some buffers, but it will re-use these.

If you test by calling `cJSON_Print` multiple times then you should see that it doesn't actually leak memory per call.

law-ko, 2022-11-03:
> I think this is as expected. `_dtoa_r` (used e.g. while printing floats) will lazy allocate some buffers, but it will re-use these.
> 
> If you test by calling `cJSON_Print` multiple times then you should see that it doesn't actually leak memory per call.

Thanks @ESP-Marius , this is tested and confirmed to be the case.

closed 2022-11-03 by law-ko