espressif/esp-idf #10069: esp_hid_host fails to build with CONFIG_BT_SSP_ENABLED=n

boernsen-development, 2022-10-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1496-gbd18096cb4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

I expected the esp_hid_host example to build successfully when CONFIG_BT_SSP_ENABLED=n.

### What is the actual behavior?

The esp_hid_host example has build errors when CONFIG_BT_SSP_ENABLED=n.

### Steps to reproduce.

1. Check out bd18096
2. Install, export and copy esp_hid_host example according to [Get Started guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/)
4. Set CONFIG_BT_SSP_ENABLED=n in sdkconfig
5. Build

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running cmake in directory /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/boernsen/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/sdkconfig
Loading defaults file /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/sdkconfig.defaults...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/boernsen/.espressif/python_env/idf5.1_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
[... 56 lines cut ...]
-- Adding linker script /home/boernsen/esp-idf/master/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/boernsen/esp-idf/master/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/boernsen/esp-idf/master/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/boernsen/esp-idf/master/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/boernsen/esp-idf/master/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/boernsen/esp-idf/master/esp-idf/components/bootloader /home/boernsen/esp-idf/master/esp-idf/components/bootloader_support /home/boernsen/esp-idf/master/esp-idf/components/efuse /home/boernsen/esp-idf/master/esp-idf/components/esp_app_format /home/boernsen/esp-idf/master/esp-idf/components/esp_common /home/boernsen/esp-idf/master/esp-idf/components/esp_hw_support /home/boernsen/esp-idf/master/esp-idf/components/esp_rom /home/boernsen/esp-idf/master/esp-idf/components/esp_system /home/boernsen/esp-idf/master/esp-idf/components/esptool_py /home/boernsen/esp-idf/master/esp-idf/components/freertos /home/boernsen/esp-idf/master/esp-idf/components/hal /home/boernsen/esp-idf/master/esp-idf/components/log /home/boernsen/esp-idf/master/esp-idf/components/bootloader/subproject/main /home/boernsen/esp-idf/master/esp-idf/components/bootloader/subproject/components/micro-ecc /home/boernsen/esp-idf/master/esp-idf/components/newlib /home/boernsen/esp-idf/master/esp-idf/components/partition_table /home/boernsen/esp-idf/master/esp-idf/components/soc /home/boernsen/esp-idf/master/esp-idf/components/spi_flash /home/boernsen/esp-idf/master/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/bootloader
[100/101] Generating binary image from built executableesptool.py v4.3
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/bootloader/bootloader.bin
[101/101] cd /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/bootloader/e.../home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/bootloader/bootloader.binBootloader binary size 0x66e0 bytes. 0x920 bytes (8%) free.
[1146/1148] Linking CXX executable esp_hid_host.elfFAILED: esp_hid_host.elf c_esp32.c.obj
: && /home/boernsen/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=/home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/esp_hid_host.map -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld CMakeFiles/esp_hid_host.elf.dir/project_elf_src_esp32.c.obj  -o esp_hid_host.elf -L/home/boernsen/esp-idf/master/esp-idf/components/soc/esp32/ld   -L/home/boernsen/esp-idf/master/esp-idf/components/esp_rom/esp32/ld   -L/home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/esp-idf/esp_system/ld   -L/home/boernsen/esp-idf/master/esp-idf/components/esp_phy/lib/esp32   -L/home/boernsen/esp-idf/master/esp-idf/components/esp_wifi/lib/esp32   -L/home/boernsen/esp-idf/master/esp-idf/components/bt/controller/lib_esp32/esp32 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_pars [... cut ...]
/home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/../main/esp_hid_gap.c:419: undefined reference to `esp_bt_gap_set_security_param'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/log/idf_py_stderr_output_42064 and /home/boernsen/eclipse-workspace/esp_hid_host_master_20221027_163932_bd18096cb4/build/log/idf_py_stdout_output_42064
```

BetterJincheng, 2022-10-28:
@boernsen-development 

```
static esp_err_t init_bt_gap(void)
{
    esp_err_t ret;
#if (CONFIG_BT_SSP_ENABLED == true)
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
#endif
    /*
     * Set default parameters for Legacy Pairing
     * Use fixed pin code
     */
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
    esp_bt_pin_code_t pin_code;
    esp_bt_gap_set_pin(pin_type, 0, pin_code);

    if ((ret = esp_bt_gap_register_callback(bt_gap_event_handler)) != ESP_OK) {
        ESP_LOGE(TAG, "esp_bt_gap_register_callback failed: %d", ret);
        return ret;
    }

    // Allow BT devices to connect back to us
    if ((ret = esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE)) != ESP_OK) {
        ESP_LOGE(TAG, "esp_bt_gap_set_scan_mode failed: %d", ret);
        return ret;
    }
    return ret;
}
```
This can help.

boernsen-development, 2022-10-28:
@BetterJincheng That worked, thanks a lot!

closed 2022-11-28 by espressif-bot