espressif/esp-idf #10072: esp emac abort

SimonPVS, 2022-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v4.3.1 (but also higher versions)

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board with esp32 and phy LAN8742A

### Power Supply used.

External 3.3V

### What is the expected behavior?

When failing to malloc for the emac, a normal error code would appear.

### What is the actual behavior?

The processor crashes due to an abort in the "newlib/locks.c" file

### Steps to reproduce.

I have a program which uses udp, tcp and websockets.
The abort occured when using the websockets intensive.

I am not sure if the issue occures on sending or receiving.

[image]

chipweinberger, 2022-10-29:
I'd recommend you turn on comprehensive heap corruption in the config settings and try again.

kostaond, 2022-11-22:
Hi @SimonPVS, did you try to investigate if there is any heap corruption as suggested by @chipweinberger?
I would also need  more detailed steps in more controlled environment to be able to reproduce. The description is too generic if it is run in your own application...

SimonPVS, 2022-11-22:
Hi, 

It happened due to memory corruption.
So thanks for the replies!

kostaond, 2022-11-22:
Great to hear you've been able to solve the problem! Therefore I'm going to close this issue.

closed 2022-11-22 by kostaond