espressif/esp-idf #10073: ESP32-S3 BLE HID Output Reports Not Sending in Windows 10

yuanyihan, 2022-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

在蓝牙键盘demo中，我并不能获得win10-pc发给键盘的report，比如获得numlock和capslock，这个问题论坛等多次有出现。

论坛：https://esp32.com/viewtopic.php?f=13&t=19076&p=70573&hilit=CapsLock#p70573
git：https://github.com/espressif/esp-idf/issues/7898

我看到demo中，SUPPORT_REPORT_VENDOR是false，HID_RPT_ID_LED_OUT=0，但是我修改了部分代码，仍然无果。

我希望：
1.给出一个PC下发到esp32S3的demo，比如给出calllback函数，我处理LED在里面。
相同的问题在ESP32C3也存在，但我没仔细调试。

yuanyihan, 2022-10-29:
昨天忘了补充版本：
idf：5.0版本
pc：win10x64
同时4.4版本也不能获得我想要的report，并没有测试master版本，但我看到内容没有修改

xiewenxiang, 2022-10-31:
Hi @yuanyihan 
Some windows devices may not support this feature.
Which API did you used to send the Output Reports on windows and what is the return value of this API?

[Sending HID Reports](https://learn.microsoft.com/en-us/windows-hardware/drivers/hid/sending-hid-reports)

I saw a note here: 

`Some devices might not support HidD_SetOutputReport and will become unresponsive if this routine is used.`

yuanyihan, 2022-10-31:
@xiewenxiang 你貌似没明白我的反馈，这个问题并没有在win的HID方面，原因有而：
1. HID键盘是标准通用的设备，不需要win系统等特殊开发吧
2. USB-HID设备，在回调中，可以明显的设置capslock等状态，这代表着至少esp32的usb是能接到win系统的report（侧面说了不是win系统API等一系列的问题）
事实上，我使用stm32做的第一版设备，还有使用esp32S3做的usb设备均可正常的使用，但一旦使用了蓝牙，就没法接到报文了。
所以，问题在于蓝牙，而不在于win的HID（因为usb的HID是正常的）
而且，我的开发，仅在于esp32，wins并没有做任何的开发（没有上位机，也没有做任何的设置）

xiewenxiang, 2022-10-31:
@yuanyihan 
从开始问题描述中， "在蓝牙键盘demo中，我并不能获得win10-pc发给键盘的report"
这里是指 win10-pc 向键盘发送 report， 但是 ESP 设备无法收到？ 这里数据方向是PC发送到ESP设备?  两者是通过 BLE 连接吗？

另外后续补充的意思是，同一个PC，同样的代码在 ESP32 上面可以工作，但是在 ESP32S3上面无法工作吗？

yuanyihan, 2022-10-31:
@xiewenxiang 
1. win10-PC 通过**USB**发送给esp32S3，能够发送成功（另一个键盘敲击CAPSLOCK键，esp32S3设备能够亮灯/灭灯）【[采用idf5.0的例子，填充tud_hid_set_report_cb函数](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_hid)】
2. win10-PC 通过**BLE**发送给esp32S3，不成功（另一个键盘敲击CAPSLOCK键，esp32S3设备**没反应**）【[采用idf5.0的例子，并尝试做了修改，失败](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo)】
3. win10-PC 通过**BLE**发送给**esp32C3**，不成功（另一个键盘敲击CAPSLOCK键，esp32C3设备**没反应**）【[采用idf5.0的例子，并尝试做了修改，失败](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo)】
4. 我现有的设备仅有一个ESP32S3和两个ESP32C3，没有其他设备，三个设备均有此问题（C3没usb-hid）；PC用了两台win10，均有此问题。
------
从开始问题描述中， "在蓝牙键盘demo中，我并不能获得win10-pc发给键盘的report"
这里是指 win10-pc 向键盘发送 report， 但是 ESP 设备无法收到？ 这里数据方向是PC发送到ESP设备? 两者是通过 BLE 连接吗？
答：是的，PC向ESP设备发送，两者是BLE链接
------
另外后续补充的意思是，同一个PC，同样的代码在 ESP32 上面可以工作，但是在 ESP32S3上面无法工作吗？
答：我这里ESP32指ESP32-S3，，能工作是ESP32S3通过tinyusb可以工作；不能工作是ESP32S3通过BLE不能工作
------
上面说的CAPSLOCK灯和数据是如下的灯，是PC向HID设备的OUTPUT（这是标准的HID数据，不是自定义的数据）
```
    /* Output 5-bit LED Indicator Kana | Compose | ScrollLock | CapsLock | NumLock */ \
    HID_USAGE_PAGE  ( HID_USAGE_PAGE_LED                   )       ,\
      HID_USAGE_MIN    ( 1                                       ) ,\
      HID_USAGE_MAX    ( 5                                       ) ,\
      HID_REPORT_COUNT ( 5                                       ) ,\
      HID_REPORT_SIZE  ( 1                                       ) ,\
      HID_OUTPUT       ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE  ) ,\
```
-----
事实上，我这几天均在查找蓝牙键盘的问题，我又查了很多使用esp32系列做蓝牙键盘的例子，全github均未发现从PC得到output的方法，最接近的一个[带有capslock灯的键盘采用的是esp32内存储键盘状态，而不是通过PC下发report](https://github.com/hhuysqt/esp32s3-keyboard),事实上，我打log发现，如[esp32论坛](https://esp32.com/viewtopic.php?f=13&t=19076&p=70573&hilit=CapsLock#p70573)所述，显示底层都没有接收到数据

xiewenxiang, 2022-10-31:
@yuanyihan 

从您的描述中可以得出，应该是找一个 win10 的笔记本，然后烧录一个 demo 到 ESP 模组里，建立BLE连接就可以复现了吧？
方便提供一个能复现的最小 demo 和复现的步骤吗？

yuanyihan, 2022-10-31:
另外，esp-idf的[BLE-HID-demo](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo/main/ble_hidd_demo_main.c#L30)里面，明确写了：
```
/**
 * Brief:
 * This example Implemented BLE HID device profile related functions, in which the HID device
 * has 4 Reports (1 is mouse, 2 is keyboard and LED, 3 is Consumer Devices, 4 is Vendor devices).
 * Users can choose different reports according to their own application scenarios.
 * BLE HID profile inheritance and USB HID class.
 */
```
而且BLE-HID的keyboard已经成功（这个keyboard 指esp32s3设备向PC发送按键，叫input，是可以的）
但LED不成功（PC向esp32S3发送led状态，叫output，不可以）
注：标准的keyboard 包含两个部分，input和output，详情参考HID标准

yuanyihan, 2022-10-31:
@xiewenxiang 
您好，关于复现的问题，您可以直接使用idf的[demo](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo)，这个demo在esp32S3设备就不成功

xiewenxiang, 2022-11-01:
@yuanyihan 
还有一些不明确的地方，测试的场景是使用一个 win10的PC， 这个PC 连接了两个键盘（其中一个是 esp32s3）对吗？ 我目前是一台 win10 的台式PC，有线的方式连接了一个键盘，BLE方式连接了一个 esp32s3，烧录的代码是 ble_hid_device_demo

问题是按下另一个键盘上的 capslock 键，但是 没有收到 win10 pc 发给 esp32s3 的 report?

yuanyihan, 2022-11-01:
@xiewenxiang 是的，就是电脑（win10）发送（从电脑到每个键盘设备led状态）output给esp32S3（esp32的灯有反应），
没有另一个键盘也没关系，win10自带的键盘，也是可以的，就是按笔记本的capslock键，所有键盘的capslock灯（包括笔记本本身）应该有反应

yuanyihan, 2022-11-01:
哦哦，，， ble_hid_device_demo可能涉及部分代码的修改，主要是callback，，，就是esp32S3处理report的代码（但，根本接不到，别说处理了）。。。。我这边换一个说法：这不是我业务代码的问题，这是官方demo不能够实现LED，也就是demo说明中的2 is keyboard and LED不成功，而且极大可能是IDF-5.0的代码bug造成的不能接收

```
/**
 * Brief:
 * This example Implemented BLE HID device profile related functions, in which the HID device
 * has 4 Reports (1 is mouse, 2 is keyboard and LED, 3 is Consumer Devices, 4 is Vendor devices).
 * Users can choose different reports according to their own application scenarios.
 * BLE HID profile inheritance and USB HID class.
 */
```

xiewenxiang, 2022-11-01:
@yuanyihan 
感谢回复，我现在复测的办法应该和你描述的一致了。我做了两个场景的测试：

场景1

一台预装 win10 的台式机 PC( dell OptiPlex 3080) 连接一个键盘和esp32s3(ble_hid_device_demo)，同时我还抓取了空口包，我发现在实物键盘上敲 capslock, PC 是不会给 esp32s3 发送任何的数据包的，那么 esp32s3 也就是不可能会收到任何的数据, 我们可以把这个先简化为单纯的 BLE 问题，BLE HID 也是基于属性协议的上层应用，如果蓝牙空口没有任何包上来，那么 HID 这里肯定也是不会有数据，只从蓝牙层面来考虑的话，需要要看 PC 为什么没有把这个report发过来，这个行为是 HID 协议规定的吗？

场景2

我觉得如果直接连接2个市面上面的其他键盘，按照您前面的描述，应该是只操作一个，就能够控制两个键盘同时亮和灭，所以我在一个PC上面，用有线的方式连接两个键盘，我在 A 键盘上面敲 capslock ，只会 A 的灯亮，B 是不会有任何反应的，反之也是一样，所以，是不是也从另一个方面证明，PC 就不会向另一个键盘发送这些 report？

yuanyihan, 2022-11-01:
@xiewenxiang 
场景一，我个人认为，问题出在BLE层，通信数据底层是BLE，然后上层HID才进一步解析。
场景二，啥？我这边可以提供视频，你敲电脑的CAPSLOCK键，正常情况下，是所有从机设备（键盘）都接到电脑的output报文啊，，，这个灯的状态是master设备通知slave设备的，是实时同步的

逻辑关系是：
1. 任何的键盘敲击，都会通过input上报给PC设备（master设备），修改电脑的状态
2. 电脑把需要同步的5个LED状态，通过HID的output报文，实时分发给各个从机。
3. 键盘A的capslock按键和键盘A的capslock灯都不直接控制，而是按键上报给PC，PC下发LED给设备；所以正常是：任何一个键盘按下，所有灯都变吧

您有其他沟通效率更高的联系方式么？我给您的邮箱发了个电子邮件

xiewenxiang, 2022-11-01:
是的，我这里抓包，发现 PC 没有通过 BLE 发送任何的 ATT 层数据包，所以问题应该是先看为什么 WIN PC 那里没有通过 BLE 发送这个包。

yuanyihan, 2022-11-01:
@xiewenxiang 
我同事，做了mac电脑的简单测试（因为我不太懂mac，而同事不懂嵌入式领域，他是个前端js工程师，所以测试仅供参考）：

首先是同事的设备：
1. 系统版本10.15.7 (19H2)
2. 电脑型号MacBook Pro (13-inch, 2017, Two Thunderbolt 3 ports)
3. 硬件配置2.3 GHz 双核Intel Core i5；16 GB 2133 MHz LPDDR3
4. 有设置其他的一些属性，让其工作在caplock，并且使用其余USB机械键盘测试正常变灯。

其次，测试结果：
和wins一致，USB键盘可以正常工作，BLE-HID设备，只input（PC能接到esp设备发送的A字符），不output（PC不向设备发送LED状态）

所有测试，剔除了休眠的影响（休眠典型的特点是：A键盘按下capslock，B键盘不会亮灯，按下B键盘任一键，B键盘LED才和电脑保持一致：B键盘休眠，暂时不实时同步）
所有测试，剔除了键盘的特殊性（不对个性的键盘，比如都没有灯，或者灯状态来自键盘内部，而不是来自PC主机的特殊键盘测试，选用的是遵循标准HID协议的蓝牙或usb键盘）

xiewenxiang, 2022-11-01:
@yuanyihan 
我们也基于 mac 做了测试，两个同事两台 mac 都是一样，两个键盘不会同步 capslock 的灯，我们之间的测试肯定是有某些差异

yuanyihan, 2022-11-01:
@xiewenxiang 
话说，乐鑫没有测试工程师么？
乐鑫idf的demo写的是2是键盘的LED，然后后面写WIN10不支持Vendor devices，那么意思就是win10的蓝牙的LED部分测试通过了啊，，
为什么2021年11月份，有人提出issue，现在demo还是没有实现需求的感觉？

诉求：
给出一个能够工作的BLE-HID-demo，has 4 Reports (1 is mouse, 2 is keyboard and LED, 3 is Consumer Devices, 4 is Vendor devices)

xiewenxiang, 2022-11-01:
[image]

yuanyihan, 2022-11-01:
@xiewenxiang 我要求你的电话，或微信方式，，或者其他沟通方式，视频会议，zoom，腾讯会议，飞书均可。

yuanyihan, 2022-11-01:
@xiewenxiang mac电脑是有同步的，但可能同步比较复杂，因为如果是led由内部系统控制，那么逻辑有问题（我称之为系统脑裂），比如现在键盘A是大写状态，键盘B是小写状态，那么系统如果没有仲裁没法工作；但如果有仲裁，但是信息不下发给设备，那么有一个设备就和自己出错了，它的状态会混乱

关于win系统的同步问题，你那边的测试结果是啥？我这边在四个电脑，，，联想X1，小米笔记本，联想拯救者y7000，华硕ROG幻16均显示同步，而且我自己做的键盘（基于stm32）还特意关注了这个部分的报文，键盘内部没有存储，具体可以参考[我的项目](https://gitee.com/yuanyihan/temp_public/tree/master/EE/KEYBOARD/STM32_p60)
总之，BLE-HID-demo项目目前应该是有问题的，它没有按照实现项目要求实现功能
```
/**
 * Brief:
 * This example Implemented BLE HID device profile related functions, in which the HID device
 * has 4 Reports (1 is mouse, 2 is keyboard and LED, 3 is Consumer Devices, 4 is Vendor devices).
 * Users can choose different reports according to their own application scenarios.
 * BLE HID profile inheritance and USB HID class.
 */
```

同时，希望懂这些的朋友一起来看看这个问题，我只是一个算法工程师，我比较擅长的领域是NLP自然语言处理，今年国庆比较闲，所以自己做了个键盘。第一版采用STM32的芯片，使用USB方式。第二版想做个蓝牙的，但是在调试过程中，发现了这个问题，这才进行反馈

esp-cjh, 2022-11-02:
@yuanyihan 确实是我们的 example 有问题。主要是以下几点：
1 report map 中 LED report 的声明存在问题。
2 属性表中 LED Output Report Characteristic 的属性有问题，没有 write without response 属性。
3 LED output report ID 不对，应该设置为 2。

我修改后用笔记本电脑测试，按 capslock 可以收到 PC 发送的数据。我会尽快提交修复。
[image]

yuanyihan, 2022-11-02:
@xiewenxiang @esp-cjh 
很高兴听到这个消息，很期待您的代码。
期待在不久后的时间在**客制化**键盘的大军中，有一大批基于esp32S3/C3/S2的蓝牙键盘的出现。

esp-cjh, 2022-11-02:
@yuanyihan 
你可以先用我提供的 patch 测试一下，需要注意：
由于修改了属性表，但 PC 对于已配对的设备一般不会再次发现服务，所以需要在 PC 设置中先删除已绑定的设备，然后使用 `idf.py -p PORT erase-flash` 清除 ESP32S3 的配对信息，再烧写固件。PC 连接 ESP32S3 后会重新发现服务。

[0001-fix-ble-hid-led-output-report.zip](https://github.com/espressif/esp-idf/files/9917432/0001-fix-ble-hid-led-output-report.zip)

yuanyihan, 2022-11-02:
@xiewenxiang @esp-cjh 
感谢，经过测试，这个patch获得了win10的LED状态，在低1位的位置，获得了CAPSLOCK的状态，并且成功驱动灯，使得win10笔记本的自带键盘，外接USB键盘，外接蓝牙键盘，ESP32S3设备的状态灯得到了同步。
```
I (17904) HID_DEMO: ESP_HIDD_EVENT_BLE_LED_REPORT_WRITE_EVT
I (17904) HID_DEMO: 00
I (18894) HID_DEMO: ESP_HIDD_EVENT_BLE_LED_REPORT_WRITE_EVT
I (18894) HID_DEMO: 02
```
再次表示感谢

closed 2022-11-04 by yuanyihan