espressif/esp-idf #10074: Espefuse

ifoxbr, 2022-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I would like you to help me better understand the "espefuse" program.
When I use ESP32S3 DevKit on both ports of the Kit (USB and USB with CP2102) I can read the status of the fuses without any problems. 
But when I try to read the fuses on my ESP32S3 board through the USB port (without CP2102 / FTDI) I get the following error. Strange because the USB port connections are the same as in the Devkit.

> espefuse.py v4.3
Connecting...
Traceback (most recent call last):
  File "runpy.py", line 194, in _run_module_as_main
  File "runpy.py", line 87, in _run_code
  File "C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\espefuse.exe\__main__.py", line 7, in <module>
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\espefuse\__init__.py", line 277, in _main
    main()
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\espefuse\__init__.py", line 215, in main
    esp = get_esp(
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\espefuse\__init__.py", line 80, in get_esp
    esp = esptool.cmds.detect_chip(port, baud, connect_mode)
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\cmds.py", line 93, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 632, in connect
    last_error = self._connect_attempt(
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 573, in _connect_attempt
    self.sync()
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 436, in sync
    val, _ = self.command(
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 366, in command
    self.write(pkt)
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esptool\loader.py", line 318, in write
    self._port.write(buf)
  File "c:\espressif\python_env\idf5.0_py3.8_env\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

KonstantinKondrashov, 2022-11-01:
Hi @ifoxbr!
The espefuse uses esptool API to detect chips. Could you try to run `esptool chip_id`? If you get the same error then it means that the issue is not in the espefuse.
This error means that the esptool can not detect a chip, but usually, it throws another error message.

ifoxbr, 2022-11-02:
Hello Konstantin
My ESP32S3 is on COM12
When running esptool chip_id, it searches all COMs I have, and the result on COM12 is Timeout.

**Serial port COM12 Connecting... COM12 failed to connect: Write timeout**

The only differences in my hardware compared to Dev Kit is that I do not have 5Volt on my USB Side.
Could be this the error ?

Thank You

ifoxbr, 2022-11-03:
Hello Konstantin
That´s the problem.
I can't even run this simple command in espefuse.
Allways receive   "serial.serialutil.SerialTimeoutException: Write timeout" 
Tried on several other boards we have, same result.
It seems to be a problem with the batch of parts we bought.
Thank you for your help, I will buy some parts from some other supplier

Regards

KonstantinKondrashov, 2022-11-05:
Can you establish the connection through UART? It is a standard way to flash an app to the chip, and after running the efuse summary cmd.
I suggest making sure first. Please check as well the chip version (ECO version).
For a certain ECO version, there was a bug that this bit was set by default.

ifoxbr, 2022-11-07:
Thank you very much Konstantin
It works fine on UART port.
Have a nice day
Thank you again for help
[]s

KonstantinKondrashov, 2022-11-07:
STRAP_JTAG_SEL exists in the new esptool.  It enables selection between usb_to_jtag or pad_to_jtag through GPIO3. https://github.com/espressif/esptool/blob/master/espefuse/efuse/esp32s3/mem_definition.py#L152
Please try to use the newest esptool version.

ifoxbr, 2022-11-07:
Hello Konstantin

Thank you so much for the great help

Regards

ifoxbr, 2022-11-07:
Hello Konstantin
On the Esptool Github page there is something strange.
[https://github.com/espressif/esptool/releases](url)
If you look at the version history it shows the latest version 3.3.2, if you scroll down the page there is version 4.3.
I think the subtitle of the last version is wrong, it must be 4.3.2.
Is it from this repository that I should download?

Regards

KonstantinKondrashov, 2022-11-08:
Sorry, I see that you use esptool 4.3 https://github.com/espressif/esptool/releases/tag/v4.3, it is correct.
You did not expose which ESP32S3 you have. The esptool has esp32s3beta2 and esp32s3 targets. The esp32s3 has the right efuse table with STRAP_JTAG_SEL. Probably your chip is recognized as esp32s3beta2 . This is why you do not see this field.

ifoxbr, 2022-11-08:
Hi,
I understood.
This is my chip
```
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:f3:4e:84
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
MAC: 7c:df:a1:f3:4e:84
```

ifoxbr, 2022-11-08:
Hello Konstantin
Good news !!
I just received a new chip from our supplier, I changed it and now everything is perfect, the efuse appears and I was able to burn it.  
I can read efuses both USB and UART. 
Problem was our first batch of parts it's what I think.

Thank you so much for the great help

Regards

closed 2022-11-16 by dobairoland