espressif/esp-idf #10075: I2C stops working after calling i2s_set_pin

rbuehlma, 2022-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.0

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

Lolin D32

### Power Supply used.

USB

### What is the expected behavior?

It should always be possible to set the volume on a ES8388 through I2C.

### What is the actual behavior?

Any I2C call to ES8388 has no effect after calling `i2s_set_pin`

Setting the volume after calling `i2s_set_pin` has no effect.

### Steps to reproduce.

1. Use sample code found here: https://github.com/thaaraak/ESP32-ES8388/blob/master/main/main.c
2. Connect ES8388 to the ESP32 using the pins found in the code
3. Add `es8388_set_voice_volume( 20 );` before the call to `i2s_set_pin`
4. Upload to the ESP.
5. The volume is correctly set
6. Move `es8388_set_voice_volume( 20 );` after the call to `i2s_set_pin`
7. The volume is not set anymore and full volume is used

rbuehlma, 2022-10-29:
I now also tried to use the new API described here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html but the same issue remains. Do you have a hint on how this could be resolved or at least debugged?

L-KAYA, 2022-10-31:
May I ask if you tried to change the GPIO, and grab the signal on I2C pins using something like logic analyzer?

But according to your description, I'm inclined to guess it might be some feature of es8388. As far as I know, the pins of I2S and I2C won't affect each other as long as they are assigned to different GPIO.

I need further investigation

rbuehlma, 2022-10-31:
Yes, I tried to also use different GPIO ports for both, I2C and I2S. Unfortunately, I do not own any logic analyzer but I will try to get access to one. What exactly would be the best starting point to measure? SDL and SDA?

You got a good point, the problem might be on ES8388 side, I will attach another I2C device and the if that continues working. I will also try to reset ES8388 after the initialization of i2s to check if that makes any difference (should have come up with that on my own).

I will add the result here probably by tomorrow.

In any case, thanks for your help

L-KAYA, 2022-10-31:
Hi @rbuehlma , my colleague tested using the [es8388 driver](https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8388/es8388.c) in [ADF repo](https://github.com/espressif/esp-adf). And seems the volume can be set successfully after I2S pin assigned (tested by the [http_mp3 example](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_http_mp3)).

Could you try the following configuration for es8388 to see if it can help? 
```
Reg Addr: Value (in Hex)
0: 12
1: 50
2: 0
3: 9
4: 3c
5: 0
6: 0
7: 7c
8: 0
9: bb
a: 0
b: 2
c: c
d: 2
e: 30
f: 20
10: 0
11: 0
12: 38
13: b0
14: 32
15: 6
16: 0
17: 18
18: 2
19: 0
1a: 66
1b: 66
1c: 8
1d: 0
1e: 1f
1f: f7
20: fd
21: ff
22: 1f
23: f7
24: fd
25: ff
26: 0
27: 90
28: 28
29: 28
2a: 90
2b: 80
2c: 0
2d: 0
2e: 1e
2f: 1e
30: 0
31: 0
```

rbuehlma, 2022-11-01:
Thanks for your help. I also tried to use the pipeline_http_mp3 example you are referencing.

If I add this code:
`audio_hal_set_volume(board_handle->audio_hal, 1);`
at this line
https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_mp3/main/play_http_mp3_example.c#L64
the volume gets very low as expected. If I add the same line on
https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_mp3/main/play_http_mp3_example.c#L119
the volume is the same as without the line and I get the following errors in the log:
```
E (2671) I2C_BUS: /home/rbuehlma/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:104 (i2c_bus_write_bytes):I2C Bus WriteReg Error

E (2681) I2C_BUS: /home/rbuehlma/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:104 (i2c_bus_write_bytes):I2C Bus WriteReg Error
```
Does this help with debugging this problem? Could it be a hardware problem? I have build my own PCB containing the ES8388, may be I did some mistake there but I'm not sure if this is possible since both I2C and I2S are working fine independently.

rbuehlma, 2022-11-01:
I did another test: I have disconnected all the I2S cables between the Lolin D32 and the ES8388 and dumped all the registers from the ES8388. As long as the I2S on the Lolin is not initialized, I received the same values you posted above. As soon as the I2S bus get initialized, dumping the values takes more time, and the received values seem to be random.

So I guess this rules out any issues with the ES8388 and its PCB and the problem needs to be something with the ESP32. I will try the same with a different ESP32 tomorrow.

L-KAYA, 2022-11-01:
> https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_mp3/main/play_http_mp3_example.c#L119
the volume is the same as without the line and I get the following errors in the log:

We test the case by setting volume at [Line 78](https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_mp3/main/play_http_mp3_example.c#L87). But I guess it might not work as well on your side. If the example can not work, I suspect it might caused by the weak pull up of the I2C pins. Usually we need the additional pull-up in the hardware, could you take a try to pull-up the I2C pins with 4.7K resistance?

L-KAYA, 2022-11-02:
I used [ESP32 LytaT v4.3 Dev board](https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/get-started-esp32-lyrat.html) to test the code you provided, the result seems correct
[image]

I still suspect the pull-up resistance, because we've met I2C write failure before while using 10k pull-up. FYI, the `ESP32 LytaT v4.3` uses 2k pull-up.

L-KAYA, 2022-11-02:
The figure above is the result on the master, and it almost same on IDF-v4.4
[image]

rbuehlma, 2022-11-02:
Actually I have an update:

I have added another I2C device (MCP23017) on the same bus to test if only the ES83888 has an issue or the ESP32. For this, I have switched to using Arduino and the "Wire" library which is now used to initialize the I2C bus in ESP32 side... Now the issue is gone (even if the MCP23017 is not connected).

Unfortunately, I do not know what exactly has solved the problem. In de next days, I will try the above test code again with the current hardware constellation to verify that moving to the Wire library did actually solve the issue.

In the mean time, thank you very much for your support!

rbuehlma, 2022-11-09:
I can't reproduce this issue anymore. Maybe it was some issue with the wires (I've used a bunch of jumper cables). Now its working without issue on a customized PCB.

Thanks for your support.

closed 2022-11-09 by rbuehlma